greiman/SdFat #428: Program compactibility SDfat 1.1.4 vs 2.2.2.

Atlan4, 2023-04-09:
The program worked correctly with the 1.1.4 library, I want to switch to 2.2.2 and later so that it also supports unicode.
However, it currently does not work even when #define SD_FAT_TYPE is set to 1 or 0.
Can you advise me what adjustments to make to make it work (really, it says that it works once and when called again it no longer responds)

old code
`    } else if(key_command == 'q') {
    if(!MP3player.isPlaying()) {
      Serial.println(F("Directory and files found :"));
      SdFile file;
      char filename[60];
      sd.chdir(dirbuffer2,true); 
      uint16_t count = 10001;
      while (file.openNext(sd.vwd(),O_READ))
      {
        file.getName(filename, sizeof(filename));
        if ( filename>0 ) {
          SerialPrintPaddedNumber(count, 5 );
          Serial.print(F(": "));
          Serial.println(filename);
          count++;
        }
        file.close();
      }
      Serial.println(F("Enter Index of directory"));

    } else {
      Serial.println(F("Busy Playing Files, try again later."));
    }`
an attempt at new ones
`#define SD_FAT_TYPE 1 //or 0?

} else if(key_command == 'q') {
    if(!MP3player.isPlaying()) {
      Serial.println(F("Directory and files found :"));

      //SdFile file; //is wrong?
      char filename[60];
      sd.chdir(dirbuffer2);
      uint16_t count = 10001;

      while (file.openNext(&dir, O_RDONLY))
      {       
        file.getName(filename, sizeof(filename));
        if ( filename>0 ) {
          SerialPrintPaddedNumber(count, 5 );
          Serial.print(F(": "));
          Serial.println(filename);
          count++;
        }
        
        file.close();
      }
      Serial.println(F("Enter Index of directory"));

    } else {
      Serial.println(F("Busy Playing Files, try again later."));
    }`

I also searched in the answers and did not find an answer. Thanks for help. I'm worried that the ATmega328 doesn't have enough memory for something like that

greiman, 2023-04-09:
I can't guess what is happening.  

The above doesn't provide a clue of how much memory your app is using.

Enabling UTF-8 may take too much memory for your app on a 328.

Atlan4, 2023-04-09:
Free RAM = 314 Should be a base line of 1017, on ATmega328 when using INTx

and what about the code? why doesn't it work

greiman, 2023-04-09:
> and what about the code? why doesn't it work

I can't remotely debug your program. 

Everything takes more RAM with UTF-8.  temp buffers must be at least twice as big.  Directory entry names must be uint16_t instead of char and when converted to UTF-8 char buffers must be twice as long.  More stack is needed for deeper calls to convert UTF-16 in directories to UTF-8.

I ran the [UnicodeFileNames](https://github.com/greiman/SdFat/blob/master/examples/UnicodeFilenames/UnicodeFilenames.ino) example.   I added `#include "FreeStack.h"` to the example.

This call at the start of setup();

```
void setup() {
  FillStack();   // <<== Added
  Serial.begin(9600);
```

This call at the end:

```
  Serial.println(UnusedStack());  // <<== Added
  Serial.println("Done!");
}
```

The result of the compile:

> Sketch uses 15782 bytes (48%) of program storage space. Maximum is 32256 bytes.
> Global variables use 1030 bytes (50%) of dynamic memory, leaving 1018 bytes for local variables. Maximum is 2048 bytes.
> 

Output from the example:

> Type any character to begin
> ls:
>          0 😀/
>           20 россиянин
>           17 très élégant
>            9 狗.txt
> After remove and rmdir
> 647
> Done!

So required RAM for stack is: 1018 - 647 = 371 bytes.

I don't think you have enough RAM.

Atlan4, 2023-04-09:
OK transfer to ATmega2560... 

What is the difference when using SDFAT type=0 or when it is =1
Which should I use if I want it to still work as in sdfat1.1.4 (now I'm using library 2.2.2.)

it's already doing something with the address books, but it won't open the file. But at least theoretically it doesn't freeze.

SdFile file; do I have to use it there? code above.

well thank you

greiman, 2023-04-09:
I did a check of SdFat size for V1.1.2 vs V2.2.2

I used the [SdFatSize](https://github.com/greiman/SdFat/blob/master/examples/examplesV1/%23attic/SdFatSize/SdFatSize.ino) example.

Here are the results:
```
V1.1.2
Sketch uses 9422 bytes (29%) of program storage space. Maximum is 32256 bytes.
Global variables use 867 bytes (42%) of dynamic memory, leaving 1181 bytes for local variables. Maximum is 2048 bytes.

V2.2.2 with same options as V1.1.2
Sketch uses 9310 bytes (28%) of program storage space. Maximum is 32256 bytes.
Global variables use 875 bytes (42%) of dynamic memory, leaving 1173 bytes for local variables. Maximum is 2048 bytes.

V2.2.2 With new defauls options
Sketch uses 9846 bytes (30%) of program storage space. Maximum is 32256 bytes.
Global variables use 880 bytes (42%) of dynamic memory, leaving 1168 bytes for local variables. Maximum is 2048 bytes.

V2.2.2 UTF-8
Sketch uses 11166 bytes (34%) of program storage space. Maximum is 32256 bytes.
Global variables use 880 bytes (42%) of dynamic memory, leaving 1168 bytes for local variables. Maximum is 2048 bytes.
```

V2.2.2 has these extra options enabled by default:
```
#define USE_FAT_FILE_FLAG_CONTIGUOUS 1
#define ENABLE_DEDICATED_SPI 1
```
So V2 uses about the same amount of RAM as V1 for the same features.

Atlan4, 2023-04-09:
The code above, rewritten for use with 2.2.2. could it work correctly?
thank 
But in the future I would like to use unicode encoding with ASCII characters

greiman, 2023-04-09:
> But in the future I would like to use unicode encoding with ASCII characters

I will never support code pages so UTF-8 is your only choice.  

Unicode uses flash and RAM.  I made it as small as I could with a great deal of effort.  

The other option for microcontrollers is [FatFs](http://elm-chan.org/fsw/ff/00index_e.html).

```
/  To enable the LFN, ffunicode.c needs to be added to the project. The LFN function
/  requiers certain internal working buffer occupies (FF_MAX_LFN + 1) * 2 bytes and
/  additional (FF_MAX_LFN + 44) / 15 * 32 bytes when exFAT is enabled.
/  The FF_MAX_LFN defines size of the working buffer in UTF-16 code unit and it can
/  be in range of 12 to 255. It is recommended to be set it 255 to fully support LFN
/  specification.
```
This is over a KB more RAM than I use.

Face it, the 328 is too small.  Probably just a bit too small.

Probably it wouldn't work with code pages since they need to be translated to UTF-16 for long filenames.

greiman, 2023-04-09:
I noticed this:
```
file.getName(filename, sizeof(filename));
if ( filename>0 ) {
```
You should be using the return value from file.getName().

```
  /**
   * Get a file's name followed by a zero byte.
   *
   * \param[out] name An array of characters for the file's name.
   * \param[in] size The size of the array in bytes. The array
   *             must be at least 13 bytes long.  The file's name will be
   *             truncated if the file's name is too long.
   * \return length for success or zero for failure.
   */
  size_t getName(char* name, size_t size);
```

Atlan4, 2023-04-10:
Good day. Why does this code run for me only once and then it only writes done!

`#include "SdFat.h"

// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 1
/*
  Change the value of SD_CS_PIN if you are using SPI and
  your hardware does not use the default value, SS.
  Common values are:
  Arduino Ethernet shield: pin 4
  Sparkfun SD shield: pin 8
  Adafruit SD shields and modules: pin 10
*/
/*
// SDCARD_SS_PIN is defined for the built-in SD on some boards.
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = 9;
#else   // SDCARD_SS_PIN
// Assume built-in SD is used.
const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
#endif  // SDCARD_SS_PIN
*/
//const uint8_t SD_CS_PIN = SD_SEL;
const uint8_t SD_CS_PIN = 9;

// Try max SPI clock for an SD. Reduce SPI_CLOCK if errors occur.
#define SPI_CLOCK SD_SCK_MHZ(4)
/*
// Try to select the best SD card configuration.
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
#endif  // HAS_SDIO_CLASS
*/
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE
//------------------------------------------------------------------------------
// Store error strings in flash to save RAM.
#define error(s) sd.errorHalt(&Serial, F(s))
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(115200);

  // Wait for USB Serial
  while (!Serial) {
    yield();
  }
    // Initialize the SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  // Open root directory
  if (!dir.open("/")) {
    error("dir.open failed");
  }

  }
//------------------------------------------------------------------------------
void loop() {

  char buffer[6]; // 0-35K+null
  int16_t fn_index = atoi(buffer);
  char filename[60];
  uint16_t count = 10001;

  // Open next file in root.
  // Warning, openNext starts at the current position of dir so a
  // rewind may be necessary in your application.
  while (file.openNext(&dir, O_RDONLY)) 
        {
        file.getName(filename, sizeof(filename));
         if ( filename>0 ) {
          SerialPrintPaddedNumber(count, 5 );
          Serial.print(F(": "));
          Serial.println(filename);
          count++;
        }
        
    file.printFileSize(&Serial);
    Serial.write(' ');
    file.printModifyDateTime(&Serial);
    Serial.write(' ');
    file.printName(&Serial);
    if (file.isDir()) {
      // Indicate a directory.
      Serial.write('/');
    }
    Serial.println();
    file.close();
  }
  if (dir.getError()) {
    Serial.println("openNext failed");
  } else {
    Serial.println("Done!");
  }

delay(1000);
}

void SerialPrintPaddedNumber(int16_t value, int8_t digits ) {
  int currentMax = 10;
  for (byte i=1; i<digits; i++){
    if (value < currentMax) {
      Serial.print("0");
    }
    currentMax *= 10;
  }
  Serial.print(value);
}
`
[image]

Atlan4, 2023-04-10:
dir.rewind(); ok

Atlan4, 2023-04-10:
ok, thank. I finally figured it out, but I have another problem
I reduced the code, it starts playing one file on VS1053 on atmega328, but when running the code on ATmega2560, it doesn't work. I WILL look for where the problem is.

closed 2023-04-10 by greiman

Atlan4, 2023-04-11:
Version sdfat1.1.4 works on both atmega328 and ATmega2560.
I modified the program, it takes a lot of ram, so I need to use atmega2560.
I chose the code that takes care of loading the file from the SD card and writing the data to the vs1053 player. the code works when uploaded to atmega328 but does not work when uploaded to atmega2560.
So I assume that your 2.2.2 library contains some specific approaches that do not work on the atmega2560? you don't know what caused it?

`
#include <SPI.h>

//Add the SdFat Libraries
#include <SdFat.h>
#include <FreeStack.h>

//and the MP3 Shield Library
#include "vs1053_SdFat.h"

// Below is not needed if interrupt driven. Safe to remove if not using.
#if defined(USE_MP3_REFILL_MEANS) && USE_MP3_REFILL_MEANS == USE_MP3_Timer1
  #include <TimerOne.h>
#elif defined(USE_MP3_REFILL_MEANS) && USE_MP3_REFILL_MEANS == USE_MP3_SimpleTimer
  #include <SimpleTimer.h>
#endif

// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 1

//const uint8_t SD_CS_PIN = SD_SEL;
const uint8_t SD_CS_PIN = 9;

// Try max SPI clock for an SD. Reduce SPI_CLOCK if errors occur.
#define SPI_CLOCK SD_SCK_MHZ(1)
/*
// Try to select the best SD card configuration.
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
#endif  // HAS_SDIO_CLASS
*/

#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
//#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
//#define USE_FAT_FILE_FLAG_CONTIGUOUS   0

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE

vs1053 MP3player;
char raz=1;
int16_t last_ms_char; // milliseconds of last recieved character from Serial port.
int8_t buffer_pos; // next position to recieve character from Serial port.
//------------------------------------------------------------------------------
// Store error strings in flash to save RAM.
#define error(s) sd.errorHalt(&Serial, F(s))
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(115200);

  // Wait for USB Serial
  while (!Serial) {
    yield();
  }
    // Initialize the SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  // Open root directory
  if (!dir.open("/")) {
    error("dir.open failed");
  }

  uint8_t result; //result code from some function as to be tested at later time.
  //Initialize the MP3 Player Shield
  result = MP3player.begin();
  //check result, see readme for error codes.
  if(result != 0) {
    Serial.print(F("Error code: "));
    Serial.print(result);
    Serial.println(F(" when trying to start MP3 player"));
    if( result == 6 ) {
      Serial.println(F("Warning: patch file not found, skipping.")); // can be removed for space, if needed.
      Serial.println(F("Use the \"d\" command to verify SdCard can be read")); // can be removed for space, if needed.
    }
  }
      MP3player.setVolume(50,50); // commit new volume sluchatka

  }
//------------------------------------------------------------------------------
void loop() {

  // Below is only needed if not interrupt driven. Safe to remove if not using.
#if defined(USE_MP3_REFILL_MEANS) \
    && ( (USE_MP3_REFILL_MEANS == USE_MP3_SimpleTimer) \
    ||   (USE_MP3_REFILL_MEANS == USE_MP3_Polled)      )

  MP3player.available();
#endif
if (raz){

  char filename[60];
  uint16_t count = 10001;
  File32 dir;
  sd.chdir("/");
  dir.open("/");
  dir.rewind();

  char filename1[] = "05.mp3";
            Serial.println(filename1);
            int8_t result = MP3player.playMP3(filename1);
            //check result, see readme for error codes.
            if(result != 0) {
                            Serial.print(F("Error code: "));
                            Serial.print(result);
                            Serial.println(F(" when trying to play track"));            
                            }
      raz=0;
      }

delay(100);
}

void SerialPrintPaddedNumber(int16_t value, int8_t digits ) {
  int currentMax = 10;
  for (byte i=1; i<digits; i++){
    if (value < currentMax) {
      Serial.print("0");
    }
    currentMax *= 10;
  }
  Serial.print(value);
}

`

greiman, 2023-04-11:
I don't have time to debug your programs.

greiman, 2023-04-11:
One more thing.  The mp3 library does not use the standard SPI protocol for use in interrupt routines.  SdFat V2.x is not designed for use in interrupt routines and may not work.   The main problem will be conflict with SPI.beginTransaction() and use of millis() for timeouts.

Atlan4, 2023-04-12:
My mistake, I wrote File instead of File32 (in version 1.1.4 there was SDfile)
     //static SdFile track;
     static File32 track;

interestingly, it didn't matter on the 328 and it was a problem on the mega2560.

I still have a problem stopping the playback. open next does not work, but LS commands do.

greiman, 2023-04-12:
You probably still have an incompatible mixture of types.  The type for sd and file must match and be the same throughout your program.  This means in your code and the mp3 library.

Here are some choices:  

The simplest is to use SdFat for the type of sd and SdBaseFile, SdFile or File for the type of files.  These types are defined by typedefs so unless you edit SdFatConfig.h, 328 will have different actual types than mega.

SdBaseFile will use the least memory.

I would use SdFat for sd everywhere and SdBaseFile for files everywhere.

Here are the typedefs:
```
#if SDFAT_FILE_TYPE == 1 || defined(DOXYGEN)
/** Select type for SdFat. */
typedef SdFat32 SdFat;
/** Select type for SdBaseFile. */
typedef FatFile SdBaseFile;
#elif SDFAT_FILE_TYPE == 2
typedef SdExFat SdFat;
typedef ExFatFile SdBaseFile;
#elif SDFAT_FILE_TYPE == 3
typedef SdFs SdFat;
typedef FsBaseFile SdBaseFile;
#else  // SDFAT_FILE_TYPE
#error Invalid SDFAT_FILE_TYPE
#endif  // SDFAT_FILE_TYPE
//
// Only define File if FS.h is not included.
// Line with test for __has_include must not have operators or parentheses.
#if defined __has_include
#if __has_include(<FS.h>)
#define HAS_INCLUDE_FS_H
#warning File not defined because __has_include(FS.h)
#endif  // __has_include(<FS.h>)
#endif  // defined __has_include
#ifndef HAS_INCLUDE_FS_H
#if SDFAT_FILE_TYPE == 1 || defined(DOXYGEN)
/** Select type for File. */
typedef File32 File;
#elif SDFAT_FILE_TYPE == 2
typedef ExFile File;
#elif SDFAT_FILE_TYPE == 3
typedef FsFile File;
#endif  // SDFAT_FILE_TYPE
#endif  // HAS_INCLUDE_FS_H
/**
 * \class SdFile
 * \brief FAT16/FAT32 file with Print.
 */
class SdFile : public PrintFile<SdBaseFile> {
 public:
  SdFile() {}
  /** Create an open SdFile.
   * \param[in] path path for file.
   * \param[in] oflag open flags.
   */
  SdFile(const char* path, oflag_t oflag) { open(path, oflag); }

  static void dateTimeCallback(void (*dateTime)(uint16_t* date,
                                                uint16_t* time)) {
    FsDateTime::setCallback(dateTime);
  }
  /**  Cancel the date/time callback function. */
  static void dateTimeCallbackCancel() { FsDateTime::clearCallback(); }
};
```
Another way is to use the base types for FAT32/FAT16. in your code and the mp3 library.
Use the type SdFat32 for sd and FatFile for for files.  This will use minimum memory and use the same types on 328 and mega.

Atlan4, 2023-04-12:
The more I look into it, the less I understand it.

  track.close(); //Close out this track - In the original program
  file.close(); //added so that after stopping the playback, it is possible to open the folders again and start the playback

I currently have UTF8 enabled - you did it nicely :)
and the modification in SdFatConfig.h that you recommended
"Your best hope is to edit SdFatConfig.h here:"

Basically, the only thing that doesn't work for me is opening another folder....
Is there a reason why it doesn't work?
after calling this subroutine, it is no longer possible to use OPENNEXT, while using the command LS, it says that I switched to the current directory
 
`      File dir;
      char filename[60];
      sd.chdir(dirbuffer2);
      dir.open(dirbuffer2);
      dir.rewind();

      uint16_t count = 10001;
      while (file.openNext(&dir, O_RDONLY))
      {
        file.getName(filename, sizeof(filename));
        if (  filename>0 ) {

          if (count == fn_index) {
            Serial.print(F("Directory is: "));
            Serial.println(filename);
            break;
          }
          count++;
        }
        file.close();
      }
        Serial.println(filename);       
// end vyberu direktory
 
      Serial.println(F("Change directory to"));
      String(dirbuffer)=("/" + String(filename) + "/");
      Serial.println(dirbuffer);
      dirbuffer.toCharArray(dirbuffer2,60);

      sd.chdir(dirbuffer2);
      dir.open(dirbuffer2);
      dir.rewind();
`

Atlan4, 2023-04-13:
uint16_t count = 10001;
  while (file.openNext(&dir, O_RDONLY))
  {
    file.getName(filename, sizeof(filename));
    if (  filename>0 ) {

      if (count == fn_index) {
        Serial.print(F("Directory is: "));
        Serial.println(filename);
        break;
      }
      count++;
    }
    file.close();
  }
    file.close(); -----------------------------------add this and work it
    Serial.println(filename);

Atlan4, 2023-04-14:
Good day.  Could you tell me where in the sdfat source codes I can find the
subroutine that takes care of utf8 encoding/decoding.  When using your
library, the names of the files in the terminal are written with
diacritics, I would also like the output of the ID3 information from the
mp3 file to be written with diacritics.  I would like to use your
subprogram (function) or copy it.  Well thank you

greiman, 2023-04-14:
Here are files and locations where encode and decode of UTF-8 happends.

Search "FsUtf" (38 hits in 8 files of 117 searched)
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\common\FsName.cpp (4 hits)
	Line 27: #include "FsUtf.h"
	Line 38:     const char* ptr = FsUtf::mbToCp(next, end, &cp);
	Line 46:       ls = FsUtf::lowSurrogate(cp);
	Line 47:       rtn = FsUtf::highSurrogate(cp);
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\common\FsUtf.cpp (3 hits)
	Line  25: #include "FsUtf.h"
	Line  26: namespace FsUtf {
	Line 114: }  // namespace FsUtf
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\common\FsUtf.h (5 hits)
	Line  25: #ifndef FsUtf_h
	Line  26: #define FsUtf_h
	Line  33: namespace FsUtf {
	Line 100: }  // namespace FsUtf
	Line 101: #endif  // FsUtf_h
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\ExFatLib\ExFatFile.cpp (1 hit)
	Line  27: #include "../common/FsUtf.h"
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\ExFatLib\ExFatFilePrint.cpp (6 hits)
	Line  27: #include "../common/FsUtf.h"
	Line 196:         if (!FsUtf::isLowSurrogate(c)) {
	Line 200:         cp = FsUtf::u16ToCp(hs, c);
	Line 202:       } else if (!FsUtf::isSurrogate(c)) {
	Line 207:       } else if (FsUtf::isHighSurrogate(c)) {
	Line 214:       char* str = FsUtf::cpToMb(cp, buf, buf + sizeof(buf));
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\ExFatLib\ExFatName.cpp (6 hits)
	Line  27: #include "../common/FsUtf.h"
	Line 122:         if (!FsUtf::isLowSurrogate(c)) {
	Line 126:         cp = FsUtf::u16ToCp(hs, c);
	Line 128:       } else if (!FsUtf::isSurrogate(c)) {
	Line 133:       } else if (FsUtf::isHighSurrogate(c)) {
	Line 141:       ptr = FsUtf::cpToMb(cp, str, end - 1);
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\FatLib\FatFileLFN.cpp (2 hits)
	Line  27: #include "../common/FsUtf.h"
	Line 469:     path = FsUtf::mbToCp(path, path + 4, &cp);
  C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\FatLib\FatName.cpp (11 hits)
	Line  27: #include "../common/FsUtf.h"
	Line 132:         if (!FsUtf::isLowSurrogate(c)) {
	Line 136:         cp = FsUtf::u16ToCp(hs, c);
	Line 138:       } else if (!FsUtf::isSurrogate(c)) {
	Line 143:       } else if (FsUtf::isHighSurrogate(c)) {
	Line 151:       ptr = FsUtf::cpToMb(cp, str, end - 1);
	Line 314:         if (!FsUtf::isLowSurrogate(c)) {
	Line 318:         cp = FsUtf::u16ToCp(hs, c);
	Line 320:       } else if (!FsUtf::isSurrogate(c)) {
	Line 325:       } else if (FsUtf::isHighSurrogate(c)) {
	Line 332:       char* str = FsUtf::cpToMb(cp, buf, end);

greiman, 2023-04-14:
Seems like there must be a utility/editor to help with this problem.

I don't know much about ID3 information but I see on the web that it can have these encodings:

> 
> Frames that allow different types of text encoding contains a text encoding description byte. Possible encodings:
> 
> $00   ISO-8859-1 [ISO-8859-1]. Terminated with $00.
> $01   UTF-16 [UTF-16] encoded Unicode [UNICODE] with BOM. All
>       strings in the same frame SHALL have the same byteorder.
>       Terminated with $00 00.
> $02   UTF-16BE [UTF-16] encoded Unicode [UNICODE] without BOM.
>       Terminated with $00 00.
> $03   UTF-8 [UTF-8] encoded Unicode [UNICODE]. Terminated with $00.

Seems like there must be programs to convert these to UTF-8 which the most popular encoding for Unicode.

Atlan4, 2023-04-15:
Good day. I'm looking at it in another terminal. And there, your utf8
encoding is unreadable, and the ID3 encoding is accented.
Specifically:
Týzden is:utf8> 54-C3BD-64-65-6E
and ID3 generation ( ISO/IEC 8859-2 (Latin-2) ) 54-FD-64-65-6E
So I found out that in the nextion display editor I can choose which
encoding to use, but only one type. Probably utf8 is promising. So I will
have to recode the id3 information into utf8 forma.
Thanks for helping me.

greiman, 2023-04-15:
ISO/IEC 8859 was a good idea in 1987 but no modern app should use code pages.

Atlan4, 2023-04-15:
Good day.
  char infobuffer[30]; data ID3 tag I use this function for the output.
Serial.write(infobuffer, a);

It is possible to call one of your functions directly to convert it to
utf8,
i.e. the input to the function will be INFOBUFFER and the output can be
written to the WRITE or PRINT terminal.

thank

greiman, 2023-04-15:
> It is possible to call one of your functions directly to convert it to utf8,

No.  You need a function to convert from your ISO/IEC 8859 code page to Unicode encoded as UTF-8.

There are hundreds of code page identifiers. ISO/IEC 8859-2 (Latin-2) is just one of these so I don't have any - they are not supported in FAT LFN or exFAT filenames.

https://www.aivosto.com/articles/charsets-codepages.html

I only wrote functions to convert encodings of Unicode code points between UTF-8, UTF-16 and 32-bit.

greiman, 2023-04-15:
A simple history of code pages and ISO/IEC 8859 which originated in 1987.

https://www-user.tu-chemnitz.de/~heha/petzold/ch02b.htm