greiman/SdFat #429: STM32l432kc Inizialization failed whe i do the SD.begin

lraton, 2023-04-13:
I have swiched from the classic sd library to this (because is more update and with the oled diplay(i2c) he doesn't like it and with the pn532 (i2c) module he doesn't let him read the cards anymore) and when i do the SD.begin it return false, whit the classic library it work, but with this no

greiman, 2023-04-13:
Try the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example.

Post any error information.

lraton, 2023-04-13:
With that example it works 'Card successfully initialized.'

lraton, 2023-04-14:
Still block when i do display.display() (send buffer to oled, i found that probably the problem is the sram but i have 64 kb) and with the pn532 still won't let me scan the rfid/nfc card

greiman, 2023-04-14:
Which STM32 board package are you using?  I have seen stack size problems in some board packages when using SPI array transfers since a 512 byte tmp buffer is needed.

lraton, 2023-04-14:
I'm using the stm32 Duino https://github.com/stm32duino/Arduino_Core_STM32

lraton, 2023-04-19:
i don't know, i can't find a solution for the oled and sd toghether, nobody know the solution :(

closed 2023-05-02 by lraton