greiman/SdFat #430: Reading of same brand of card, only newer, fails with SD_CARD_ERROR_CMD18

devloloper, 2023-04-17:
Hi, 

I have encountered a problem with a brand of cards that suddenly stopped working when I purchased a new batch of them. Formatting them has not helped. There are differences such as the SD specification and the OEM used but I am not sure if those differences mean that the new card is incompatible in some way with either the hardware or the software being used.

I have tried with both SdFat 2.2.0 and 2.2.2, both give the same error.

Hardware:
Card: 16 Gb Superior Pro MicroSDHC [https://www.silicon-power.com/web/us/product-microSDHC_10](url)
Board: Arduino Due [https://store.arduino.cc/products/arduino-due](url)
Shield: Ethernet shield 2 [https://store.arduino.cc/products/arduino-ethernet-shield-2](url)

Logs from SDInfo are below:

Failed card:

```
Card type: SDHC

sdSpecVer: 6.00
HighSpeedMode: true
 
Manufacturer ID: 0X9F
OEM ID: TI
Product: SPCC 
Revision: 0.7
Serial number: 0X31480047
Manufacturing date: 3/2034
 
cardSize: 15556.67 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros
 
OCR: 0XC0FF8000
 
SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,30375936
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
 
volumeBegin failed. Is the card formatted?
SD errorCode: SD_CARD_ERROR_CMD18 = 0XC
SD errorData = 0X1F

```
Old working card:

```
Card type: SDHC
sdSpecVer: 5.00
HighSpeedMode: true
 
Manufacturer ID: 0X9C
OEM ID: SO
Product: USD00
Revision: 0.2
Serial number: 0X607003F1
Manufacturing date: 8/2018
 
cardSize: 15920.53 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros
 
OCR: 0XC0FF8000
 
SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,31086592
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
 
Scanning FAT, please wait.
 
Volume is FAT32
sectorsPerCluster: 64
fatStartSector:    8796
dataStartSector:   16384
clusterCount:      485600
freeClusterCount:  485573

```

greiman, 2023-04-17:
Could be a counterfeit.  This is a clue something is wrong.

> Manufacturing date: 3/2034 

A simple read blocks command failed.

> SD errorCode: SD_CARD_ERROR_CMD18 = 0XC
> SD errorData = 0X1F

Try the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example and edit this line to insure there is no interference from the Ethernet controller.

`const int8_t DISABLE_CHIP_SELECT = -1;`

greiman, 2023-04-17:
I found a Silicon Power micro SD and it does work on DUE and has a CID like yours.

> init time: 82 ms
> 
> Card type: SDHC
> sdSpecVer: 6.00
> HighSpeedMode: true
> 
> Manufacturer ID: 0X9F
> OEM ID: TI
> Product: SPCC
> Revision: 0.0
> Serial number: 0X6230000D
> Manufacturing date: 4/2021
> 
> cardSize: 31268.54 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> 
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,61063168
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT32
> sectorsPerCluster: 64
> fatStartSector:    9670
> dataStartSector:   24576
> clusterCount:      953856
> freeClusterCount:  429567
> 
> type any character to start

I ran the bench example and it works at high speed.

> 
> Type any character to start
> FreeStack: 92552
> Type is FAT32
> Card size: 31.27 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X9F
> OEM ID: TI
> Product: SPCC
> Revision: 0.0
> Serial number: 0X6230000D
> Manufacturing date: 4/2021
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 3843.20,53609,113,131
> 3855.05,48366,113,131
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4262.57,1245,117,118
> 4266.21,1243,117,118

devloloper, 2023-05-03:
You are probably correct in that they are either counterfeits or a bad batch. I tried the quickstart with no luck.

I have ordered some other microSD cards that I will try instead and I will use these cards for other projects as they seem to work fine in a PC.

closed 2023-05-03 by devloloper

greiman, 2023-05-03:
Often counterfeits work on a PC since they are configured for the SDIO protocol.  SPI is not used much.

Watch for problems as the card is used, sometimes counterfeits have less flash than stated and lose data.

See this:

https://photographylife.com/fake-memory-cards