greiman/SdFat #433: Seeedstudio W5500 Ethernet Shield

dakotahorstman, 2023-05-08:
Attempting to read any sort of information from the SD card using the `Sdinfo` sketch results in some hairy output from my SD cards (both a Gigastone 8GB and Kingston 16GB result in this below output). I am using the Seeedstudio W5500 Ethernet Shield here: https://www.seeedstudio.com/W5500-Ethernet-Shield.html

`Quickstart` is actually able to read the list of files off of each card; however, `SdFormatter` errors out by reporting the cards have a GB size of 0.

**SdInfo: (W5500 shield, Gigastone card)**
```
SdFat version: 2.2.2

Disabling SPI device on pin 10

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 44 ms

Card type: SDHC
sdSpecVer: 1.01
HighSpeedMode: false

Manufacturer ID: 0X3
OEM ID: ␀␀
Product: ␁␀␐␒␃
Revision: 2.0
Serial number: 0XC8803182
Manufacturing date: 8/2001

cardSize: 0.07 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: false
dataAfterErase: zeros

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X0,0X0,0X0,0X4,0XFE,0XC0,0XFF,0,12859392
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

volumeBegin failed. Is the card formatted?
```

I retested everything with a genuine Arduino Ethernet shield with the same cards and this works just fine.

**SdInfo: (W5100 shield, Gigastone card)**
```
SdFat version: 2.2.2

Disabling SPI device on pin 10

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 236 ms

Card type: SDHC
sdSpecVer: 5.00
HighSpeedMode: true

Manufacturer ID: 0X74
OEM ID: J`
Product: SD16G
Revision: 2.0
Serial number: 0X613
Manufacturing date: 11/2021

cardSize: 7989.10 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: ones

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X4,0X1,0X4,0XC,0XFE,0XC2,0XFF,2048,15601664
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT32
sectorsPerCluster: 8
fatStartSector:    2080
dataStartSector:   32496
clusterCount:      1946396
freeClusterCount:  1946395
```

It would be great if I could get assistance in exploring the root cause issues with the W5500 version.

dakotahorstman, 2023-05-08:
Note that I can write to the SD cards using `SD.h` from Arduino just fine with the W5500 shield. Just curious and hoping for some direction on a cause of these issues.

greiman, 2023-05-08:
I don't know anything about the W5500 shield so I can't help much.

SD.h is a wrapper for 1 2009 version of SdFat.  The default SPI speed for that version was very slow and there was no dedicated SPI option.

The default option for SdInfo is dedicated SPI at the highest SPI rate supported by the board less than or equal to 16 MHz.

I suggest that you edit the program in [this area](https://github.com/greiman/SdFat/blob/57900b21d21655c513ef7e344f55e8190e612dcb/examples/SdInfo/SdInfo.ino#L27) and select shared SPI at 4 MHz.

greiman, 2023-05-09:
Also edit this area:
```
/*
  Set DISABLE_CS_PIN to disable a second SPI device.
  For example, with the Ethernet shield, set DISABLE_CS_PIN
  to 10 to disable the Ethernet controller.
*/
const int8_t DISABLE_CS_PIN = -1;
```

dakotahorstman, 2023-05-09:
Okay, that got me on the right track. I actually had to set the dedicated clock to anything under 8MHz as I already set the `DISABLE_CS_PIN`.

`SdInfo` is correctly working now. Thanks!

closed 2023-05-09 by dakotahorstman

greiman, 2023-05-09:
The AVR SPI controller only allows clock divisors that are a power of two so 4 MHz is the actual frequency when you request anything in  the range 4 <= freqMhz < 8

There are seven divisors with 64 twice.

[image]

dakotahorstman, 2023-05-09:
Ah, so if I specify 6, 7, or 7.99, it is cut to 4 then. Did not know this, so thank you for the extra info.