greiman/SdFat #440: SdFat ESP32

andynvkz, 2023-08-08:
Tell me if I use the same code, it works on STM32, but does not see the file on ESP32

`  if (card.begin(SD_CS, SD_SCK_MHZ(25)))
  {
    File firmware=card.open("firmware.bin");
    if (firmware) {`
if you change File to SdFile, I get that SdFile is not compatible with the stream.
I'm trying to use update.h with sdcard

greiman, 2023-08-08:
Stream is not a parent of SdFile.

The typedef File is already used by the system in ESP32 so you probably need to use the type FsFile.

Here are possible typedefs for File:

```
// Only define File if FS.h is not included.
// Line with test for __has_include must not have operators or parentheses.
#if defined __has_include
#if __has_include(<FS.h>)
#define HAS_INCLUDE_FS_H
#endif  // __has_include(<FS.h>)
#endif  // defined __has_include
#ifndef HAS_INCLUDE_FS_H
#if SDFAT_FILE_TYPE == 1 || defined(DOXYGEN)
/** Select type for File. */
typedef File32 File;
#elif SDFAT_FILE_TYPE == 2
typedef ExFile File;
#elif SDFAT_FILE_TYPE == 3
typedef FsFile File;   // <------------------------ Default if not 32K flash AVR
#endif  // SDFAT_FILE_TYPE
#elif !defined(DISABLE_FS_H_WARNING)
#warning File not defined because __has_include(FS.h)
#endif  // HAS_INCLUDE_FS_H
```

andynvkz, 2023-08-09:
if you use FsFile, functions such as readDir and dirName are lost, it turns out they need to be transferred from FatFile to FsFile

andynvkz, 2023-08-09:
It’s very sad, but OTA over WIFI conflicts:
In file included from C:\Users\-\Documents\Arduino\libraries\SdFat\src/FatLib/FatLib.h:27,
                 from C:\Users\-\Documents\Arduino\libraries\SdFat\src/SdFat.h:33,
                 from D:\MP3Player\ESP32-VS1053\ESP32-VS1053.ino:5:
C:\Users\-\Documents\Arduino\libraries\SdFat\src/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
 class File : public FatFile, public Stream {
       ^~~~
In file included from C:\Users\-\Documents\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:27,
                 from C:\Users\-\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:23,
                 from D:\MP3Player\ESP32-VS1053\ESP32-VS1053.ino:3:
C:\Users\-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS\src/FS.h:47:7: note: previous definition of 'class fs::File'
 class File : public Stream
       ^~~~
In file included from C:\Users\-\Documents\Arduino\libraries\SdFat\src/FatLib/FatLib.h:28,
                 from C:\Users\-\Documents\Arduino\libraries\SdFat\src/SdFat.h:33,
                 from D:\MP3Player\ESP32-VS1053\ESP32-VS1053.ino:5:
C:\Users\-\Documents\Arduino\libraries\SdFat\src/FatLib/FatFileSystem.h: In member function 'fs::File FatFileSystem::open(const char*, oflag_t)':
C:\Users\-\Documents\Arduino\libraries\SdFat\src/FatLib/FatFileSystem.h:95:13: error: 'class fs::File' has no member named 'open'
     tmpFile.open(vwd(), path, oflag);
``

greiman, 2023-08-09:
> if you use FsFile, functions such as readDir and dirName are lost, it turns out they need to be transferred from FatFile to FsFile

SdFs and FsFile support both FAT32/FAT16 and exFAT.  exFAT does not have an equivalent structure and there are no short file names so there can be no readDir function.

readDir is only supported for FAT files.

If you need FAT32/FAT16 only  functions replace SdFat with SdFat32 and replace File with File32.

>  'class fs::File' has no member named 'open' tmpFile.open(vwd(), path, oflag);

Same problem.  FAT and exFAT have different properties so only functions common to both are in FsFile.

File32 has FAT only support.

ExFile has exFAT only support.

andynvkz, 2023-08-09:
If you need FAT32/FAT16 only functions replace SdFat with SdFat32 and replace File with File32.

how to turn it on?

'ESP32-VS1053:110:1: error: 'SdFat32' does not name a type; did you mean 'SdFat'?
 SdFat32     card(&SPISD);
 ^~~~~~~
 SdFat
ESP32-VS1053:111:1: error: 'File32' does not name a type; did you mean 'File'?
 File32      root;
 ^~~~~~
 File
ESP32-VS1053:112:1: error: 'File32' does not name a type; did you mean 'File'?
 File32      cur_file;
 ^~~~~~
 File
ESP32-VS1053:113:1: error: 'File32' does not name a type; did you mean 'File'?
 File32      cur_folder;
 ^~~~~~
 File
D:\MP3Player\ESP32-VS1053\ESP32-VS1053.ino: In function 'void find_tag()':
ESP32-VS1053:319:13: error: 'cur_file' was not declared in this scope
`

greiman, 2023-08-09:
I have no idea what you are doing.  Look at the examples like [QuickStart](https://github.com/greiman/SdFat/blob/57900b21d21655c513ef7e344f55e8190e612dcb/examples/QuickStart/QuickStart.ino#L26).