greiman/SdFat #443: Strategy for dealing with a corrupted file or filesystem?

cfeied, 2023-10-11:
Hi Bill,

I know you have addressed related questions in the past, and I have read all those threads I could find both here and elsewhere. Most of them focus on preventing corruption. My question is slightly different: I'm not seeking to avoid corruption -- at this point I'm trying to understand what options may exist once corruption occurs.

1. Is reformatting the drive the only realistic option for recovery?

2. Is there any diagnostic function or series of functions that can be called to screen a card for health vs corruption?

3. Is there any method for testing whether a specific file is affected by corruption?

3. Is there any present or future possibility of removing all corrupted files and restoring the file system to health while leaving unaffected files in place? 

In our setting, the most common manifestation of corruption is inability to open a file for writing with certain flag patterns, and subsequently the inability to delete that file using ordinary SdFat commands. Forcibly deleting such a file using a PC utility results in later problems with the card. Every case that I have seen appears to have resulted from loss of power during a period of file system activity. The file open for writing is generally the one affected, but there is soft evidence that additional files may sometimes be affected even if they were previously written and closed. 

It would be highly desirable to be able to repair such a card in situ, losing any affected files while preserving other files that are unaffected. If this is impossible, it would still be useful if we could verify which files were unaffected, so we could copy off all verifiably "good" files, reformat, and copy the files back.  I understand that this capability does not exist today -- my question is whether it is even possible or whether it's impossible, given the way SD cards work.

Thanks!

Craig

greiman, 2023-10-11:
Unfortunately verifying or repairing a FAT/exFAT file system is not practical on a MCU.

Here are two links that give you an idea what is involve.

The algorithms in these papers might work for very small files on small SD cards.  The time is not linear with size so it could take many hours for 32 GB or larger SD cards.

See this:

https://elinux.org/images/5/54/Elc2011_munegowda.pdf

and

https://www.enggjournals.com/ijet/docs/IJET14-06-04-708.pdf

This is beyond anything I have interest in pursuing.

On Windows or Linux You could use a GB of memory with access times in nanoseconds instead of milliseconds for external storage.

cfeied, 2023-10-12:
Thank you for your answer and for the references. 

Long ago I worked on a log-based filesystem (8080 assembler) for CP/M (I'll be raising a glass to Gary Kildall tonight).

I'll just have the application log file openings and closings. If there's a missing close() at program startup I'll copy everything off, reformat, copy everything back, and validate crc where appropriate.

closed 2023-10-12 by cfeied