greiman/SdFat #444: Refresh SD-Card !!!

DR-DNK, 2023-10-28:
Hello , I am using ESP32S3 and SD Card to perform file transfer via FTP . My ESP32S3 is a ftp server and filezilla is a FTP client.  I am transfering file  from filezilla to my esp32s3 which is working successfully . Now I have added another functionality of USBMSC , so I am able to mount my esp32s3 which is attached to SD-Card as mass storage device . So I am able to view files of my SD-Card. 

Now when my file is transferred successfully into my SD-Card via FTP . I am not able to see my files . For that every time I have to unplug my device and plug it again to see the changes .

I want my device to refresh itself so that the file changes are updated automatically .

Does SdFat or USBMSC has that functionality to refresh SD-Card by itself ?

Libraries I am using are :
SdFat v2.2.2
SimpleFTPServer v2.1.7
USBMSC

greiman, 2023-10-28:
Seems like it shouldn't work if you are accessing the SD as a USBMMC device and by SimpleFTPServer.  Both probably cache SD info.

I don't know anything about SimpleFTPServer  or ESP32 USBMSC so I can't help you

DR-DNK, 2023-11-03:
Hey !!! @greiman  ,
Can you please guide me with the following .
When I am dragging and dropping files in Sd-Card , the new changes are getting reflected .But when I am transferring files via FTP client , to see the changes , I have to unplug and plug the controller again , so that Sd-Card is re-initialized .
I have used sync function , but I found out that sync only works with changes within files.

My question is why the changes are getting reflected via drag and drop and not via FTP file transfer ?

I assure you with one thing that my file which is transfered via FTP  is there within the Sd-Card as I am able to get it listed via ls function , and also I am able to see it after plugging my controller again .

greiman, 2023-11-03:
If and SD card is accessed by two programs things like this will happen.  Info about the SD is cached in both programs.

Don't be surprised if you SD becomes corrupt if you mount it as an USB device and access it with a another program.

If microcontrollers had a true file system layer like Linux, Mac OS, or Windows this wouldn't happen.

DR-DNK, 2023-11-04:
Hey @greiman , you might be right . 
But when I am using ls function , provided in SdFat library ,
`sd.ls();`
the new file which is getting transferred via FTP client to the controller is getting listed in the Serial monitor which assures that the file resides in the SD-Card . It's just that the SD-Card is not displaying that file or the changes made from the filezilla are not reflected .

What I think  is if changes made by dragging and dropping files , is reflected and read by the SD-Card why the file which is transferred via filezilla is not reflecting here .

Hey @greiman , I really need your help here.

greiman, 2023-11-04:
I told you twice, there are multiple caches for the SD file structures.   SdFat does not know another program is making changes to the SD.  SdFat may have a copy of directory entries so changes won't be reflected in ls().

If you are careful to never use SdFat to modify the SD, you might be able to do the following.  If SdFat and FTP both modify files, the SD will be corrupt.

```
   if (!sd.begin(<your arguments>)) {
     // handle error
   }
   sd.ls();  // will  read directory into cache.
   sd.end();  // will flush cache
```
WARNING: This will reinit the SD and may cause the FTP program to fail or corrupt the SD.

Again, allowing two programs to have raw access a file device is always dangerous.