greiman/SdFat #446: Help needed with accessing files from SD flash from both the cores in esp32

rshashikanth, 2023-11-02:
Hi Greiman,

Thank you for this awesome library and your work.

I am using Esp32 and SD flash to read wav files and play it through I2S

I use your library to read the files from Flash, when I am accessing the flash from any one of the core of esp32, it works fine

But, if I try to access flash from both the cores simultaneously, esp32 restarts continuously

My intention is to play wav files, check if it is running, repeat etc. from core 1
and run the wav buffer continuously on core 0

If I run everything in a single core, it works but very slow

Everything  was working fine, when I was using SD library, like reading from both the cores
but it takes a lot of Free heap and literally none is left to run wifi and other stuff

But when I discovered your library, I was happy for how less Free hep memory it consumes

Please help, in resolving this issue. Just wanted to read files from the flash or SD card from both the cores of esp32

Thanks again for your great work

greiman, 2023-11-02:
SdFat is not multi-core safe.  You will need to use a mutex or other method to fix it.  SdFat will then likely also be slow since checking the status will block the playing calls.

There are too many problems with conflicts on ESP32 so I no longer try fix problems like this on ESP32.

rshashikanth, 2023-11-03:
Thank you, after using mutex and much tweaking, I am able to use your library to run multiple wav files at once in a buffer with altered speeds. That to with a fraction of free heap memory which SD was consuming

Thanks again