greiman/SdFat #448: How to modify the timestamp of a folder1

jie326513988, 2023-11-04:
How to modify the timestamp of a folder1
Hello
I already know how to change the timestamp of a file, 
but I can't change the timestamp of a folder no matter what I try. Can you help me? 
Thanks

```
SdFs sd;
// Set the file time to the current time
bool setFileTime_current(String path)
{
  // Check whether there is a clock chip on the I2C address. Check whether the power is off. Check whether the data is logical.
  if (clockChipPowerDown() && clockChipReadCheck())
  {
    dateTime = adapter.readDateTime(); // Get clock data
    FsFile uf = sd.open(path, O_WRITE);
    uint16_t n = dateTime.year + 2000, y = dateTime.month, r = dateTime.dayOfMonth,
             s = dateTime.hour, f = dateTime.minute, m = dateTime.second;
    uint8_t ok = 0;
	
    if(uf.timestamp(T_ACCESS, n, y, r, s, f, m)) {Serial.println("T_ACCESS"); bitWrite(ok, 0, 1);}
    if(uf.timestamp(T_CREATE, n, y, r, s, f, m)) {Serial.println("T_CREATE"); bitWrite(ok, 1, 1);}
    if(uf.timestamp(T_WRITE, n, y, r, s, f, m))  {Serial.println("T_WRITE");  bitWrite(ok, 2, 1);}
	
    uf.close();
	
    if (ok == B00000111) return 1;
    else return 0;
    
  }
  return 0;
}
```

greiman, 2023-11-04:
SdFat has always checked for a file in timestamp.  You could try modifying this [line](https://github.com/greiman/SdFat/blob/57900b21d21655c513ef7e344f55e8190e612dcb/src/FatLib/FatFile.cpp#L1284) and this [line](https://github.com/greiman/SdFat/blob/57900b21d21655c513ef7e344f55e8190e612dcb/src/ExFatLib/ExFatFileWrite.cpp#L466).

Change `isFile()`  to `isFileOrSubDir()`.

I have not tested this.  Let me know if it works.

jie326513988, 2023-11-04:
Thank you very much, it worked
The files that need to be modified are \SdFat\src\FsLib\FsFile.h

```
// 设置文件或文件夹的时间戳为当前的时间
bool setFileTime_current(String path)
{
  // 检查I2C地址上有无时钟芯片 检查是否掉电 检查数据是否符合逻辑
  if (clockChipPowerDown() && clockChipReadCheck())
  {
    //i2c_init();
    dateTime = adapter.readDateTime(); // 获取时钟数值
    SdFile dir;
    FsFile uf = sd.open(path, O_WRITE);
    if(!uf) uf = sd.open(path);  // 当文件打开失败时以文件夹方式打开
    //Serial.println("*** 设置时间 ***");
    //Serial.print("path:");Serial.println(path);
    //Serial.print("uf:");Serial.println(uf);
    uint16_t n = dateTime.year + 2000, y = dateTime.month, r = dateTime.dayOfMonth,
             s = dateTime.hour, f = dateTime.minute, m = dateTime.second;
    uint8_t ok = 0;
    if(uf.timestamp(T_ACCESS, n, y, r, s, f, m)) {Serial.println("T_ACCESS"); bitWrite(ok, 0, 1);}
    if(uf.timestamp(T_CREATE, n, y, r, s, f, m)) {Serial.println("T_CREATE"); bitWrite(ok, 1, 1);}
    if(uf.timestamp(T_WRITE, n, y, r, s, f, m))  {Serial.println("T_WRITE");  bitWrite(ok, 2, 1);}
    uf.close();
    if (ok == B00000111) return 1;
    else return 0;
  }
  return 0;
}
```

closed 2023-11-04 by jie326513988