greiman/SdFat #453: SdFat SDIO on RP2040 with USB stick support

Stass48, 2023-11-16:
For now, the question is to quickly copy files from a USB drive to an SD card, using SDIO for the memory card and GPIO to implement a USB host on the RPI PICO. I found a possibility to implement a USB host:

[PIO_USB.zip](https://github.com/greiman/SdFat/files/13380209/PIO_USB.zip)

To run the example you need to install the libraries:

- Adafruit_TinyUSB_Library (With all dependencies)
- Pico_PIO_USB (With all dependencies)

As you may have noticed, this example uses the SdFat Adafruit Fork library, however, you can remove it and install the latest version of SdFat, you just need to edit the parameter:

#ifndef USE_BLOCK_DEVICE_INTERFACE
#define USE_BLOCK_DEVICE_INTERFACE **1**
#endif  // USE_BLOCK_DEVICE_INTERFACE

In this case, the compiler produces a lot of Warnings (unlike the SdFat Adafruit Fork library), but the sketch still compiles and works.
So I would like to know from you, as the author, whether it is possible to somehow copy files from a USB drive to an SD card using SDIO?
Thank you very much for your answer and help.

greiman, 2023-11-16:
I have no interest in USB MSC for RP2040 using Adafruit_TinyUSB_Library and Pico_PIO_USB.

PIO SDIO would be total overkill for access to the SD.  SPI would probably be fast enough.  

I did a PIO SPI for SdFat [here](https://github.com/greiman/SdFat/pull/445#issuecomment-1790719432).  I get about 7,100 KB/sec read/write to a SD.

At some point I will release PIO SDIO for SD cards on SdFat.  I currently get about 25,000 KB/Sec read/write.

Let me know how fast read is from the USB drive.

greiman, 2023-11-17:
Forget about using any of my fast PIO with the Adafruit_TinyUSB_Library and Pico_PIO_USB.  

Pico_PIO_USB uses almost all of the two PIO memories.

I suspect it will be slow since the decode loops for USB NRZI decoding take lots of instructions.

Stass48, 2023-11-18:
> Let me know how fast read is from the USB drive.

I'm trying to implement copying, but I can't do it. The code in this form simply reads the file, but the reading never ends. I do not understand why(

```
#include "SPI.h"
#include "SdFat.h"
#include "usbh_helper.h"

Adafruit_USBH_MSC_BlockDevice msc_block_dev;
FatVolume fatfs;
bool is_mounted = false;
bool usb_ready = false;
bool usb_checked = false;

// SD
SdFat sd;
const uint8_t SD_CS = SS;  // chip select for sd

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(10);

  USBHost.begin(1);

  Serial.println("Init SD card...");
  if (sd.begin(SD_CS, SPI_FULL_SPEED)) {
    Serial.println("SD OK!");
  } else {
    Serial.println("SD ERROR!");
  }

  while (!usb_checked) delay(1);

  if (usb_ready) {
    Serial.println("USB OK!");
  } else {
    Serial.println("NO USB!");
  }

  Serial.println("Starting copying file");
  copyFile("bench.dat", "bench.dat");
}

void setup1() {
  rp2040_configure_pio_usb();
}

void loop() {
  USBHost.task();
}

void loop1() {
  USBHost.task();
}

void copyFile(const char* srcFileName, const char* destFileName) {
  File32 srcFile = fatfs.open(srcFileName, O_READ);
  if (!srcFile) {
    srcFile.close();
    Serial.println("Error while file opening for reading!");
  } else {
    Serial.println("File opened for reading!");
  }

  // File32 destFile = sd.open(destFileName, FILE_WRITE);
  // if (!destFile) {
  //   destFile.close();
  //   Serial.println("Error while file opening for writing!");
  // } else {
  //   Serial.println("File opened for writing!");
  // }

  size_t buf = 4096;
  uint8_t data[buf];
  while ((buf = srcFile.read(data, buf)) > 0) {
    // destFile.write(data, buf);
  }

  srcFile.close();
  // destFile.close();

  Serial.println("Done");
}

extern "C" {

  // Invoked when device is mounted (configured)
  void tuh_mount_cb(uint8_t daddr) {
    (void)daddr;
  }

  /// Invoked when device is unmounted (bus reset/unplugged)
  void tuh_umount_cb(uint8_t daddr) {
    (void)daddr;
  }

  // Invoked when a device with MassStorage interface is mounted
  void tuh_msc_mount_cb(uint8_t dev_addr) {

    // Initialize block device with MSC device address
    msc_block_dev.begin(dev_addr);

    // For simplicity this example only support LUN 0
    msc_block_dev.setActiveLUN(0);

    is_mounted = fatfs.begin(&msc_block_dev);

    if (is_mounted) {
      usb_ready = true;
    } else {
      usb_ready = false;
    }

    usb_checked = true;
  }

  // Invoked when a device with MassStorage interface is unmounted
  void tuh_msc_umount_cb(uint8_t dev_addr) {
    Serial.printf("Device removed, address = %d\r\n", dev_addr);

    // unmount file system
    is_mounted = false;
    fatfs.end();

    // end block device
    msc_block_dev.end();

    usb_ready = false;
    usb_checked = true;
  }
}
```

> Init SD card...
> SD OK!
> USB OK!
> Starting copying file
> File opened for reading!

Stass48, 2023-11-18:
Я пока что хочу попробовать обычный SPI и PIO USB. Строки вида:

**File** srcFile = fatfs.open(srcFileName, O_READ);
выдает ошибку компиляции, то же самое касается карты памяти:
File destFile = sd.open(destFileName, FILE_WRITE);

File32 srcFile = fatfs.open(srcFileName, **O_READ**);
или 
File32 srcFile = fatfs.open(srcFileName, **FILE_READ**);
ни на что не влияет.

greiman, 2023-11-18:
I looked at the PIO USB and don't want to waste my time.  I am closing this.

closed 2023-11-18 by greiman