greiman/SdFat #454: STM32duino and SdFat

JFBow, 2023-11-19:
Hello Greiman,
I used, under IDE Arduino 2.2.1, STM32duino 2.6.0 and SdFat 2.2.2 : perfect, it worked perfectly. 
Since I upgraded from STM32duino 2.6.0 to 2.7.0, it generates errors, here they are:

```
c:\Users\JF\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiSTM32Core.cpp: In member function 'void SdSpiArduinoDriver::send(const uint8_t*, size_t)':
c:\Users\JF\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiSTM32Core.cpp:63:18: error: no matching function for call to 'SPIClass::transfer(uint8_t*, uint8_t [512], size_t&)'
   63 |   m_spi->transfer(const_cast<uint8_t*>(buf), rxBuf, count);
      |   ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\JF\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiDriver.h:90,
                 from c:\Users\JF\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiSTM32Core.cpp:26:
C:\Users\JF\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.7.0\libraries\SPI\src/SPI.h:202:21: note: candidate: 'virtual uint8_t SPIClass::transfer(uint8_t)'
  202 |     virtual uint8_t transfer(uint8_t _data);
      |                     ^~~~~~~~
C:\Users\JF\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.7.0\libraries\SPI\src/SPI.h:202:21: note:   candidate expects 1 argument, 3 provided
C:\Users\JF\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.7.0\libraries\SPI\src/SPI.h:204:18: note: candidate: 'virtual void SPIClass::transfer(void*, size_t)'
  204 |     virtual void transfer(void *buf, size_t count);
      |                  ^~~~~~~~
C:\Users\JF\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.7.0\libraries\SPI\src/SPI.h:204:18: note:   candidate expects 2 arguments, 3 provided

exit status 1

Compilation error: exit status 1
```

Is there a solution by keeping the STM32duino 2.7.0 version ?

greiman, 2023-11-20:
I will remove the custom STM32 code.  

For now use the standard Arduino API by editing SdFatConfig.h and change this.
`
#define SPI_DRIVER_SELECT 1  // << was zero`

> Can I quote you on the provision of fast SPI DMA by STMicroelectronics ?

Yes.  Bet they will hear form other users with apps that need fast SPI.

STM32 is now slower than an UNO R3

 Uno R3:
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 689.18,760,728,736
> 689.18,760,728,736
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 663.75,776,760,765
> 663.75,776,760,765

 STM32 Nucleo L476RG
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 528.65,983,965,966
> 528.60,983,965,967
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 519.59,995,982,983
> 519.59,995,982,983

greiman, 2023-11-20:
You can do better by using this in SdFatConfig.h

```
#define SPI_DRIVER_SELECT 1
...
#define USE_SPI_ARRAY_TRANSFER 1
```
STM32 Nucleo L476RG
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1101.08,480,462,463
> 1100.84,480,462,463
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1095.05,478,465,465
> 1094.57,477,465,466

Please don't close this since I don't want to change SdFat for a while since I will break all apps using other versions of the STM32 board package before 2.7.0.

I may just modify the STM32 code and keep a custom driver using `transfer(buf, count)`.

Still the `transfer(buf, count)` API can be much faster with DMA.

greiman, 2023-11-20:
Here is Teensy 4.1 using DMA SPI with:
```
#define SPI_DRIVER_SELECT 1
...
#define USE_SPI_ARRAY_TRANSFER 1
```

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 5175.98,1221,98,98
> 5186.72,110,98,98
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 5181.35,107,98,98
> 5213.76,106,98,98

So even with the standard Arduino API you can go fast.

Teensy 4.1 also has 4-bit SDIO:

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 21929.82,48,22,22
> 21645.02,414,22,23
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 22727.27,167,22,22
> 22624.43,166,22,22

All of these test used 512 byte read/write size.

fpistm, 2023-11-21:
Hi,
I will provide a PR which will be able to support all core version.
About DMA, I know. Time is always the missing ingredient. Anyway, any contribution are welcome :)

greiman, 2023-11-21:
@fpistm
Why should I waste time on STM32?

Here is a logic analyzer trace of transfer(buf, count)  40 MHz clock.  almost 600 ns dead time between bytes.

[image]

SPI transfer rate is about 1.27 MB/sec  - DMA should be 5 MB/sec.

fpistm, 2023-11-21:
When I said "any contribution are welcome". I talk generally.  This do not mean you should implement the SPI DMA support. :)

fpistm, 2023-11-21:
That's the point, to be or not to be fully Arduino Compatible. :thinking: 
I will think to bring back this API including the `const` :wink: 
FYI, I will release in the coming days a 2.7.1 version with some fixes and hope including way to fix this issue (with or without fix in SdFat).

greiman, 2023-11-21:
> I will think to bring back this API including the const 

That would be great. 

Some versions of the the `transfer(txBuf, rxBuf, count)` treat a null pointer for txBuf as send 0XFF and a null pointer for rxBuf as discard receive data.

When a SPI device is full duplex like a SD card in SPI mode you can't send junk if you only want receive data.

Many board support packages now have a API like this.  `myFunction` is only used if DMA is supported.

`SPI.transfer(tx_buffer, rx_buffer, length, myFunction);`

> tx_buffer: array of Tx bytes that is filled by the user before starting the SPI transfer. If NULL, default dummy 0xFF bytes will be clocked out.
> 
> rx_buffer: array of Rx bytes that will be filled by the slave during the SPI transfer. If NULL, the received data will be discarded.
> 
> length: number of data bytes that are to be transferred
> 
> myFunction: user specified function callback to be called after completion of the SPI DMA transfer. It takes no argument and returns nothing, e.g.: void myHandler(). Can be NULL if not using a callback.

fpistm, 2023-11-23:
Hi @greiman 
could you file an issue to request to add this API, please?
Thanks in advance.

JFBow, 2023-11-24:
Good morning,
thank you very much to both of you for your discussions, I hope that this will move forward in the right direction and also a version 2.7.1 !
I lost the usb of my blackpill so I can't test your suggestions now.
I will find and fix the problem and get back to you.
THANKS.

JFBow, 2023-12-21:
Good morning,
thank you, I was able to test version 2.7.1 of stm32duino with SdFat, it works without modifying my code, as with version 2.6.0
Thanks again.