greiman/SdFat #459: Running two SdCards are not working

sufius, 2023-12-15:
I'm trying to follow the official example (v1, v2 not found) of running two cards. but  i can run always just one alone, never together. the second card initialization is always failing, why?

I'm using esp32 DevKitC v4 on PlatformIO with SdFat version 2.

```
#include <SdFat.h>

#define PIN_CS_SDCARD 13
#define SPI_SPEED_SDCARD SD_SCK_MHZ(4)

#define PIN_CS_CHIP 5
#define SPI_SPEED_CHIP SD_SCK_MHZ(20)

#define SD_CONFIG_SDCARD SdSpiConfig(PIN_CS_SDCARD, SHARED_SPI, SPI_SPEED_SDCARD)
#define SD_CONFIG_CHIP SdSpiConfig(PIN_CS_CHIP, SHARED_SPI, SPI_SPEED_CHIP)

SdFat32 SD_SDCARD;
SdFat32 SD_CHIP;

void setup()
{
  Serial.begin(115200);

  // Wait for USB Serial
  while (!Serial)
  {
    yield();
  }

  // disable sd2 while initializing sd1
  pinMode(PIN_CS_SDCARD, OUTPUT);
  digitalWrite(PIN_CS_SDCARD, HIGH);

//                     _                      
//    ___ __ _ _ __ __| |     ___  _ __   ___ 
//   / __/ _` | '__/ _` |    / _ \| '_ \ / _ \
//  | (_| (_| | | | (_| |   | (_) | | | |  __/
//   \___\__,_|_|  \__,_|    \___/|_| |_|\___|

  if (!SD_CHIP.begin(SD_CONFIG_CHIP))
  {
    Serial.println("CHIP initialization failed!");
    return;
  }
  else
  {
    Serial.println("CHIP initialization successful!");
  }

//                     _     _                 
//    ___ __ _ _ __ __| |   | |___      _____  
//   / __/ _` | '__/ _` |   | __\ \ /\ / / _ \ 
//  | (_| (_| | | | (_| |   | |_ \ V  V / (_) |
//   \___\__,_|_|  \__,_|    \__| \_/\_/ \___/ 
                                            
  if (!SD_SDCARD.begin(SD_CONFIG_SDCARD))
  {
    Serial.println("SDCARD initialization failed!");
    return;
  }
  else
  {
    Serial.println("SDCARD initialization successful!");
  }

  Serial.println("SDCARD directory:");
  Serial.println(PIN_CS_SDCARD);
  SD_SDCARD.ls(LS_R | LS_DATE | LS_SIZE);

  Serial.println("CHIP directory:");
  Serial.println(PIN_CS_CHIP);
  SD_CHIP.ls(LS_R | LS_DATE | LS_SIZE);
}

void loop()
{
  
}
```

greiman, 2023-12-15:
I copied your program and ran it on a Mega that has two cards.  

I only edited the CS pin numbers.  

I compiled and ran with two 32 GB FAT32 cards.

Here is the output:

> CHIP initialization successful!
> SDCARD initialization successful!
> SDCARD directory:
> 53
> 2023-01-01 00:00   50000000 bench.dat
> CHIP directory:
> 10
> 2023-01-01 00:00    1048576 FatLogger00.bin
> 2023-01-01 00:00    5000000 bench.dat

Can't help more since I don't have your hardware.

sufius, 2023-12-15:
okay, thank you, i will try to change my cs pin.