greiman/SdFat #462: addApbChangeCallback(): duplicate func when using with TFT_eSPI

chconnor, 2023-12-26:
Hello!

ESP32-DevKitC-VIE (WROOM-IE module) on vscode / platformIO, SdFat 2.2.2

I'm using TFT_eSPI to drive an ILI9488 display over VSPI, and using SdFat to access an SD card built in to the display.

Everything seems to work OK: screen works, SD card file writing works, etc, but when I call sd.cardBegin(SD_CONFIG) I get the following on the terminal:

`[  1397][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4016d858 arg=0x3ffc0810`

Some googling found the [same/similar error here](https://github.com/Bodmer/TFT_eSPI/issues/1509), but that thread didn't prove helpful.

Apparently TFT_eSPI uses its [own SPI instance internally](https://github.com/Bodmer/TFT_eSPI/discussions/2954#discussioncomment-7476753) -- not sure if that could be related.

I configure SdFat like this:

```
#define SD_CS_PIN 32
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16))
#define DISABLE_CS_PIN -1
```

...if I set DISABLE_CS_PIN to the CS for the TFT display (pin 5), then call pinMode and digitalWrite(DISABLE_CS_PIN, HIGH);
the problem still happens. (Admittedly I'm not clear on when I should be setting that pin high, but even if I do it right before I call sd.begin() the problem still happens.)

It should be possible to share the VSPI bus between the screen and the SD card, is that right? (Of course it's working already, but I mean without a "duplicate func" error.)

In case it's useful, TFT config:

#define USER_SETUP_ID 21
#define ILI9488_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5
#define TFT_DC    2
#define TFT_RST  -1 // using EN (aka RST), so do this
#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  16000000

I have the ESP-PROG debugger in case there is anything useful I can do with that to be of assistance.

Thanks for any ideas! And thanks for the library.

greiman, 2023-12-27:
> Apparently TFT_eSPI uses its [own SPI instance internally](https://github.com/Bodmer/TFT_eSPI/discussions/2954#discussioncomment-7476753)  -- not sure if that could be related.

This is likely the problem.  SdFat depends on  beginTransaction/endTransaction for the standard instance of SPI to control sharing.

There is a way to try using the internal instance.  There is an additional parameter for SdSpiConfig.  It may not work depending on the nature of the SPI instance.

If the instance is TftSpi do this:

`#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16) , &TftSpi)`

If you only have a pointer to the instance use:

`#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16) , pTftSpi)`

chconnor, 2023-12-27:
Fantastic, thanks! That has removed the error line. I was also having issues with the SD not working sometimes and that seems to also have been fixed (crossing my fingers.) Thanks for the help!

closed 2023-12-27 by chconnor

Thecactusman0, 2024-01-24:
@chconnor Would you mind posting your fully working code? Or atleast the initialisation? Ive been having a lot of trouble getting tft espi to play nicely with sdfat and this almost exactly my problem. How did you get the instance from tft_espi in the SdSpiConfig?

chconnor, 2024-01-24:
Hi -- sure --

TFT_eSPI is initialized as above, with the exception that SPI_READ_FREQUENCY is now:
```#define SPI_READ_FREQUENCY  10000000```
(but I'm guessing that doesn't matter and that the SD library sets the bus to it's frequency as needed? Or maybe this actually mattered... I don't remember...)

SD initialization looks like something like this:

```
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(10), &tftspi)
#define SD_CS_PIN 0 // this is my pin for CS for the SD card; connects to my display module

  SPIClass tftspi = tft.getSPIinstance();
  if (!sd.cardBegin(SD_CONFIG))
  {
    // SD card initialization failed
    if (isSpi(SD_CONFIG)) // copy/pasted code; this is likely redundant?
    {
      // Is SD_CS_PIN set to the correct value? Does another SPI device need to be disabled?
    }
    printSDError(); // custom error printing function
    return(false);
  }
```

Thecactusman0, 2024-01-25:
Alright ive now got tft espi and sdfat working together with no issues so far. Here is the full example sketch I used for anyone who needs it. My SPI_DRIVER_SELECT is set to 0 in the config file. Thanks chconnor that was the last piece to getting it working.

`#include <SPI.h> //spi library
#include <TFT_eSPI.h>  // tft lcd library
#include "SdFat.h" //sd card library

//Sd card pins
#define SDCSPin 5 

TFT_eSPI tft = TFT_eSPI();               // Invoke custom library
//SoftSpiDriver<misoPin,mosiPin, sckPin> softSpi;

SdFs sd;
SdFile file;
SdFile root;

void setup() {
  Serial.begin(115200);

  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLUE);

  SPIClass tftspi = tft.getSPIinstance();
  if (!sd.begin(SdSpiConfig(SDCSPin, SHARED_SPI, SD_SCK_MHZ(10), &tftspi))) {
    Serial.println("SD start failed");
  }

  tft.fillScreen(TFT_GREEN);
  
  delay(100);
  readTest();

  tft.setCursor(10,10);
  tft.print("Testing");
}

void loop() {}

void readTest()
{
  root.open("/", O_RDONLY);
  int i = 0;
  while (file.openNext(&root, O_RDONLY)) 
  {  
    char buffer[255];
    file.getName(buffer, 255);
    // Skip directories and hidden files.
    if (!file.isSubDir() && !file.isHidden()) 
    {
      Serial.println(buffer);
      i++;
    }
    file.close();
  }
}

`