greiman/SdFat #464: [question]How to speed up sd writes on Arduino Uno r4?

7-rate, 2024-01-13:
## How to speed up sd writes on Arduino Uno r4?

I am building a robot and want to log about 64 bytes of information per ms to a csv file.
As a test, if I run sync() every 512 bytes on R4, it takes about 10ms.
Is there any way to reduce sync() to less than 1ms?

7-rate, 2024-01-13:
The following code confirms this.
https://github.com/7-rate/arduinoR4_SD_bench/tree/main

I expected that it would be faster if I wrote while switching the use of the buffer for writing to sd and the buffer for logging, but I was wrong.
It seems that FreeRTOS does not switch tasks while sync() is running.

greiman, 2024-01-13:
> As a test, if I run sync() every 512 bytes on R4, it takes about 10ms.
> Is there any way to reduce sync() to less than 1ms?

Calling sync() kills performance.  It caused a huge amount of I/O.  It's almost as much I/O as closing and opening the file.

[This](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioLogger/TeensySdioLogger.ino) example can log data fast.

[Here](https://github.com/greiman/SdFat/issues/456#issuecomment-1843003087) is a version that uses SPI on a STM32.  You will go much slower on Uno R4.  I hope you are not using shared SPI.  SD cards go extremely slow in shared mode. 

You may get an improvement in SPI performance by editing SdFatConfig.h and setting these defines.

At about line 122:
`#define SPI_DRIVER_SELECT 1   //  was zero `

At about line 144
`#define USE_SPI_ARRAY_TRANSFER 1  // was zero`

7-rate, 2024-01-14:
Thanks for letting me know.  
If possible, I would like to do this via SPI, not SDIO.

>> At about line 122:.
>> #define SPI_DRIVER_SELECT 1 // was zero 
>> 
>> At about line 144
>> #define USE_SPI_ARRAY_TRANSFER 1 // was zero

I tried the above, but the results were not faster.

I am trying to use a FreeRTOS task to trigger a task that buffers to RAM and flushes to SD card when the buffer is full. 
https://github.com/7-rate/arduinoR4_SD_bench/blob/main/src/sdfat_wrapper.cpp#L42
Aiming at logging every 1ms by buffering while flushing.

However, when I do a sync() on the task that flushes, the task does not switch tasks and it ends up not being faster.

This may be the wrong place to ask this, but do you know why FreeRTOS does not switch tasks while sync() is running?

greiman, 2024-01-14:
> I am trying to use a FreeRTOS task to trigger a task that buffers to RAM and flushes to SD card when the buffer is full.

Don't call sync(). It kills performance look at the examples for fast logging.

Throw away your code that will never work!

Put data you want to log in a large circular buffer.  It must hold at least 200-400 ms of data since an SD can go busy for that long.

Write a low priority task to write data to the SD.

You must preallocate a file since searching for free clusters can take seconds.  See the example I pointed out above.

[Here](https://github.com/greiman/FreeRTOS-Arduino/blob/master/libraries/FreeRTOS_ARM/examples/frFifoDataLogger/frFifoDataLogger.ino) is a really old example using FreeRTOS.  This example was written before I added preallocate.

greiman, 2024-01-14:
> I tried the above, but the results were not faster.

I checked using the [bench](https://github.com/greiman/SdFat/blob/master/examples/bench/bench.ino) example.

Here is the default config with an Uno R4:

> Type any character to start
> FreeStack: 25564
> Type is FAT32
> Card size: 32.01 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0X390D14D2
> Manufacturing date: 6/2015
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 612.44,852,830,833
> 612.29,852,830,833
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 569.93,897,893,895
> 569.93,897,893,895

Here is the result with Array Transfer:

> Type any character to start
> FreeStack: 25564
> Type is FAT32
> Card size: 32.01 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0X390D14D2
> Manufacturing date: 6/2015
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1695.49,319,298,299
> 1696.07,317,298,299
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1841.62,279,274,275
> 1841.62,279,274,275

Write is about 2.7 times faster.  Max write latency goes from 879 μs to 319 μs.

greiman, 2024-01-14:
Here is proof that Uno R4 can log a 64 byte CSV string every ms.  There is a bit of jitter in the log time.  Must be an interrupt that is slow in the R4,

I took the STM32 SPI example and edited it for CSV.
```
// Test Uno R4 with write busy and RingBuf in a data logger demo.

#include "RingBuf.h"
#include "SdFat.h"

const uint8_t SD_CS_PIN = SS;
#define SPI_CLOCK SD_SCK_MHZ(50)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)

// Interval between points for 1,000 sps.
#define LOG_INTERVAL_USEC 1000

// Size to log 64 byte lines at 1,000 sps for more ten minutes.
#define LOG_FILE_SIZE 64 *1000 * 600 

// Space to hold more than 200 ms of data for 64 byte lines at 1,000 sps.
#define RING_BUF_CAPACITY 64 * 200

#define LOG_FILENAME "UnoR4Logger.csv"

SdFs sd;
FsFile file;

// RingBuf for File type FsFile.
RingBuf<FsFile, RING_BUF_CAPACITY> rb;

void logData() {
  // Initialize the SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  // Open or create file - truncate existing file.
  if (!file.open(LOG_FILENAME, O_RDWR | O_CREAT | O_TRUNC)) {
    Serial.println("open failed\n");
    return;
  }
  // File must be pre-allocated to avoid huge
  // delays searching for free clusters.
  if (!file.preAllocate(LOG_FILE_SIZE)) {
    Serial.println("preAllocate failed\n");
    file.close();
    return;
  }
  // initialize the RingBuf.
  rb.begin(&file);
  Serial.println("Type any character to stop");

  // Max RingBuf used bytes. Useful to understand RingBuf overrun.
  size_t maxUsed = 0;

  uint32_t maxJitterMicros = 0;

  // Min spare micros in loop.
  int32_t minSpareMicros = INT32_MAX;

  // Actual log time micros
  uint32_t m;

  // Start time.
  uint32_t logTime = micros();

  // Log data until Serial input or file full.
  while (!Serial.available()) {
    // Amount of data in ringBuf.
    size_t n = rb.bytesUsed();
    if ((n + file.curPosition()) > (LOG_FILE_SIZE - 20)) {
      Serial.println("File full - quitting.");
      break;
    }
    if (n > maxUsed) {
      maxUsed = n;
    }
    // Not busy only allows one sector before possible busy wait.
    // Write one sector from RingBuf to file.
    if (n >= 512 && !file.isBusy()) {
      // This is the limiting factor for sample rate.
      // The STM32F411 SPI library is slow so this takes around 350 usec.
      if (512 != rb.writeOut(512)) {
        Serial.println("writeOut failed");
        break;
      }
    }
    // Time for next point.
    logTime += LOG_INTERVAL_USEC;
    int32_t spareMicros = logTime - micros();
    if (spareMicros < minSpareMicros) {
      minSpareMicros = spareMicros;
    }
    if (spareMicros <= 0) {
      Serial.print("Rate too fast ");
      Serial.println(spareMicros);
      break;
    }
    // Wait until time to log data.
    // This loop is so fast that jitter is not over one usec.
    do {
      m = micros();
    } while (m < logTime);
    const char* csv = "1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ\n";
    rb.print(csv);
    if (rb.getWriteError()) {
      // Error caused by too few free bytes in RingBuf.
      Serial.println("WriteError");
      break;
    }
    m -= logTime;
    if (m > maxJitterMicros) {
      maxJitterMicros = m;
    }
  }
  // Write any RingBuf data to file.
  rb.sync();
  file.truncate();
  file.rewind();
  // Print first twenty lines of file.
  for (uint8_t n = 0; n < 20 && file.available();) {
    int c = file.read();
    if (c < 0) {
      break;
    }
    Serial.write(c);
    if (c == '\n') n++;
  }
  Serial.print("fileSize: ");
  Serial.println((uint32_t)file.fileSize());
  Serial.print("maxBytesUsed: ");
  Serial.println(maxUsed);
  Serial.print("minSpareMicros: ");
  Serial.println(minSpareMicros);
  Serial.print("maxJitterMicros: ");
  Serial.println(maxJitterMicros);
  file.close();
  sd.end();
}
void clearSerialInput() {
  for (uint32_t m = micros(); micros() - m < 10000;) {
    if (Serial.read() >= 0) {
      m = micros();
    }
  }
}
void setup() {
  Serial.begin(9600);
  while (!Serial) {
  }
}

void loop() {
  clearSerialInput();
  Serial.println("Type any character to start");
  while (!Serial.available()) {
  }
  clearSerialInput();
  logData();
}
```
Here is the output:

> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> 1234567890,abcdefghijklmnopqrstuvwxyz,ABCDEFGHIJKLMNOPQRSTUVWXYZ
> fileSize: 1560390
> maxBytesUsed: 706
> minSpareMicros: 640
> maxJitterMicros: 39

7-rate, 2024-01-15:
Thank you very much!
I got the results I was hoping for.

However, what is a little disturbing is that it did not work with the following settings.
```#define SPI_CLOCK SD_SCK_MHZ(50)```

In my environment (Arduino Uno R4), I had to drop down to 22MHz to get it to work.

closed 2024-01-15 by 7-rate