greiman/SdFat #465: Examples of SDFat

VantagePro2, 2024-01-13:
Good evening!
I am dealing with the examples of the SDFat library (version 2.2.2).
When compiling the Unicode Filenames example for the Arduino Due board, I get an error:

Arduino: 1.8.19 (Windows 10), Плата:"Arduino Due (Programming Port)"

UnicodeFilenames:96:2: error: #error USE_UTF8_LONG_NAMES must be non-zero in SdFat/src/SdFatCongfig.h
 #error USE_UTF8_LONG_NAMES must be non-zero in SdFat/src/SdFatCongfig.h
 ^
exit status 1
#error USE_UTF8_LONG_NAMES must be non-zero in SdFat/src/SdFatCongfig.h

What is the reason for this error?
There is 
// #define USE_UTF8_LONG_NAMES 1
in SdFatCongfig.h

greiman, 2024-01-13:
Edit SdFatConfig.h and remove the comment slashes so `USE_UTF8_LONG_NAMES` is define to one.

VantagePro2, 2024-01-14:
Everything works.
Thank you very much.
Could you explain to me why there were the comment slashes for this line?

VantagePro2, 2024-01-14:
Now it's clear
Thanks a lot