greiman/SdFat #469: Did not work at ESP32-CAM

JoergTiedemann, 2024-02-17:
Hi I'm using latest version on a ESP-32-CAM board.
But i didn't got it to run
I use the followingfor initialization:
`
  if (!SD.begin(SS, SD_SCK_MHZ(10))) {
    Serial.println("SD Card Mount Failed");
    return;
  }`
but everything was i got was 
SD-card mount failed.
SO what is the right configuration for an ESP32-CAM board ?

greiman, 2024-02-17:
What happens if you run the [QuickStart](https://github.com/greiman/SdFat/blob/master/examples/QuickStart/QuickStart.ino) example?

JoergTiedemann, 2024-02-18:
did not work:
I got Standard SD running with this parameters:

`#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI(VSPI);
sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
if(!SD.begin(SD_CS, sdSPI)){ 
...`

so in my ESP32-CAM board the CS Pin is definitly GPIO 13
if I enter 13 in the serial Monitor of quickstart example i got the following message:
`13

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?

errorCode: 0x1, errorData: 0x0

Restarting

Enter the chip select pin number: `
because of i got standard SD library running i suppose that somethin with SPI Interface to access SDFat is going wrong
problably you can help me or give me some tips how to get the example working on ESP32-CAM

greiman, 2024-02-18:
Try the following.

```
#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI(VSPI);
#define #define SD_CONFIG SdSpiConfig(SD_CS, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(10), &sdSPI)

...

  sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!sdSPI.begin(SD_CONFIG)) { Serial.println("SD Card Mount Failed"); return; }

```

I don't have your ESP32 board so I can't test it.

greiman, 2024-02-18:
I did compile the following.

```
#include "SdFat.h"
SdFat SD;

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI(VSPI);
#define SD_CONFIG SdSpiConfig(SD_CS, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(10), &sdSPI)

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CONFIG)) { Serial.println("SD Card Mount Failed"); return; }
}
void loop() {
}
```

JoergTiedemann, 2024-02-18:
this works,
thank you for support :-)
you should at minimum add an addidional example in your repo please
found many posts on google about people with same problem

greiman, 2024-02-18:
> you should at minimum add an addidional example in your repo please

There is a [example](https://github.com/greiman/SdFat/blob/master/examples/UserSPIDriver/UserSPIDriver.ino) of more general way to use non standard SPI.

The arguments for SdSpiConfig are defined in the html documentation.

[image]

The problem is that there are now hundreds arduino compatible boards.   I can't really even test on very many.

See [this](https://en.wikipedia.org/wiki/List_of_Arduino_boards_and_compatible_systems).