greiman/SdFat #470: MMC_ERASE

folkertvanheusden, 2024-02-18:
Hi,

Are you planning to implement MMC_ERASE? It's like trim/unmap for SD cards, see https://superuser.com/questions/18906/can-i-use-something-like-the-ssd-trim-command-on-a-sd-card

greiman, 2024-02-18:
Look at the SdFormatter example [here](https://github.com/greiman/SdFat/blob/57900b21d21655c513ef7e344f55e8190e612dcb/examples/SdFormatter/SdFormatter.ino#L95C1-L95C6).

This does a very fast erase of all data on the card.

I plan to make an SD erase function callable from an app.

I have a large number of additions and additions for the next version of SdFat but I currently don't have time to work on my open source libraries.  

A new release take a lot of time since I must test it on a substantial number of cards using the most popular board support packages.

folkertvanheusden, 2024-02-18:
Looking forward to it!