greiman/SdFat #471: Wrapper to fs::FS  to use SDfat in ESPAsyncWebserver

JoergTiedemann, 2024-02-19:
Is there any wrapper to use Sdfat with a ESPAsyncWebserver to convert sdfat to to fs::FS  because of ESPAsyncWebserver needs in AsyncWebServerRequest::send a FS object
or how can I do that ?

greiman, 2024-02-19:
I don' know of any use of SdFat with ESPAsyncWebserver.

I avoid using ESP32/ESP8266 because the SdFat often conflicts with the ESP file system.

JoergTiedemann, 2024-02-19:
I found a wrapper, will post it next weekend 
Investors to use the lib on Esp32 because of firebase library by @mobizt requires it because of your library has adjustable speed for spi interface

VeloSteve, 2024-03-14:
Any chance you can post that wrapper?  SD.h is unreliable for me (on Arduino Nano ESP32), but SdFat is stable, so I need to use SdFat.

JoergTiedemann, 2024-03-14:
Hi gimme a chance this weekend
I will try to search the wrapper and post it, if you didn't got any post from me remember me latest this Sunday

JoergTiedemann, 2024-03-16:
[sd_fat32_fs_wrapper.zip](https://github.com/greiman/SdFat/files/14623380/sd_fat32_fs_wrapper.zip)
In the attachment you will find the wrapper
just include the file

I use it als following

`SdFat sd_fat_fs;
`
`fs::FS sdfilesys = fs::FS(fs::FSImplPtr(new SdFat32FSImpl(sd_fat_fs)));
`

If you want to use it with ESP Async Webserver in a response at server.on...
i.g. to respond a picture from an sd card 
`String picturefilene ="picture.jpg"
`
`request->send(sdfilesys,>picturefilename>, "image/jpg", false);
`
hope this helps

VeloSteve, 2024-03-21:
Thank you! I missed your post for a few days and came up with the workaround of replacing send(SD, filename . . . ) calls with  sendChunked and my own simple function which gets data from the file using SdFat.  Your wrapper is probably nicer, so I'll try it out.