greiman/SdFat #472: pico-sdk only rp2040 implementation

bog-dan-ro, 2024-02-22:
Hello there,

I'd like to add support for rp2040 but based only on pico-sdk, I'd love to upstream it, so I'd like to ask if such a contribution is something that the maintainer(s) will accept.

Why pico-sdk only? Mostly because the arduino-pico can't be used with cmake projects .

greiman, 2024-02-22:
I have no interest in supporting SdFat on new platforms.  I will not being doing much much new development of SdFat.

A file system should be integrated into the operating system not a library like SdFat.  When I started SdFat in 2008 for an Arduino with 1 KB of RAM, I assumed it would be replaced with a proper filesystem.

The pico-sdk should have an integrated file system with SD being one of the file devices.

It would be fairly easy to port SdFat to pico-sdk.  I use SdFat on the ChibiOS/RT RTOS.  You need to replace a couple of calls to millis(), digitalWrite(), and pinMode().

You need a wrapper for SPI like in [this](https://github.com/greiman/SdFat/blob/master/examples/UserSPIDriver/UserSPIDriver.ino) example.

I have a very fast pio SDIO driver for rp2040 that can do over 20 MB/sec.  I will post it as an example eventually.

bog-dan-ro, 2024-02-22:
Thanks a lot for your quick reply!

>  have no interest in supporting SdFat on new platforms. I will not being doing much much new development of SdFat.

In this case I'll not waste time to try to isolate the Arduino stuff and I'll just remove it or replace it with pico-sdk.

> The pico-sdk should have an integrated file system with SD being one of the file devices.

It should, but ... Right now I'm using https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico but is buggy.

Thanks a lot for the link. Regarding PIO SDIO implementation, how big is it? I'm asking as I might not be able to use it as I'm a bit short on PIO space (on both of PIOs :) ).

bog-dan-ro, 2024-03-10:
@greiman did you manage to post the pio SDIO driver? 
Thanks in advance :) !

greiman, 2024-03-10:
I looked at the SDIO driver but became busy with other things.

You probably can't use it since it uses almost all memory in one PIO. I implemented it to be fast for small transfers like the SdFat dedicated SPI mode but that takes more memory.

I also need to decide whether to do more development.  There are lots of possible configuration options so I need to make it easy to choose the pins and speed options.  This can't be done at run time since it must be compiled into the PIO code.

I also need to decide whether to develop a second mode.  SD cards start in the very old "Standard mode" which was limited to 25 MHz.  Modern cards can run much faster in that mode.  I find all cards I have tested can run at 62.6 MHz with 250 MHz CPU speed.

Implementing "High Speed" mode is a big job since it has totally different bus timing but the SD start in "Standard" mode. So a reload of the PIO code is required during the initialization process.  25 years ago 50 MHz bus speed required an unusual timing.

I also don't want to increase the amount time I spend on maintenance of SdFat.  I don't enjoy development on the Arduino platform.  I had high hope for Arduino15 years ago when I released the first version of SdFat.  I now don't use Adruino for my projects.

bog-dan-ro, 2024-03-11:
First and foremost, I'd like to say thanks a lot Bill for all your effort on this project. As a FOSS contributor myself, I know it is hard to maintain a project that is used by so many people even if we like it or not ;-).

I managed to free one of the PIOs, next I'll try to see if I can adapt [ZuluSCSI's SDIO](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/ZuluSCSI_platform_RP2040/sdio_RP2040.pio) implementation. Is your implementation much different?

greiman, 2024-03-11:
>  Is your implementation much different?

My implementation is very different.  I started from scratch, I didn't copy any existing code.

Modern  SD cards require vary large transfers for maximum speed.  I do this for both read and write by keeping the SD in read or write mode as long as possible.  To do this I stop the SDIO clock between 512 byte transfers.

Here is a typical result of the [bench](https://github.com/greiman/SdFat/blob/master/examples/bench/bench.ino) example for 512 byte reads and writes.  This is overclocked at 250 MHz CPU.

This is on an [Adafruit Metro RP2040](https://www.adafruit.com/product/5786).  I also test on a Pico.

> Type is exFAT
> Card size: 128.18 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: YD4QD
> Revision: 3.0
> Serial number: 0XCBDD66A0
> Manufacturing date: 9/2023
> 
> FILE_SIZE_MB = 50
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 24366.41,48,19,20
> 24366.41,37,19,20
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 24402.08,77,20,20
>24402.08,169,20,20

greiman, 2024-03-11:
I looked at the current state of my RP2040 SDIO.  The source needs to be cleaned up since I was still playing with different approaches. 

I looked at the High Speed mode again. Card output valid timing has a long delay from the rising edge of clock.  I might try it again at some point.

Here is High Speed mode timing:

[image]

androda, 2024-08-02:
> You probably can't use it since it uses almost all memory in one PIO. I implemented it to be fast for small transfers like the SdFat dedicated SPI mode but that takes more memory.

Existing RP2040 SDIO systems like the aforementioned https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico already use most (or all) of the instruction memory in one of the PIO units.  It's not a downside for yours to use most of the instruction memory, rather it seems to be the status quo.  :)

greiman, 2024-08-03:
@androda

I had hoped to post a version of SdFat with RP2040 SDIO some time ago but have been very busy and away from home so not had time to finish testing all the mods in the new version.

androda, 2024-08-15:
@greiman 
If you'd like another set of eyes on it, let me know.  I have an Agilent logic analyzer frame with 16717A card which ought to be more than up to the task of checking SD waveforms.

greiman, 2024-08-15:
Here is a test version of SdFat with RP2040 SDIO.  Try it and let me know the result.

[SdFatRp2040.zip](https://github.com/user-attachments/files/16626181/SdFatRp2040.zip)

It is setup with two examples, Rp2040SdioSetup and bench.  I test with an Adafruit RP2040 Metro and a Pi Pico.

I use the AdaFruit [PiCowBell](https://www.adafruit.com/product/5200) on Pico.  The blue proto board has pins for a logic analyzer.

[image]

I tested at 133 MHz and 250 MHz.  Here is my results for the bench example:

At 133MHz:

> Type is exFAT
> Card size: 128.18 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: YD4QD
> Revision: 3.0
> Serial number: 0XF024666C
> Manufacturing date: 7/2023
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 12819.69,2005,38,38
> 12657.42,2414,38,38
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 12593.65,88,39,39
> 12562.01,83,39,39

At 250 MHz:

> Type is exFAT
> Card size: 128.18 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: YD4QD
> Revision: 3.0
> Serial number: 0XF024666C
> Manufacturing date: 7/2023
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 23472.68,1181,20,21
> 23472.68,1220,20,21
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 23808.00,45,20,21
> 23808.00,49,20,21

androda, 2024-08-19:
@greiman 
Yes, pre-ordered some of the Pico 2.  I'm also excited about the RP2350B, given the additional IO pins.  We should see some really interesting projects with this new chip.

androda, 2024-08-19:
Got things set up this morning for building the RP2040 library, here are the first results (no clock speed changes yet, just compiled as-is and run).
```Type any character to start
FreeStack: 253848
Type is exFAT
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: WX32G
Revision: 8.0
Serial number: 0X830F12B8
Manufacturing date: 3/2023

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
12406.15,3134,37,39
12164.67,6934,38,39

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
12593.65,84,39,39
12562.01,68,39,39

Done
Type any character to start

greiman, 2024-08-19:
Write is often slower for big writes since there will be occasional large latencies while the card programs flash.

Your card has max latencies of about 50 ms.  For 133 MHz the average latency was 3-4 ms greater than the min latency.

For the card I used with a 500 MB file, the max latency is 8-10 ms and average latency is one ms greater than the min.

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 12961.08,10603,37,38
> 12952.69,8179,37,38

For some apps occasion large write latency is very undesirable. 

Here is my card for 500 MB at 250 MHz.  Max of 8 ms write latency. 

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 24273.01,8285,20,20
> 24271.83,8275,20,20
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 23776.68,85,20,21
> 23774.42,197,20,21

greiman, 2024-09-10:
I did first tests with a Pico 2.  I need to do some mods so the third PIO can be used and there are problems with some cards at 250 MHz.

Here are two tests with a Lexar Silver Plus at 250 MHz.

512 byte buffer with a 5 MB file:
> FreeStack: 515984
> Type is exFAT
> Card size: 125.67 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0XAD
> OEM ID: LS
> Product: LX128
> Revision: 2.1
> Serial number: 0X7009FC3A
> Manufacturing date: 4/2024
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 26453.33,31,19,19
> 26314.11,22,19,19
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 26314.11,243,19,19
> 26314.11,216,19,19

32,768 byte buffer with a 50 MB file.

> FreeStack: 483216
> Type is exFAT
> Card size: 125.67 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0XAD
> OEM ID: LS
> Product: LX128
> Revision: 2.1
> Serial number: 0X7009FC3A
> Manufacturing date: 4/2024
> 
> FILE_SIZE_MB = 50
> BUF_SIZE = 32768 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 27963.74,1179,1170,1171
> 27792.66,11351,1170,1177
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 27381.48,1208,1194,1196
> 27396.49,1198,1194,1196

DatanoiseTV, 2024-09-18:
> I did first tests with a Pico 2. I need to do some mods so the third PIO can be used and there are problems with some cards at 250 MHz.
>
> Here are two tests with a Lexar Silver Plus at 250 MHz.
>
> 512 [...]

Does it mean it would use all 3 PIOs or 3rd PIO? I am using 1 PIO for I2S Audio on the [PicoADK v2](https://datanoise.shop/products/picoadk-v2), so it would be good to have 1-2 spare PIOs.

greiman, 2024-09-19:
> Does it mean it would use all 3 PIOs or 3rd PIO?

It uses one PIO.  It looks for a PIO with almost all memory available.  Then it claims two state machines.

See `static bool pioInit()`  at line 260 of SdFat/src/SdCard/Rp2040Sdio/PioSdioCard.cpp.