greiman/SdFat #473: ESP32 S3 always fails to mount card

derekriter, 2024-02-22:
Hello all!
I have been working with an Adafruit ESP32-S3 Feather (4MB Flash, 2MB PSRAM) using Arduino code and PlatformIO. The SD card + reader I have attached work flawlessly when attached to an Arduino Uno, but always fail with error codes 0x17, 0xFF on an ESP. I have spent a lot of time looking through other forums and issues on this repo and none of the solutions seem to work for me.

I understand that greiman doesn't plan to support ESP boards, but if somebody could explain to me why it doesn't work, then maybe I could try to patch the library for my project.

Any help is greatly appreciated.

greiman, 2024-02-23:
0x17, 0xFF is the basic initialization command which occurs at very low SPI clock.  Usually means an SPI problem.

Is the SD connected to the correct pins?

Are you trying to map the SPI pins?

derekriter, 2024-02-23:
Hi, thanks for your help. I believe the SPI interface is connected correctly, as if I change it at all then I get 0x01, with the SPI failing completely. I am using SPI.begin to set the SPI bus to the pins I am using. My code roughly goes as follows

```
#define SD_FAT_TYPE 0
SdFat sd;

void setup() {
    SPI.begin(SD_SCK, SD_MISO, SD_MOSI, -1);
    if(!sd.begin(SD_CS)) {
        sd.initErrorHalt();
    }
}
```

derekriter, 2024-02-23:
Ok, I will try that later today when I am able.

derekriter, 2024-02-23:
It didn't work. I am still getting 0x17. My exact code is
```
#include <Arduino.h>
#include <SPI.h>

#include "SdFat.h"

#define SD_FAT_TYPE 0

const uint8_t SD_CS_PIN = 6;
const uint8_t SD_MISO_PIN = 10;
const uint8_t SD_MOSI_PIN = 9;
const uint8_t SD_SCK_PIN = 11;

#define SD_CONFIG SdSpiConfig(SD_CS_PIN, USER_SPI_BEGIN | DEDICATED_SPI, SD_SCK_MHZ(10))

#if SD_FAT_TYPE == 0
SdFat sd;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
#elif SD_FAT_TYPE == 2
SdExFat sd;
#elif SD_FAT_TYPE == 3
SdFs sd;
#else
#error Invalid SD_FAT_TYPE
#endif

void setup() {
    Serial.begin(115200);
    
    Serial.println("Type any character to start");
    while(!Serial.available()) {
        yield();
    }
    
    SPI.begin(SD_SCK_PIN, SD_MISO_PIN, SD_MOSI_PIN, -1);
    if(!sd.begin(SD_CONFIG)) {
        sd.initErrorHalt();
    }
}
void loop() {
    
}
```
And console output
```
Type any character to start
begin() failed
Do not reformat the SD.
SdError: 0X17,0XFF
```

I also tried moving the ```#define SD_FAT_TYPE 0``` into the beginning of SdFat.h, and it had no effect, so I believe that it takes global effect.

greiman, 2024-02-23:
I can't do more since I don't have your hardware.

[Here](https://github.com/greiman/SdFat/issues/469#issuecomment-1951343756) is a similar ESP32 issue where the above worked.

derekriter, 2024-02-23:
OK, thank you for your efforts. If I can't get it to work, I might just have to bit-bang it.

closed 2024-02-23 by derekriter

greiman, 2024-02-24:
You need to verify the SPI is available on the pins you selected.   Here is a pinout for Feather ESP32-S3:
[image]

Looks like the option is:
MOSI  pin 11
MISO  pin 13
SCK    pin  12

Standard pins are:

SCK  pin 36
MOSI pin 35
MISO pin 37

greiman, 2024-02-24:
Here is the IO MUX pin functions from the ESP32-S3 datasheet:
[image]

greiman, 2024-02-24:
I found a ESP32-S2 Feather someone gave me.  It has the same pinout as the ESP32-S3 Feather.

I edited your sketch to use the correct pins and added a call to ls().

```
#include <Arduino.h>
#include <SPI.h>

#include "SdFat.h"

#define SD_FAT_TYPE 0

const uint8_t SD_CS_PIN = 10;
const uint8_t SD_MISO_PIN = 13;
const uint8_t SD_MOSI_PIN = 11;
const uint8_t SD_SCK_PIN = 12;

#define SD_CONFIG SdSpiConfig(SD_CS_PIN, USER_SPI_BEGIN | DEDICATED_SPI, SD_SCK_MHZ(10))

#if SD_FAT_TYPE == 0
SdFat sd;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
#elif SD_FAT_TYPE == 2
SdExFat sd;
#elif SD_FAT_TYPE == 3
SdFs sd;
#else
#error Invalid SD_FAT_TYPE
#endif

void setup() {
    Serial.begin(115200);
    
    Serial.println("Type any character to start");
    while(!Serial.available()) {
        yield();
    }
    
    SPI.begin(SD_SCK_PIN, SD_MISO_PIN, SD_MOSI_PIN, -1);
    if(!sd.begin(SD_CONFIG)) {
        sd.initErrorHalt();
    }
    sd.ls();
}
void loop() {}
```
Here is the output:

> bench.dat
> SF411Logger.csv
> STM32F411Logger.csv
> STM32F411ADC/
> STM32F411Logger/