greiman/SdFat #474: Unable to format flat Spiflash module with ESP8266

Mikey31200, 2024-02-25:
Hello,
I want to use an Spiflash module W25Q128FV with ESP8266 module.
Unfortunately, I cannot be able to format the memory with the sdFormatter sketch.

The connections seem to be correct as soon as I can execute properly the RawHardwareTest sketch from this library:
https://github.com/PaulStoffregen/SerialFlash/tree/master/examples/RawHardwareTest/RawHardwareTest.ino

Raw SerialFlash Hardware Test

Read Chip Identification:
  JEDEC ID:     EF 40 18
  Part Nummber: W25Q128FV
  Memory Size:  16777216 bytes
  Block Size:   65536 bytes

Reading Chip...

Writing 32 signatures

Double Checking All Signatures:
  all 8192 signatures read ok

Checking Signature Pairs
  all 4095 signature pairs read ok

Checking Read-While-Write (Program Suspend)
  write 256 bytes at 512
  write time was 398 microseconds.
  read-while-writing: 00 00 00 00 15 F5 95 4B 
  test passed, good read while writing

But when I try to execute sdFormatter, I get the following error message :

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 15
Edit SD_CS_PIN to change the SD chip select pin.

Type any character to start

This program can erase and/or format SD/SDHC/SDXC cards.

Erase uses the card's fast flash erase command.
Flash erase sets all data to 0X00 for most cards
and 0XFF for a few vendor's cards.

Cards up to 2 GiB (GiB = 2^30 bytes) will be formated FAT16.
Cards larger than 2 GiB and up to 32 GiB will be formatted
FAT32. Cards larger than 32 GiB will be formatted exFAT.

Warning, all data on the card will be erased.
Enter 'Y' to continue: Y
error: card init failed.
No card, wrong chip select pin, or wiring error?
SD errorCode: SD_CARD_ERROR_CMD0 = 0x1
SD errorData = 0xff

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Could you help me to solve my problem, please ?

Thank you.

greiman, 2024-02-25:
SdFat will not support your SPI Flash module.

SdFat only supports SD cards that conform to the [SD standard](https://www.sdcard.org/downloads/pls/) command set.

Mikey31200, 2024-02-26:
Hi, 
Thank you for your quick response. Do you know if an other library can be used for this purpose (format SPI Flash module in Fat) ?

greiman, 2024-02-26:
I think Adafruit has added a SPI flash driver to their port of SdFat.

There are other SPI flash file systems but usually FAT is not considered appropriate for SPI flash.

FAT assumes wear leveling is handled at the drive level.  SPI flash expects wear leveling to be handled by the file system.

Here are tow popular SPI flash file systems

https://github.com/littlefs-project/littlefs

https://github.com/pellepl/spiffs

Mikey31200, 2024-03-16:
Thank you for you support

closed 2024-03-16 by Mikey31200