greiman/SdFat #477: Unable to share the same SPI bus with TFT_eSPI on ESP32

JerryZRF, 2024-03-24:
I am using a LCD screen which shares its SPI bus with a SD reader on ESP32.
```c
// User_Setup.h in TFT_eSPI
#define ST7796_DRIVER

#define SPI_FREQUENCY  40000000

#define TFT_MISO  25
#define TFT_MOSI  27
#define TFT_SCLK  26
#define TFT_CS    13  // Chip select control pin
#define TFT_DC    14  // Data Command control pin
#define TFT_RST   12  // Reset pin (could connect to RST pin)
```
```c
//SdFatConfig.h in SdFat
#define SDFAT_FILE_TYPE 3
#define ENABLE_DEDICATED_SPI 0
#define SPI_DRIVER_SELECT 2
```
```c++
// My code
const uint8_t SD_CS = 33;
SoftSpiDriver<TFT_MISO, TFT_MOSI, TFT_SCLK> softSpi;
#define SD_CONFIG SdSpiConfig(SD_CS, SHARED_SPI, SD_SCK_MHZ(40), &softSpi)
```
When I try to init the SdFat and the the TFT_eSPI, the SdFat doesn't work.
```c++
sd.begin(SD_CONFIG); // It succeeds.

tft.begin();
tft.setRotation(1);
tft.fillScreen(TFT_BLUE);

uint32_t size = sd.card()->sectorCount();  // It returns 0.
```
And when I try to init the TFT_eSPI and the the SdFat, the TFT_eSPI doesn't work.
```c++
tft.begin();
tft.setRotation(1);
tft.fillScreen(TFT_BLUE);

sd.begin(SD_CONFIG) // It succeeds.

uint32_t size = sd.card()->sectorCount();  // It returns the right value.
tft.println("Succeed");  // Nothing happends
Serial.println("Succeed");  // It succeeds.
```

greiman, 2024-03-24:
Are you pulling chip select high on the second device to to be initialized so it won't interfere with initializing of the first device?

sud-sharma, 2024-08-09:
I am also facing the same issue. Did you find a solution ?

Andy2No, 2024-08-09:
Not all SD breakouts are the same.  The ones that use a logic level shifter to work with 5V Arduinos tend not to work well with 3.3V MCUs.  SD SPI is 3.3V so the sort of breakout you need should just have some resistors and capacitors on it, no voltage regulator or level shifter.

archearth, 2024-09-06:
I am in the same situation. I was able to get the SD to mount and list the files inside but after that when I tried to load a function that displays an image on the TFT screen it does not work. I am trying to use SdFs and the TFT_eSPI library and read/display images from the SD card.  If anyone has found a solution or has one, it will be much appreciated.

giri-v, 2025-01-18:
I have a ESP32-3248S035 board that also combines SD and TFT apparently. Any solution would be great. I'm happy to ship a board to someone to test.

giri-v, 2025-01-20:
I've gotten it to work. Steps should be done in order to make TFT_eSPI and an SD card share Pin 5 as the Chip Select (TFT_CS and SD_CS). Here is what I did:

```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI(); // Create object "tft"
SPIClass tftspi = tft.getSPIinstance();

#include <SD.h>

#define SD_CS 5
#define SD_SCLK 18
#define SD_MISO 19
#define SD_MOSI 23

SPIClass SDSPI(VSPI);

void initSD()
{
    pinMode(5, OUTPUT);
    digitalWrite(5, HIGH); // This appears to be the IMPORTANT step

    tftspi.begin(18, 19, 23); // Looks like this explicitly uses the TFT_eSPI provided SPI port
    tftspi.setFrequency(1000000);
    if (!SD.begin(5, tftspi))
    {
        Log.errorln("SD Card Mount Failed");
        return;
    }
}

void setupDisplay()
{
    Log.infoln("Setting up display.");
    tft.init();
    tft.setRotation(2);
    tft.fillScreen(TFT_BLACK);
}

void setup()
{
    setupDisplay();

    initSD();

    // Now you can use the SD card AND graphics
}

void loop()
{
}

```

I had to cut and paste from my codebase so it may not run as is, but I hope this helps with the proper sequencing to get them working.

greiman, 2025-01-20:
I said above that you always must insure multiple devices on an SPI bus don't interfere.  Best to set all chips selects high before initializing any.
 
see Above.
> Are you pulling chip select high on the second device to to be initialized so it won't interfere with initializing of the first device?