greiman/SdFat #478: Problems with multiple SD card

argentario84, 2024-04-10:
Hi, i have a project that write on SD card data from GPS and MPU6050 every 250ms with ESP32 (CS= Pin 5)

I have 4 SD card.

1) an old one that i can initialize and mount
https://i.etsystatic.com/25910604/r/il/61ae76/4951623534/il_794xN.4951623534_pjuc.jpg

The problem with this card is that sometimes i get strange chars instead of good one

> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00' 
> 01;*
> 80424;10025825;4356063	9+14480000;7;0.20;1.00;-0.00;-
> $ ;-0.01;*
> 80424;10025825;435:'33;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 80424;10025825;435606333;114480000;7;0.20;1.00;-0.00;-0.00;-0.01;*
> 

`_myFile = _sd.open(_nameFile, O_WRITE | O_APPEND | O_CREAT);
            _myFile.println(_strTempSDcardData);
            _myFile.close();`

I print the content of _strTemSDcardData in serial and is good so the error is on the writing on the card.
I thinked the reason was old card so i tried others card

2) a new one (arrived yesterday) 
https://m.media-amazon.com/images/G/29/apparel/rcxgs/tile._CB483369964_.gif
formatted fat32 with SD card formatter

i obtain: 
SD: Mount failed
08:28:10.425 -> SD errorCode: SD_CARD_ERROR_ACMD41 = 0x17
08:28:10.458 -> SD errorData = 0x1

3) SanDisk Ultra 80 MB/s 64 GB SDXC
formatted exFat
https://i.ebayimg.com/images/g/IokAAOSwhaBasYLY/s-l500.jpg
I obtain:
08:24:53.850 -> SD errorCode: SD_CARD_ERROR_ACMD41 = 0x17
08:24:53.884 -> SD errorData = 0xff

4) Microsd with SdAdapter
Sandisk Ultra 128Gb
https://i.ebayimg.com/images/g/SkkAAOSw~vJl-qxw/s-l500.jpg
formatted exFat

I obtain:
08:27:24.461 -> SD errorCode: SD_CARD_ERROR_ACMD41 = 0x17
08:27:24.494 -> SD errorData = 0xff

Connection are good i check with tester and the first SD card initialize and write

I'm getting crazy to solve this problem, why i can't initialize other sd? why the first one get strange char?
`if (!_sd.begin(SS, SD_SCK_MHZ(2))) {`
i tried different MHZ but the errors still come

Thanks in advance for possible answers.

greiman, 2024-04-10:
I have no idea.  You say:

>  "Connection are good i check with tester and the first SD card initialize and write" 

but all these errors look like wiring/ hardware problems.

argentario84, 2024-04-10:
I solved the problem of strange chars.

Instead of writing every 250ms i concatenate the string and write the data in SD every 10 points.
`if (_strTempSDcardData != "0;") {
            if (val == 10) {
              _newInsert = false;
              digitalWrite(PINOUT, HIGH);
              _myFile = _sd.open(_nameFile, O_WRITE | O_APPEND | O_CREAT);
              _myFile.print(myS);
              _myFile.close();
              delay(500);
              val = 0;
              myS = "";
            } else {
              myS += _strTempSDcardData;
              val++;
            }
          }`

Wiring seems ok but i think are too long and to much noise for access SD every 250ms? Could this be the cause of the failure of the other cards to initialize?

greiman, 2024-04-10:
> Wiring seems ok but i think are too long and to much noise for access SD every 250ms? Could this be the cause of the failure of the other cards to initialize?

I really can't guess what causes hardware problems without being able to put a scope or logic analyzer on the SD card.