greiman/SdFat #480: SdFat 2.22 with USER_SPI1 pin + another USER_SPI2 in SPI_Mode3

sobido, 2024-05-01:
Hi Bill
I have your Sfat2.22 on ESP32 DEV module on Arduino with special USER_SPI (GPIO_SPI_SCK, GPIO_SPI_MISO, GPIO_SPI_MOSI, GPIO_SPI_CS).
SD card access work fine as soon as I have only one SPI in use.
Now I need to add another User SPI (Gyro)  like (GPIO_IMU_SCLK, GPIO_IMU_DOUT, GPIO_IMU_DIN, GPIO_IMU_CS).
Moreover this SPI need to be in SPI_Mode3.
I could not find a solution to use both at the same time. So far , I need to SPI2.end() before to get access to the SD Card in SPI1.

I tried to find a solution with also your Sdfat_Beta version, but don't know how to do that .
Do you have a suggestion please ?

greiman, 2024-05-01:
> Moreover this SPI need to be in SPI_Mode3.

You can have devices with different modes on one bus. This is set by:
SPI.beginTransaction(settings);

See [this](https://www.arduino.cc/reference/en/language/functions/communication/spi/spisettings/).

sobido, 2024-05-01:
Hi Thanks
I found finally a solution . Maybe not the more beautiful one, but seems to work:
I used your SdFat_Beta instead of SdFat ,and changing priorly  the SPI_DRIVER_SELECT = 2 in the SdFat/SdFatConfig.h , and after using the **softSpiDriver** like in your "SoftwareSpi"example. 
I just had to be aware that I  define well my GPIO_IMU_SCLK, GPIO_IMU_DOUT, GPIO_IMU_DIN, GPIO_IMU_CS, and GPIO_SPI_CS as const unint8_t , and not as static gpio_num_t , as I did for all the rest of my gpio configuration.

I can now avoid any SPI.end() on my User.SPI Gyro or any sd.end() on the USER.SPI SDfat and have both User.SPI are working together.

Many Thanks for your help