greiman/SdFat #482: Problem with SdFat and TwoWire instance.

alexandrumot, 2024-05-26:
Hello. I'm using Bluepill on PlatformIO with platform = ststm32. I get the SD to work with the following simple configuration:
```
#define SD_CS_PIN PB12
static SPIClass sdSPI(PB15, PB14, PB13, SD_CS_PIN);
#define SD1_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(18), &sdSPI)
 
SdFat SD;
 
File logs;

void setup() {
  Serial.print("Initializing SD card...");
 
  if (!SD.begin(SD1_CONFIG)) {
    Serial.println("Initialization failed!");
    return;
  }
  Serial.println("Initialization done.");
}

void loop() {
  logs = SD.open("telemetry.txt", FILE_WRITE);
  
  if (logs) {
    logs.print("text");
    logs.close();
    Serial.println("done.");
  else {
    Serial.println("error opening telemetry.txt");
  }

}
```

But when I use the begin method of instance TwoWire Wire2(PB11, PB10); for my bmp280 I get error that I cannot open file (SD Initialization success in either case). If I use the default Wire instance it writes to file, but then the bmp280 does not work.

Has anyone encountered a similar problem and solved it?

alexandrumot, 2024-05-26:
It appears there is some collision between I2C2 pins PB11 SDA, PB10 SCL, and SdFat. When I use I2C1 PB8 SCL and PB9 SDA, it writes to SD correctly. One more interesting fact is that it fails to write only when the MCU successfully gets data from BMP280 on I2C2. But I need to use I2C2 for the BMP.

greiman, 2024-05-27:
I have never mapped pins on a Bluepill and don't have one so I can't try anything or help.

If you map SPI pins with SdFat, you may need to call `SPI.begin(<args>)` before `sd.begin()` if any arguments are required in the `SPI.begin()` call. 

Add USER_SPI_BEGIN to this definition like this:

`#define SD1_CONFIG SdSpiConfig(SD_CS_PIN, USER_SPI_BEGIN | DEDICATED_SPI, SD_SCK_MHZ(18), &sdSPI)`

If this fails, try the [stm32duino](https://www.stm32duino.com/) forum for help.

alexandrumot, 2024-05-28:
I see... It appears to be a STM32 core problem. I managed to remap default I2C pins. It writes now.

closed 2024-05-28 by alexandrumot