greiman/SdFat #483: sd.begin(SPI_CONFIG) breaks my DMA ADC datalogger on the SAMD51

cdd0042, 2024-05-29:
I have a data logger that takes in inputs from 2 ADCs and writes them to 2 circular buffers using DMA. This part works well. I am able to read data in and get it to the buffers at about 2.4 Mb/s.

The issue I was running into was how to store it. I eventually figured out how to use SdFat with the SPI configuration that utilizes adafruit's zeroDMA library to handle the SD card writes. I was able to write buffers to the SD card at about 2.7 Mb/s in my stand alone test code. This should be fast enough to keep up with the DMA data logger.

The issues arose when I tried to combine the two. Have data incoming over DMA get written to the SD card. The DMA initialization and setup all works perfectly until sd.begin(SD_CONFIG) is called. I am not sure what is going wrong or what is conflicting in the SdFat library with my DMA logger.

Any help would be great.

`// WIP: Use SAMD51's DMA Sequencing to read A0, A1, A3, and A4 on ADC0 and read A2 on ADC1 without CPU intervention and write them to an SD Card. Optional AC trigger and DAC sine wave generator

#include "FreeStack.h"
#include "SdFat.h"
#include "sdios.h"

#define error(s) sd.errorHalt(&Serial, F(s))
const uint8_t SD_CS_PIN = 5;
#define SPI_CLOCK SD_SCK_MHZ(50)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)

SdFat sd;
File file;

const size_t BUF_SIZE = 512;

volatile boolean results0Part0Ready = false;
volatile boolean results0Part1Ready = false;
volatile boolean results1Part0Ready = false;
volatile boolean results1Part1Ready = false;
uint16_t adcResults0[BUF_SIZE*2];                                                       // ADC results array 0
uint16_t adcResults1[BUF_SIZE*2];                                                       // ADC results array 1

uint16_t r0p0[BUF_SIZE];
uint16_t r0p1[BUF_SIZE];
uint16_t r1p0[BUF_SIZE];
uint16_t r1p1[BUF_SIZE];

// ADC0 INPUTCTRL register MUXPOS settings 
uint32_t inputCtrl0[] = { ADC_INPUTCTRL_MUXPOS_AIN0,                              // AIN0 = A0
                          ADC_INPUTCTRL_MUXPOS_AIN5,                              // AIN5 = A1
                          ADC_INPUTCTRL_MUXPOS_AIN3,                              // AIN3 = A3
                          ADC_INPUTCTRL_MUXPOS_AIN4 };                            // AIN4 = A4

//uint16_t sintable[256];                                                         // Sine table

typedef struct                                                                    // DMAC descriptor structure
{
  uint16_t btctrl;
  uint16_t btcnt;
  uint32_t srcaddr;
  uint32_t dstaddr;
  uint32_t descaddr;
} dmacdescriptor ;

volatile dmacdescriptor wrb[DMAC_CH_NUM] __attribute__ ((aligned (16)));          // Write-back DMAC descriptors
dmacdescriptor descriptor_section[DMAC_CH_NUM] __attribute__ ((aligned (16)));    // DMAC channel descriptors
dmacdescriptor descriptor __attribute__ ((aligned (16)));                         // Place holder descriptor

dmacdescriptor linked_descriptor[2] __attribute__ ((aligned (16)));               // Linked descriptors

void setup() {
  Serial.begin(115200);                                                           // Start the native USB port
  while(!Serial);                                                                 // Wait for the console to open
  
  

  pinMode(10, OUTPUT);                                                            // Initialise the output on D10 for debug purposes
  pinMode(11, OUTPUT);                                                            // Initialise the output on D11 for debug purposes

  DMAC->BASEADDR.reg = (uint32_t)descriptor_section;                              // Specify the location of the descriptors
  DMAC->WRBADDR.reg = (uint32_t)wrb;                                              // Specify the location of the write back descriptors
[... 238 lines cut ...]
    digitalWrite(10, LOW);
  }
}

void DMAC_4_Handler()                                                             // Interrupt handler for DMAC channel 4
{
  static uint8_t count1 = 0;                                                      // Initialise the count 
  if (DMAC->Channel[4].CHINTFLAG.bit.SUSP)                                        // Check if DMAC channel 4 has been suspended (SUSP) 
  {  
    DMAC->Channel[4].CHCTRLB.reg = DMAC_CHCTRLB_CMD_RESUME;                       // Restart the DMAC on channel 4
    DMAC->Channel[4].CHINTFLAG.bit.SUSP = 1;                                      // Clear the suspend (SUSP)interrupt flag
    if (count1)                                                                   // Test if the count1 is 1
    {
      results1Part1Ready = true;                                                  // Set the results 1 part 1 ready flag
    }
    else
    {
      results1Part0Ready = true;                                                  // Set the results 1 part 0 ready flag
    }
    count1 = (count1 + 1) % 2;                                                    // Toggle the count1 between 0 and 1 
    digitalWrite(11, HIGH);                                                       // Toggle the output high then low on D11 for debug purposes
    digitalWrite(11, LOW);
  }
}
`

greiman, 2024-05-30:
I can't help with Adafruit's SAMD51 DMA.  I tried to use it in one of my apps and gave up.

I suggest you use the Adafruit port of [SdFat](https://github.com/adafruit/SdFat) and post issues with Adafruit.

I was going publish fast example using Adafruit's DMA for SAMD but decided it was too flaky.

cdd0042, 2024-05-30:
Hey Greiman,

That makes sense.  It has been tricky.

I started with the Adafruit Fork of SdFat, but I was trying to achieve high write speeds to keep up with my datalogging ADCs and I don't think the Adafruit Fork supports the SAMD51 DMA based SPI transfers.

I used this example to get my high write speeds.  I was able to replicate it with the normal SdFat library: https://github.com/greiman/SdFat/issues/434

I was just curious if you knew anything about which registers the begin() method uses so that I could identify any conflicts in my DMA datalogger.  I can't find the root begin() anywhere.

raider-snake, 2024-06-04:
@greiman Is there anything in SdFat specifically when sd.begin(SD_CONFIG) is called that would enable the DMAC on a M4. Does SdFat have ZeroDMA as a dependency and if so where is that called in the library? I have done some testing with regard to the code above and found some interesting issue. I reconfigured the code write out the DMA controller and channel registers and added #define DEBUG_SD_BEGIN where I can easily configure if sd.begin is called or not. When sd.begin is not called (#define DEBUG_SD_BEGIN 0), the DMA runs normally and GPIO 10 and 11 (when on Grand Central board) pulse for 3us with a period of ~850 us indicating that the buffers are being filled and all of the DMA channels are sequencing properly (you can also see this in the registers that are printed). When sd.begin is called (#define DEBUG_SD_BEGIN 1), there is no pulsing on either GPIO. The registers indicate that something has changed the DMAC BASEADDR and WRB but none of the channel descriptor addresses. This causes CH2 (previously CH5) to throw a transfer error and suspend flag from the memory misalignment and halts the rest of the DMA sequence. Below is the code and serial monitor output which shows the error.
`#include "FreeStack.h"
#include "SdFat.h"
#include "sdios.h"

#define error(s) sd.errorHalt(&Serial, F(s))
const uint8_t SD_CS_PIN = 83;
#define SPI_CLOCK SD_SCK_MHZ(50)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)

SdFat sd;
File file;

// Debug Parameters
#define DEBUG_R0_P0 0
#define DEBUG_R0_P1 0
#define DEBUG_R1_P0 0
#define DEBUG_R1_P1 0
#define DEBUG_SD_BEGIN 1

volatile boolean results0Part0Ready = false;
volatile boolean results0Part1Ready = false;
volatile boolean results1Part0Ready = false;
volatile boolean results1Part1Ready = false;
uint16_t adcResults0[2048];                                                       // ADC results array 0
uint16_t adcResults1[2048];                                                       // ADC results array 1

// ADC0 INPUTCTRL register MUXPOS settings 
uint32_t inputCtrl0[] = { ADC_INPUTCTRL_MUXPOS_AIN0,                              // AIN0 = A0
                          ADC_INPUTCTRL_MUXPOS_AIN5,                              // AIN5 = A1
                          ADC_INPUTCTRL_MUXPOS_AIN3,                              // AIN3 = A3
                          ADC_INPUTCTRL_MUXPOS_AIN4 };                            // AIN4 = A4

//uint16_t sintable[256];                                                         // Sine table

typedef struct                                                                    // DMAC descriptor structure
{
  uint16_t btctrl;
  uint16_t btcnt;
  uint32_t srcaddr;
  uint32_t dstaddr;
  uint32_t descaddr;
} dmacdescriptor ;

volatile dmacdescriptor wrb[DMAC_CH_NUM] __attribute__ ((aligned (16)));          // Write-back DMAC descriptors
dmacdescriptor descriptor_section[DMAC_CH_NUM] __attribute__ ((aligned (16)));    // DMAC channel descriptors
dmacdescriptor descriptor __attribute__ ((aligned (16)));                         // Place holder descriptor

dmacdescriptor linked_descriptor[2] __attribute__ ((aligned (16)));               // Linked descriptors

void setup() {
  Serial.begin(115200);                                                           // Start the native USB port
  while(!Serial);                                                                 // Wait for the console to open

  #if DEBUG_SD_BEGIN
    // Initialize SD card before setting up DMA
    if (!sd.begin(SD_CONFIG)) {
      Serial.println("SD initialization failed!");
      while (1);  // Halt if SD card initialization fails
    } else {
      Serial.println("SD initialization succeeded!");
    }
  #endif

  pinMode(10, OUTPUT);                                                            // Initialise the output on D10 for debug purposes
  PORT->Group[PORTB].DIRSET.reg = PORT_PB23;                                      // Initialise the output on D11 for debug purposes TODO: make it autoselect based on board 

  /DMAC->BASEADDR.reg = (uint32_t)descriptor_section;                              // Specify the location of the descriptors
  /DMAC->WRBADDR.reg = (uint32_t)wrb;                                              // Specify the location of the write back descriptors
  // DMAC->BASEADDR.reg = 0x20002550;                                                // Specify the location of the descriptors
[... 321 lines cut ...]
    DSTADDR: 20000D3E
    DESCADDR: 200027A0
Channel 4
  CHCTRLA: 204602
  CHCTRLB: 0
  CHINTFLAG: 0
  CHINTENSET: 4
  CHINTENCLR: 4
  Descriptor Address: 20002590
    BTCNT: 1024
    SRCADDR: 43002040
    DSTADDR: 20001D3E
    DESCADDR: 200027B0
=== ADC Status ===
ADC0 RESULT: 3B
ADC0 INTFLAG: 0
ADC0 STATUS: 0
ADC1 RESULT: 2E
ADC1 INTFLAG: 0
ADC1 STATUS: 0
=== Other Important Registers ===
DMAC BASEADDR: 20000120 // Former BASEADDR was 20000550 which is 16 byte aligned with CH0
DMAC WRBADDR: 20000320
DMAC CTRL: F02
`

greiman, 2024-06-04:
> @greiman Is there anything in SdFat specifically when sd.begin(SD_CONFIG) is called that would enable the DMAC on a M4. 

SdFat does not know about DMA.  It may cause DMA to be used if the SPI library is implemented with DMA.

 SdFat only calls SPI.begin() with no arguments, SPI.beginTransaction(), SPI.transfer(), SPI.endTransaction() and possibly SPI.end() if sd.end() is called.

SdFat supports three  SPI transfer signatures:

```
// Arduino standard library
uint8_t transfer(uint8_t data);  // Rarely DMA
void transfer(void *buf, size_t count);  // May be DMA

// Adafruit, Teensy, and other non Arduino boards.
void transfer(const void* txBuf, void *rxBuf, size_t count);  // Often uses DMA.
```
Use of array transfers requires editing SdFatConfig [here](https://github.com/greiman/SdFat/blob/052d38e2c6ed64d862d8867b32e37f36b8c065ec/src/SdFatConfig.h#L167).

Baron-Paelen, 2024-09-10:
Hi @raider-snake, just commenting to see if you found any solutions yet or if greiman's solution above helped you. I'm working on something pretty similar on the same board and would appreciate any updates you may have!

Baron-Paelen, 2025-04-07:
For anyone else running into this issue, replace the original definitions of `wrb` and `descriptor_section` with:
```cpp
volatile dmadesc (*wrb)[DMAC_CH_NUM] __attribute__ ((aligned (16)));          // Write-back DMAC descriptors (changed to array pointer)
dmadesc (*descriptor_section)[DMAC_CH_NUM] __attribute__ ((aligned (16)));    // DMAC channel descriptors (changed to array pointer)
dmadesc descriptor __attribute__ ((aligned (16)));                            // Place holder descriptor
```

and add this to your setup somewhere:
```cpp
descriptor_section =  reinterpret_cast<dmadesc(*)[DMAC_CH_NUM]>(DMAC->BASEADDR.reg); //point array pointer to BASEADDR defined by SD.begin
    wrb =  reinterpret_cast<dmadesc(*)[DMAC_CH_NUM]>(DMAC->WRBADDR.reg);                 //point array pointer to WRBADDR defined by SD.begin
    DMAC->CTRL.reg = DMAC_CTRL_DMAENABLE | DMAC_CTRL_LVLEN(0xf); 
```

Fix found in [this thread](https://forum.arduino.cc/t/samd51-adc0-and-adc1-multiplexing-with-dma/1238002/7?u=paelen1234).