greiman/SdFat #485: exFat sd cards not working with Sdinfo.ino example, only FAT32 card

JacobDel, 2024-06-03:
Example source code: https://github.com/greiman/SdFat/blob/master/examples/SdInfo/SdInfo.ino

The only change made in the script was to change the CS pin (see full script below):
```
// // SDCARD_SS_PIN is defined for the built-in SD on some boards.
// #ifndef SDCARD_SS_PIN
// const uint8_t SD_CS_PIN = SS;
// #else   // SDCARD_SS_PIN
// const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
// #endif  // SDCARD_SS_PIN
const uint8_t SD_CS_PIN = 4;
```

microcontroller: arduino nano

sd card extension board (simple level shifters): Catalex MicroSD Card Adapter v1.0 11/01/2013 v MicroSD Card Adapter

**Working card** (see serial logs below):
- Transcend Premium 300x 64gb A1 - FAT32 formatted

**Not working cards**:
1. SanDisk Ultra 128GB A1 - (factory formatted) exFat format
2. Samsung Pro Plus 128GB A2 - exFat formatted

**128GB cards with exFat format**:
```
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_INVALID_CARD_CONFIG = 0x2b
SD errorData = 0x0

type any character to start
```
**64GB card with FAT32 format**
```
init time: 439 ms

Card type: SDXC
sdSpecVer: 3.00
HighSpeedMode: true

Manufacturer ID: 0X74
OEM ID: JE
Product: USDU1
Revision: 0.2
Serial number: 0XC8105DF6
Manufacturing date: 9/2014
CID HEX: 744A45555344553102C8105DF600E955

cardSize: 63383.27 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros
CSD HEX: 400E00325B590001D83D7F800A4000A5

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0XC,0XFE,0XFF,0XFF,32768,123762688
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

[... 271 lines cut ...]
  }
  printCardType();
  cout << F("sdSpecVer: ") << 0.01 * scr.sdSpecVer() << endl;
  cout << F("HighSpeedMode: ");
  if (scr.sdSpecVer() > 101 && sd.card()->cardCMD6(0X00FFFFFF, cmd6Data) &&
      (2 & cmd6Data[13])) {
    cout << F("true\n");
  } else {
    cout << F("false\n");
  }
  cidDmp();
  csdDmp();
  cout << F("\nOCR: ") << uppercase << showbase;
  cout << hex << ocr << dec << endl;
  if (!mbrDmp()) {
    return;
  }
  if (!sd.volumeBegin()) {
    cout << F("\nvolumeBegin failed. Is the card formatted?\n");
    errorPrint();
    return;
  }
  dmpVol();
}
```

greiman, 2024-06-03:
Did you edit SdFatConfig.h and modify SPI_DRIVER_SELECT [here](https://github.com/greiman/SdFat/blob/aadedbf206068181b9aedc5c2e7f302b246eeea7/src/SdFatConfig.h#L164)?

The error SD_CARD_ERROR_INVALID_CARD_CONFIG happens when SPI_DRIVER_SELECT  is >= 2 and no spi pointer is provided in `SdSpiConfig(`).

[Here](https://github.com/greiman/SdFat/blob/aadedbf206068181b9aedc5c2e7f302b246eeea7/src/SdCard/SdSpiCard.cpp#L136) is where the error occurs.

JacobDel, 2024-06-03:
Thank you Greiman for the quick response.

I did not modify SPI_DRIVER_SELECT in SdFatConfig.h.
Something I forgot to mention is the library version, 2.2.3., which should be the latest as I pulled it straight from git.
 

> The error `SD_CARD_ERROR_INVALID_CARD_CONFIG` happens when `SPI_DRIVER_SELECT` is >= 2 and no spi pointer is provided in `SdSpiConfig(`).

The error is not triggered in SdSpiCard.cpp.
The error comes from [here](https://github.com/greiman/SdFat/blob/aadedbf206068181b9aedc5c2e7f302b246eeea7/src/SdFat.h#L326) instead.

# Proof

Changes made to the code:
```
  /** \return SD card error code. */
  uint8_t sdErrorCode() {
    if (m_card) {
      return m_card->errorCode();
    }
    return 123; //SD_CARD_ERROR_INVALID_CARD_CONFIG;
  }
```
Serial output with the format32 card:
```
SdFat version: 2.2.3

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 488 ms

Card type: SDXC
sdSpecVer: 3.00
HighSpeedMode: true

Manufacturer ID: 0X74
OEM ID: JE
Product: USDU1
Revision: 0.2
Serial number: 0XC8105DF6
Manufacturing date: 9/2014
CID HEX: 744A45555344553102C8105DF600E955

cardSize: 63383.27 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros
CSD HEX: 400E00325B590001D83D7F800A4000A5

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0XC,0XFE,0XFF,0XFF,32768,123762688
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT32
sectorsPerCluster: 64
fatStartSector:    32800
dataStartSector:   63010
clusterCount:      1933319
freeClusterCount:  885415
```
Serial output with the exFat card:
```
type any character to start

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_UNKNOWN = 0X7B
SD errorData = 0X0
```

greiman, 2024-06-03:
There is a bug for the error code when SdFs fails.

Please change [this](https://github.com/greiman/SdFat/blob/aadedbf206068181b9aedc5c2e7f302b246eeea7/src/SdCard/SdCard.h#L69) function in SdCard.h to the following:

```
  bool cardBegin(SdSpiConfig spiConfig) {
    m_card = m_cardFactory.newCard(spiConfig);
    return m_card && !m_card->errorCode();
  }
```

Rerun the Sdinfo example and post the error code.

greiman, 2024-06-03:
The above had the wrong function.

JacobDel, 2024-06-03:
Serial output after proposed modifications:
```
SdFat version: 2.2.3

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_ACMD41 = 0x17
SD errorData = 0x1

type any character to start
```

JacobDel, 2024-06-03:
Thanks!
I'll close the issue until I can test this with other sd card adapters/sniffers.

closed 2024-06-03 by JacobDel