greiman/SdFat #486: Allocating bin file with 0s

pablobaezlinero, 2024-06-03:
I am writing my own version of the example "ExFatLogger", which will save the data only in a single ``.bin`` file, and not convert it into the csv.

However, when I test the code and reuse it, the previous logged data is in the same place in the memory as the new data, so it can be difficult to see when the data-logging ends (I dont close the file: only interrupt by disconecting the Arduino).

Would it be possible to fill the preallocated bin file with zeros instead of whatever it is in the memory it occupies?

I got this idea from the difference in C between calloc and malloc.

Thanks!

greiman, 2024-06-03:
> Would it be possible to fill the preallocated bin file with zeros instead of whatever it is in the memory it occupies?

You can do what ever you want.  I don't have time to help you write code.

I am closing this.

closed 2024-06-03 by greiman

pablobaezlinero, 2024-06-03:
First of all, thanks for your reply.

I was asking in case it was an already implemented option, rather than having to program it myself. I guess based on your response that this is not the case then.

Thank you