greiman/SdFat #487: CS_Pin has to be driven high on ESP32 (to be low at the SDCARD) because of some logic inbetween

ImkereiWetzel, 2024-06-11:
Hi,

is there a way to drive the CS_Pin manually, because I have used a little quirk because of lack of pins. 
My CS-Pin driven low, drives TFT-CS low and SD-CS  high, and if CS-Pin driven high exactly the other ways around.

Thanks in advance

greiman, 2024-06-12:
I can't suggest a solution since I have no idea what hardware you have.

SdFat does have an option for using a [port expander](https://learn.adafruit.com/adafruit-mcp23017-i2c-gpio-expander). but this is probably not practical.

You would need to provide the two functions described [here](https://github.com/greiman/SdFat/blob/052d38e2c6ed64d862d8867b32e37f36b8c065ec/src/SdFatConfig.h#L222).

ImkereiWetzel, 2024-06-12:
Thanks! I have an ESP32 and connected is an Adafruit 3.5 TFT w/ capacitive touch w/ onboard SD slot.
I already use multiple port expanders for other purposes, so my solution uses a little single gate inverter (NOT-Gate) to select the lcd if the pin is low, and select the SD if high. 
I will try to define my own functions and will come back if it works.

greiman, 2024-06-13:
>  my solution uses a little single gate inverter (NOT-Gate) to select the lcd if the pin is low, and select the SD if high.

It won't work.  CS needs to be toggled in many places in the SD driver and is not just low when the driver is active.