greiman/SdFat #492: QuickStart fails to compile on ESP32 V3.0.3

drmpf, 2024-08-16:
Installed SdFat V2.2.3, did not change config
compiling QuickStart on Sparkfun ESP32 thing, with ESP32 board support V3.0.3
gives the following compile errors

```
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp: In member function 'int StdioStream::printDec(float, uint8_t)':
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:276:15: error: 'fmtFloat' was not declared in this scope; did you mean 'float'?
  276 |   char *ptr = fmtFloat(value, buf + sizeof(buf), prec);
      |               ^~~~~~~~
      |               float
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp: In member function 'int StdioStream::printDec(uint16_t)':
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:310:15: error: 'fmtDec' was not declared in this scope; did you mean 'fmtHex'?
  310 |   char *ptr = fmtDec(n, buf + sizeof(buf));
      |               ^~~~~~
      |               fmtHex
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp: In member function 'int StdioStream::printDec(uint32_t)':
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:345:15: error: 'fmtDec' was not declared in this scope; did you mean 'fmtHex'?
  345 |   char *ptr = fmtDec(n, buf + sizeof(buf));
      |               ^~~~~~
      |               fmtHex
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp: In member function 'int StdioStream::printHex(uint32_t)':
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:371:22: error: invalid conversion from 'uint32_t' {aka 'long unsigned int'} to 'char*' [-fpermissive]
  371 |   char *ptr = fmtHex(n, buf + sizeof(buf));
      |                      ^
      |                      |
      |                      uint32_t {aka long unsigned int}
In file included from ...\Arduino\libraries\SdFat\src\FatLib\FatFile.h:35,
                 from ...\Arduino\libraries\SdFat\src\FatLib\StdioStream.h:32,
                 from ...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:25:
...\arduino\libraries\sdfat\src\common\fmtnumber.h:36:20: note:   initializing argument 1 of 'char* fmtHex(char*, uint32_t)'
   36 | char* fmtHex(char* str, uint32_t n);
      |              ~~~~~~^~~
...\Arduino\libraries\SdFat\src\FatLib\StdioStream.cpp:371:29: error: invalid conversion from 'char*' to 'uint32_t' {aka 'long unsigned int'} [-fpermissive]
  371 |   char *ptr = fmtHex(n, buf + sizeof(buf));
      |                         ~~~~^~~~~~~~~~~~~
      |                             |
      |                             char*
...\arduino\libraries\sdfat\src\common\fmtnumber.h:36:34: note:   initializing argument 2 of 'char* fmtHex(char*, uint32_t)'
   36 | char* fmtHex(char* str, uint32_t n);
      |                         ~~~~~~~~~^
In file included from ...\Arduino\libraries\SdFat\src\FatLib\fstream.cpp:25:
...\Arduino\libraries\SdFat\src\FatLib\fstream.h: In member function 'virtual void fstream::getpos(FatPos_t*)':
...\Arduino\libraries\SdFat\src\FatLib\fstream.h:140:14: error: 'getpos' is not a member of 'FatFile'
  140 |     FatFile::getpos(pos);
      |              ^~~~~~
...\Arduino\libraries\SdFat\src\FatLib\fstream.h: In member function 'virtual void fstream::setpos(FatPos_t*)':
...\Arduino\libraries\SdFat\src\FatLib\fstream.h:164:14: error: 'setpos' is not a member of 'FatFile'
  164 |     FatFile::setpos(pos);
      |              ^~~~~~
...\Arduino\libraries\SdFat\src\FatLib\fstream.h: In member function 'virtual void ifstream::getpos(FatPos_t*)':
...\Arduino\libraries\SdFat\src\FatLib\fstream.h:225:14: error: 'getpos' is not a member of 'FatFile'
  225 |     FatFile::getpos(pos);
      |              ^~~~~~
...\Arduino\libraries\SdFat\src\FatLib\fstream.h: In member function 'virtual void ifstream::setpos(FatPos_t*)':
...\Arduino\libraries\SdFat\src\FatLib\fstream.h:237:14: error: 'setpos' is not a member of 'FatFile'
  237 |     FatFile::setpos(pos);
      |              ^~~~~~
 
```

greiman, 2024-08-17:
The directory structure for the above is not correct for V2.2.3 StdioStream.cpp is not located here:

\SdFat\src\FatLib\StdioStream.cpp

It is [here](https://github.com/greiman/SdFat/blob/master/src/iostream/StdioStream.cpp).

Many other files are wrong for V2.2.3 in your example above.

I used ESP 3.0.4:
[image]

And QuickStart compiles:

[image]

drmpf, 2024-08-17:
Tried again this time I used Arduino Import Library to install .zip file instead of using WinZip to unzip it.
Now it compiles.  Thanks

closed 2024-08-17 by drmpf