greiman/SdFat #497: error compiling on Thinary Every board

rickyelqasem, 2024-10-06:
get this error with sdfat 2.2.3:

SpiDriver\SdSpiChipSelect.cpp:34:26: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

greiman, 2024-10-06:
I don't understand why the error is at line 34 of SdSpiChipSelect.cpp

[Here](https://github.com/greiman/SdFat/blob/master/src/SpiDriver/SdSpiChipSelect.cpp#L34) is line 34.

Seems like it should be a line  31 or 37 with `digitalWrite()`.

I suspect the Thinary Every board support package need the second argument of `digitalWrite()` to be either HIGH or LOW.

Try replacing
` { digitalWrite(pin, level); }`
with
`{ digitalWrite(pin, level ? HIGH : LOW); }`

rickyelqasem, 2024-10-07:
Made no difference. But there error is slightly different. Here is the full error;

C:\Users\ricky\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiChipSelect.cpp: In function 'void sdCsWrite(SdCsPin_t, bool)':
C:\Users\ricky\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiChipSelect.cpp:31:68: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'
 void sdCsWrite(SdCsPin_t pin, bool level) { digitalWrite(pin, level); }

rickyelqasem, 2024-10-07:
Scratch that .. after noticing the last error was on 31, I changed:
void sdCsWrite(SdCsPin_t pin, bool level) { digitalWrite(pin, level ? HIGH : LOW); }
and that worked.