greiman/SdFat #499: recovery strategy

folkertvanheusden, 2024-11-11:
Hi,

I created a program which exports an SD-card on e.g. a ESP32 as an iSCSI device.
Now when reading & writing large amounts of data, I occasionally get an error: a file.read() (or write) returns -1 instead of what is requested.
What is the best strategy to recover from this? Reboot? retry x times? re-initialize the SDFat library?

Thanks!

greiman, 2024-11-12:
If you access the SD both by SdFat and as an iSCI device you will get errors since both devices will cache data.  The file system will become corrupt.

This is the reason network file systems were developed in the 1980s.

folkertvanheusden, 2024-11-12:
That's not how my program works.
My program accesses a file on the sd-card and serves that as an iSCSI device to the network.

greiman, 2024-11-13:
I can't guess what is happening.  Try printing the error code after a failed read/write.

You can call [errorPrint](https://github.com/greiman/SdFat/blob/052d38e2c6ed64d862d8867b32e37f36b8c065ec/src/SdFat.h#L284) like this:

  `sd.errorPrint(&Serial, "read failed");`

What SPI rate are you using?  I have seen errors on some ESP32 boards at SPI clock rates over 20 MHz.

closed 2024-11-17 by folkertvanheusden