greiman/SdFat #501: SdFat compatibility with ESP8266

VantagePro2, 2024-12-20:
Hi,
The issue of compatibility of the sd version with the ESP8266 was already discussed two years ago, 
See: https://github.com/greiman/SdFat/issues/327
But I still haven't figured out if this issue has been resolved.
I'm trying to run an SdInfo sketch from the examples.
The connection between NodeMCU (ESP8266) and microSD Shield is as shown in the example at the link: https://www.instructables.com/SD-Card-Module-With-ESP8266/
When the SD_CS_PING parameter was set to "SS" there was the massage on monitor that the microSD card  cannot be initialized on pin 8.  
When the SD_CS_PAGING parameter was set to "8" there wasn't any messages are displayed on the monitor.
What is the error?

greiman, 2024-12-20:
You should use the version of SdFat [here](https://github.com/earlephilhower/ESP8266SdFat) for ESP8266.

VantagePro2, 2024-12-21:
Good evening!
I downloaded the SdFat from https://github.com/earlephilhower/ESP8266SdFat.
Also I deleted all another versions SdFat libraries from \\Arduino\libraries\
Unfortunatelly I don't get info about SD card in Port Monitorю
Even I defined SD_CS_PIN = SS/4/8/10 there wasn't info about SD card

greiman, 2024-12-21:
You need to post on the above site.  I did not port it to ESP8266.