greiman/SdFat #502: Few compile errors when building using an Atmega128

rickyelqasem, 2024-12-24:
Atmega128 is an AVR / Megacore using the Arduino framework so presumed sdfat would compile without errors but I got a few:

.pio\libdeps\Upload_ISP\PCM\TMRpcm.cpp:1442:61: error: use of deleted function 'SdFile::SdFile(const SdFile&)'
             unsigned long pos = searchMainTags(xFile, datStr);
                                                             ^
In file included from .pio\libdeps\Upload_ISP\PCM\TMRpcm.cpp:9:0:
.pio\libdeps\Upload_ISP\SdFat\src/SdFat.h:468:7: note: 'SdFile::SdFile(const SdFile&)' is implicitly deleted because the default definition would be ill-formed:
 class SdFile : public PrintFile<SdBaseFile> {
       ^~~~~~
.pio\libdeps\Upload_ISP\SdFat\src/SdFat.h:468:7: error: use of deleted function 'PrintFile<FsBaseFile>::PrintFile(const PrintFile<FsBaseFile>&)'
In file included from .pio\libdeps\Upload_ISP\SdFat\src/ExFatLib/ExFatFile.h:883:0,
                 from .pio\libdeps\Upload_ISP\SdFat\src/ExFatLib/ExFatVolume.h:27,
                 from .pio\libdeps\Upload_ISP\SdFat\src/ExFatLib/ExFatLib.h:28,
                 from .pio\libdeps\Upload_ISP\SdFat\src/SdFat.h:31,
                 from .pio\libdeps\Upload_ISP\PCM\TMRpcm.cpp:9:
.pio\libdeps\Upload_ISP\SdFat\src/ExFatLib/../common/ArduinoFiles.h:43:7: note: 'PrintFile<FsBaseFile>::PrintFile(const PrintFile<FsBaseFile>&)' is implicitly deleted because the 
default definition would be ill-formed:
 class PrintFile : public print_t, public BaseFile {
       ^~~~~~~~~
.pio\libdeps\Upload_ISP\SdFat\src/ExFatLib/../common/ArduinoFiles.h:43:7: error: 'FsBaseFile::FsBaseFile(const FsBaseFile&)' is private within this context
In file included from .pio\libdeps\Upload_ISP\SdFat\src/FsLib/FsLib.h:31:0,
                 from .pio\libdeps\Upload_ISP\SdFat\src/SdFat.h:33,
                 from .pio\libdeps\Upload_ISP\PCM\TMRpcm.cpp:9:
.pio\libdeps\Upload_ISP\SdFat\src/FsLib/FsFile.h:75:3: note: declared private here

greiman, 2024-12-24:
The problem is in this call  `searchMainTags(xFile, datStr)` a copy constructor is used for `xFile`. 

This can cause file corruption since the state of the file is lost when the constructed object is destroyed on return from `searchMainTags()`.  See [this](https://github.com/greiman/SdFat/blob/052d38e2c6ed64d862d8867b32e37f36b8c065ec/src/SdFatConfig.h#L50).

rickyelqasem, 2024-12-25:
So I changed #define FILE_COPY_CONSTRUCTOR_PRIVATE 1 to #define FILE_COPY_CONSTRUCTOR_PRIVATE 2 and the problem went away but can you explain why that happened?

greiman, 2024-12-25:
> but can you explain why that happened?

I disable the copy constructor since SdFat is designed to have only one instance of the File object for an open file.  The state of the file is maintained in the file object.  I you call a function with a copy of the file object, any changes of the file state will be lost.

See [this](https://www.geeksforgeeks.org/difference-between-call-by-value-and-call-by-reference-in-cpp/).

You should design functions with File object arguments as call by reference or call by pointer, not call by value.

```
File myFile;

void callByReference(File& file) {
    file.write(123);
}
void callByPointer(File* file) {
  file->write(123);
}
void callByValue(File file) {
  file.write(123);
}
void test() {
  callByReference(myFile); // OK
  callByPointer(&myFile); // OK
  callByValue(myFile);  // Error
}
```

You are lucky that it works in your app but if you have problems don't post an issue.