greiman/SdFat #503: error when compiling

Setoula, 2024-12-28:
C:\Users\bAppData\Local\Temp\ccb469Tf.ltrans0.ltrans.o:(.rodata+0x6): undefined reference to `Print::write(unsigned char const*, unsigned int)'
c:/users/b/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

greiman, 2024-12-28:
I have no idea how this involves SdFat.

brightproject, 2024-12-29:
> C:\Users\bAppData\Local\Temp\ccb469Tf.ltrans0.ltrans.o:(.rodata+0x6): undefined reference to `Print::write(unsigned char const*, unsigned int)' [...]

It's really hard to help you without knowing what microcontroller, what IDE and what framework you're using.
I've had this happen when building code for a framework

> https://github.com/platformio/platform-ststm32

It's really hard to help you without knowing what microcontroller, what IDE and what framework you're using.
I had this happen when building code for a framework

> https://github.com/orgs/stm32duino/discussions/2523

Then I used a project that was written for Linux and required opening files.
You seem to have something to do with the main file, again, write more input data and the task will become clearer🙂