greiman/SdFat #508: Explanation about openNext vs open functions

SherifMEldeeb, 2025-01-22:
### **Issue Description:**
I am reading a hidden directory content that was created earlier how ever for some reason that directory sometimes gets empty files with 0bytes size and weird names like
```
Reading directory test
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ?(???
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ?@??
         0 2025-01-21 17:11 %?
         0 2025-01-21 17:11 %?
         0 2025-01-21 17:11 `???
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ??!??
         0 2025-01-21 17:11 ???!
         0 2025-01-21 17:11 ???!?
         0 2025-01-21 17:11 ??!
         0 2025-01-21 17:11 ???!?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ??.??
         0 2025-01-21 17:11 ?(?
         0 2025-01-21 17:11 ?@?!
         0 2025-01-21 17:11 a???
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ????
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??!??
         0 2025-01-21 17:11 ???
         0 2025-01-21 17:11 ?.?
         0 2025-01-21 17:11 ?.B
         0 2025-01-21 17:11 ????@?
         0 2025-01-21 17:11 ??@
         0 2025-01-21 17:11 ?@??
         0 2025-01-21 17:11 ?@
         0 2025-01-21 17:11 ?@?
         0 2025-01-21 17:11 ?@??
         0 2025-01-21 17:11 ?$
         0 2025-01-21 17:11 ?$?
         0 2025-01-21 17:11 ??3
         0 2025-01-21 17:11 ?B??
         0 2025-01-21 17:11 ?D?
         0 2025-01-21 17:11 D?
         0 2025-01-21 17:11 e?
         0 2025-01-21 17:11 e?
         0 2025-01-21 17:11 e??
         0 2025-01-21 17:11 e?
         0 2025-01-21 17:11 e?AB
         0 2025-01-21 17:11 e???@
         0 2025-01-21 17:11 e?B
         0 2025-01-21 17:11 g???
         0 2025-01-21 17:11 m?
         0 2025-01-21 17:11 t?
         0 2025-01-21 17:11 u?
         0 2025-01-21 17:11 ?
         0 2025-01-21 17:11 ??
         0 2025-01-21 17:11 ?b@??
```
As I understood while. debugging this issue that openNext and printfile names gets their files from cached entries or something like that how ever all CRUD operations like remove, read, write, exists use open method which I suppose it uses the actual card content so the problem is these are useless files for me so I want to delete them when ever they are found but all CRUD operations failes on these files except for reading their name I am not sure how to fix this or how to debug it.

### **Current Behavior:**
Can't remove or open 0 bytes files exist in a directory.

### **Expected Behavior:**
If reading directory contents returns x number of files I should be able to interact with all these files.

### **Setup:**
Controller:                  Esp32s3
Library Version:         2.1.2

### **Card Info:**
```
init time: 5 ms
SdFat version: 20102

Card type: SDHC

Manufacturer ID: 0X0
OEM ID: 
Product: 
Version: 0.0
Serial number: 0X996B0000
Manufacturing date: 10/2016

cardSize: 4026.53 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
OCR: 0XC0FF8000
```

Andy2No, 2025-01-22:
@SherifMEldeeb Can you replicate the behaviour with other cards?  Could it just be that this one is faulty, fake or corrupt?

A lot of cards are sold with hacked controller chips to report a larger capacity, though that's probably more likely with cards that claim to be larger than 4GB.  Those cards fail once more data is written to them than they can actually hold, or could fail sooner.

SherifMEldeeb, 2025-01-22:
@Andy2No Greetings Thanks for quick response.

Actually the original card is 4 GB but I have replicated the issue on a brand new 8GB card however I just found out that the files names were in UTF8 format so I just enabled the UTF8 format support in sdFat then it worked as expected !!!

Now I am not sure does this really fix the issue or just this case. I will try to test it further the close the issue if everything as expected.

Thanks again.