greiman/SdFat #510: CardInfo and QuickStart for ESP32

VantagePro2, 2025-01-25:
Good evening!
I'm trying to connect SD shield to ESP32 by SPI:
MISO ->D19
MOSI->D23
SCK->D5
CS->D18
However, nothing happens after uploading the sketches CardInfo and QuickStart.
I used two different voltage supply - 3.3 and 5 V - for different shields.
But there aren't any information in Arduino IDE Monitor.
Why?

greiman, 2025-01-26:
> However, nothing happens after uploading the sketches CardInfo and QuickStart.

Is there no output from QuickStart?  Do you see this?

> Be sure to edit DISABLE_CHIP_SELECT if you have
> a second SPI device.  For example, with the Ethernet
> shield, DISABLE_CHIP_SELECT should be set to 10
> to disable the Ethernet controller.
> 
> SD chip select is the key hardware option.
> Common values are:
> Arduino Ethernet shield, pin 4
> Sparkfun SD shield, pin 8
> Adafruit SD shields and modules, pin 10
> 
> Enter the chip select pin number: 

If you see this, what happens after entering a chip select pin number?

You should see something like this with error info.

> Enter the chip select pin number: 10
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CHIP_SELECT to disable another device.
> 
> SD initialization failed.
> Do not reformat the card!
> Is the card correctly inserted?
> Is chipSelect set to the correct value?
> Does another SPI device need to be disabled?
> Is there a wiring/soldering problem?
> 
> errorCode: 0x1, errorData: 0xff

VantagePro2, 2025-01-27:
Yes,
I checked  the different pins.
Also I used two different boards ESP32 and some SD shields (3.3 and 5 volts)
Unfortunately there wasn't any info in the monitor.
So I decided to ask you

greiman, 2025-01-27:
I can't help since I don't have your hardware here and Serial.print does not work.

Try the [BackwardCompatibility](https://github.com/greiman/SdFat/blob/master/examples/BackwardCompatibility/BackwardCompatibility.ino) example.  It is very simple.

VantagePro2, 2025-01-27:
OK
Thanks for support