greiman/SdFat #513: Long time data logging in small files

Anjali8340, 2025-02-15:
Hello,
i am using Teensy4.1 to log data into sd card. i am logging sample of data in every millisec, each sample is of 16 bytes. When file size exceed file size limit which is 16MB i create a new file and close the previous one. i am using ring buffer to store data and write to sd card only when 512 bytes are collected. 

Below function is called every whenever data logging is required 
void buff_fill(uint32_t t)
{
  size_t n = rb.bytesUsed();
  
 
  rb.write(startBytes, NUM_START_BYTES); // 3 bytes
  rb.write(&t, NUM_T_BYTES); //4 bytes
  rb.write(currentState); // 1 bytes
  rb.write(&Trig_count, NUM_COUNT_BYTES); //6bytes
  rb.write(&DAC_value, NUM_DAC_BYTES); //2bytes
  if (n >= SECTOR_SIZE && !file.isBusy())
  {
    rb.writeOut(SECTOR_SIZE);
    if (file.curPosition() >= MAX_FILE_SIZE)
    {
      createNewFile();
    }    
  }

bool createNewFile()
{
  // Close the current file if it's open
  if (file.isOpen())
  {
    // if (!rb.sync())
    //   return false;
    // rb.sync();
    // file.truncate();
    file.sync(); // Ensure data is written before closing
    file.close();
  }
  getFileName(fc);
  file.open(fileName, O_CREAT | O_WRONLY);
  file.preAllocate(MAX_FILE_SIZE);

  return true;
}

Problem is whenever a new file is created a huge delay is observed that block my code for more than minimum 5 millisec and max 50 millisec and some samples are lost. Logging is interrupt based . Whenever an interrupt occur a flag is set and i call this buff_fill function to write data. Interrupt occur every millisec so i cannot offer a code block of more than 900 microsec.

What should i do? Can you suggest me a better way to approach this issue

greiman, 2025-02-15:
You could create a series of per-allocated files.  Creating a file and allocating space on an SD can take an arbitrary amount of time since there is no free list for space.  A linear search is done for directory entries and file data space.

hattesen, 2025-02-17:
@greiman I assume you med "pre allocated".
Can you provide a link to example code for pre-allocating file space on an SD-card?

greiman, 2025-02-17:
Sorry about the typo it is per-allocate.

First open an empty file like [this](https://github.com/greiman/SdFat/blob/67e26476f15a3bae5f390d91b6cc01830920c55d/examples/bench/bench.ino#L194).  Then allocate a block of contiguous space like [this](https://github.com/greiman/SdFat/blob/67e26476f15a3bae5f390d91b6cc01830920c55d/examples/bench/bench.ino#L219). 

For best performance a contiguous file write is important.  512 bytes sectors are emulated in very large flash pages.  Record Units, RUs, can be as large as 512KB on modern cards.

I have used an array of file objects when the size of each file must be limited.  That way you don't need to close and open files while logging data.

`  FsFile files[NFILE];`

Anjali8340, 2025-03-26:
@greiman i tried logging by opening series of files using an array of file object. It worked, thank you.
i needed to create an array of ringbuff object also for every file object.
object array is declared like this:
```
FsFile fileArr[NFILES];
RingBuf<FsFile, RingBuffCap> rb[NFILES];
```

this function is called once in setup 
```
void createfiles()
{
  for (size_t i = 0; i < NFILES; i++)
  {
    fileArr[i].open(fileName, O_WRONLY | O_CREAT | O_TRUNC);
    fileArr[i].preAllocate(MAX_FILE_SIZE);
    rb[i].begin(&fileArr[i]);
    getFileName();
  }
}

```
this function is called whenever i want to log data( which is every millisec):
```
void buff_fill(uint32_t t)
{
  //size_t n = rb[fileIndex].bytesUsed();
  rb[fileIndex].write(startBytes, NUM_START_BYTES);
  rb[fileIndex].write(&t, NUM_T_BYTES);
  rb[fileIndex].write(currentState);
  rb[fileIndex].write(&Trig_count, NUM_COUNT_BYTES);
  rb[fileIndex].write(&DAC_value, NUM_DAC_BYTES);
  if (rb[fileIndex].bytesUsed() >= SECTOR_SIZE && !fileArr[fileIndex].isBusy())
  {
    //Serial.println("sync"); 
    rb[fileIndex].writeOut(SECTOR_SIZE);
    if ((fileArr[fileIndex].curPosition() >= MAX_FILE_SIZE) && ((fileIndex+1) < NFILES))
    {
      
      // if ((fileIndex+1) < NFILES)
      // {
        // fileArr[fileIndex].sync();
        // fileArr[fileIndex].close();
        // fileArr[fileIndex].truncate();
        //rb.sync();
        fileIndex++;
        //rb.begin(&fileArr[fileIndex]);
        // x = micros();
        // rb[fileIndex].begin(&fileArr[fileIndex]);
        
      //}
      // rb.sync();
      // fileArr[fileIndex].sync();
      // fileArr[fileIndex].close();
      // fileArr[fileIndex].truncate();
      // fileIndex++;
      // rb.begin(&fileArr[fileIndex]);
      // openNewFile();
    }
  }
}

```
but using number of ring buffers is taking huge space is there any other way to use single ring buffer for all file object. 

if i am calling this function rb.begin(&fileArr[fileIndex]); then first rb.writeout() is taking anonymous time which is very large in some instances.

greiman, 2025-03-26:
> but using number of ring buffers is taking huge space is there any other way to use single ring buffer for all file object.

You only need one ring buffer.

When you switch files use rb.brgin() to select the new file:

`rb.begin(&fileArr[i]);`

When you finish a file do this:

```
rb.sync();  // Write any data in the buffer to the file.
fileArr[i].close(); // close the file.
```

Anjali8340, 2025-04-02:
It is working. but i will need large number of files like 100 files to log 5 days data. in that case creating 100 objects will use huge memory.

Andy2No, 2025-04-03:
@Anjali8340 You could maybe have two tasks, one per core, with one task setting a flag then opening a file, then clearing the flag.  Meanwhile the other task would either write results to the file, or save them to a buffer, depending on the status of the flag.  While the flag is clear, indicating the file is open for writing, that second task can write results from the buffer to the file if it's not empty, or write directly to the file if it's empty.

If there's a non-blocking way to open a file, it could be done without the need for multi tasking - just ask to open the file and then poll for it being ready.  In the meantime, write to the buffer instead.

A circular buffer would be a good way of doing it.

greiman, 2025-04-03:
> You could maybe have two tasks, with one task setting a flag then opening a file, then clearing the flag. Meanwhile the other task would either write results to the file

This will be very slow.  Modern SD cards have huge flash pages and are only  fast for sequential access.  Internal buffering and algorithms are designed for sequential access.  An SD Record Uint can be as large as 512 KB.

The 512 byte sector is just a transfer unit not a flash page. The physical 512 byte sector dates back over 20 years when the first SD cards we 16 MB, not 256GB.

The SDIO controller is not designed to have two streams, one interrupt and the other background.  It has a 512 byte FIFO and write to the SD overlaps execution of the program.  Transfer to the FIFO only takes a few µs.

Too bad you are limited to many small files.  Modern SD cards are designed for large video files.

If possible I would write to a large file and split the large to small files for the application that needs small files.