greiman/SdFat #514: Problems with small capacity memory card

camd97, 2025-02-28:
Hi, I tried to use a 128MB class 4 micro SD card from Cloudisk without success.

After formatting it with the SD Card Formatter tool, the library manages to initialize the card, open a file and write to it for a while, then the library fails to initialize the SD anymore.

SD_SCK_MHZ(1)

When trying to format with the SdFormatter.ino sketch

the error is:

Options are:
E - erase the card and skip formatting.
F - erase and then format the card. (recommended)
Q - quickly format the card without erasing.

Enter option: F

Erasing
error: erase failed
SD errorCode: SD_CARD_ERROR_ERASE = 0x27
SD errorData = 0xff

Thanks for your comment.

greiman, 2025-02-28:
> SD errorCode: SD_CARD_ERROR_ERASE = 0x27

This error code is:
> 0X27,SD_CARD_ERROR_ERASE - Card did not accept erase commands

The SD Association Format tool does not use the low level erase command since they can't be sent through most USB adapters.  By default the SD association format tool does not erase the data areas of the card.

 If you select "Overwrite format", It just erases using write commands which is very slow.  

[image]

I don't implement Overwrite since it takes so long using SPI on Arduino boards.

I have no Idea why the Cloudisk fails to initialize after some time.

A possibility is not properly closing a file open for write 

I suggest you try  the standard SD.h library.  It is based on a 2009 version of SdFat and only small cards with limited features existed then.  I rarely test with card 2GB or smaller.  I don't have any Cloudisk cards and suspect they do not conform to Ver 3.1 of the SD spec.

from SD/src/utility/SdFat.h
```
/* Arduino SdFat Library
   Copyright (C) 2009 by William Greiman

   This file is part of the Arduino SdFat Library

```

greiman, 2025-02-28:
I found an old HanDisk 128 MB class 4 sd.

I used a Adafruit RP2350 Feather with an Adalogger SD Wing.

I ran the SdFormatter example:

> Enter option: F
> 
> Erasing
> .
> All data set to 0xff
> Erase done
> Writing FAT .....................................
> Format Done
> Run the SdInfo example for format details.

I ran SdInfo:

> init time: 3 ms
> 
> Card type: SD2
> sdSpecVer: 2.00  <<<<<<<<<<<<<<<<<< Really Old spec
> HighSpeedMode: true
> 
> Manufacturer ID: 0X0
> OEM ID: 0
> Product: APPSD
> Revision: 0.0
> Serial number: 0XBB5
> Manufacturing date: 5/2018  <<<<<<<<<<<<<<< Fairly New
> CID HEX: 00300041505053440000000BB50125C5
> 
> cardSize: 125.83 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> CSD HEX: 007F0032535A803BEEBBFF9F16800053
> 
> OCR: 0X80FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X3,0X4,0X0,0X6,0X7,0XE0,0XBF,99,245661
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT16
> sectorsPerCluster: 32
> fatStartSector:    100
> dataStartSector:   192
> clusterCount:      7674
> freeClusterCount:  7365

I ran the bench example:
FreeStack: 515992
Type is FAT16
Card size: 0.13 GB (GB = 1E9 bytes)

Manufacturer ID: 0X0
OEM ID: 0
Product: APPSD
Revision: 0.0
Serial number: 0XBB5
Manufacturing date: 5/2018

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2096.44,61076,139,243
2044.99,60962,139,250

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
3728.56,163,136,136
3728.56,156,136,136

camd97, 2025-02-28:
Thank you very much for your prompt response, @greiman!

The cards are actually not old; they were manufactured in 2023, and [we purchased a batch of 100](https://www.amazon.com/Cloudisk-Micro-128MB-Capacity-Memory/dp/B0D2QHJ4B4/ref=cm_cr_arp_d_product_top?ie=UTF8&th=1) a few weeks ago.

SdInfo for Cloudisk-Micro-128MB:

> Manufacturer ID: 0X0
> OEM ID: 0
> Product: APPSD
> Revision: 0.0
> Serial number: 0X81F
> Manufacturing date: 12/2023
> CID HEX: 0030004150505344000000081F017C33
> 
> cardSize: 125.83 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> CSD HEX: 007F0032535A803BEEBBFF9F16800053
> 
> OCR: 0X80FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X0,0X20,0X0,0XC,0X4B,0X3B,0XF,32,245728
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT16
> sectorsPerCluster: 8
> fatStartSector: 64
> dataStartSector: 544
> clusterCount: 30652
> freeClusterCount: 30648

I am currently testing it with the standard SD.h library, and it is working without any issues. I have written more than 3000 lines to my file, and we will continue monitoring its performance over the next few days. Thanks for your suggestion.

It would be great if the SdFat.h library could support this type of card, as they are currently available on the market. At a large-scale manufacturing level, their price is very favorable for reducing costs. Clearly, I am not certain how long they will remain operational or whether they will last the entire lifetime of the device.

greiman, 2025-02-28:
I look through old micro SD cards people have sent me.  

I found these two from a batch someone had problems with.  They worked reliably in devices like phones, PCs, Macs, etc that use the SDIO 4-bit but had reliability problems in SPI mode for the user.

[image]

They seem much like my HanDisk microSD cards internally.  Formatted with a different non-standard format.

Both lack a true Manufacturer ID or OEM ID.  The HanDisk and Cloudisk have the same Product name, APPSD.  They are clearly not certified by the SD Association.

Both of these seem to work with the bench example for my test setup.  No way I will try to find why you have problems with  uncertified Chinese micro SD cards.

First Disk:
> Type is FAT16
> Card size: 0.13 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X0
> OEM ID: 0
> Product: APPSD
> Revision: 0.0
> Serial number: 0X5ED
> Manufacturing date: 5/2018
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 2123.14,59492,139,240
> 2072.11,59383,139,246
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 3736.92,161,136,136
> 3736.92,158,136,136

Second Disk:

> Type is FAT16
> Card size: 0.13 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X0
> OEM ID: 0
> Product: APPSD
> Revision: 0.0
> Serial number: 0X1EF
> Manufacturing date: 5/2018
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 2139.50,58326,139,239
> 2088.55,58078,138,244
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 3736.92,145,136,136
> 3736.92,152,136,136

camd97, 2025-02-28:
Thank you very much, @greiman, for the clarification. I understand your point.

Do you know of any other similarly priced cards available on the market that are supported by the SdFat.h library and certified by the SD Association?

camd97, 2025-02-28:
Thank you, @greiman, for the clarification and the useful information. It makes sense that the market has shifted towards SDXC cards. I appreciate your time and insights!

closed 2025-02-28 by camd97