greiman/SdFat #515: Two SDIOs on RP2040

Stass48, 2025-03-11:
Hello! I’m really sorry for the silly question—maybe someone has already asked this (but I couldn’t find it).
	1.	Could you please tell me if it’s possible to use two SD cards on a single RP2040 (for data copying and other tasks)?
	2.	Will other PIO-based interfaces, such as I2S, still work properly in this setup?
	3.	Will there still be any remaining PIO resources available on the controller with such a configuration?

I apologize if my questions seem too basic. I truly appreciate your help and willingness to discuss these topics. I remember our conversation a year ago—you helped me a lot back then. Thank you so much for everything!

greiman, 2025-03-11:
>1.  Could you please tell me if it’s possible to use two SD cards on a single RP2040 (for data copying and other tasks)?

You can use many SPI cards but only one SDIO card.  There is not sufficient memory in a PIO block to add support for multiple cards on one PIO controller.

> 2. Will other PIO-based interfaces, such as I2S, still work properly in this setup?
> 3. Will there still be any remaining PIO resources available on the controller with such a configuration?

PIO SDIO uses one PIO block.  RP2040 has two blocks, RP2350 has three blocks.  SdFat checks for free PIO resources and reserves them.  Some PIO interfaces libraries do not check/reserve PIO resources and will not work reliably.

Stass48, 2025-03-11:
Thank you for your answer 🙏

_You can use many SPI cards but only one SDIO card. There is not sufficient memory in a PIO block to add support for multiple cards on one PIO controller._

Does this apply only to the RP2040? Or does the RP2350 have the same issue with a lack of PIO memory?

greiman, 2025-03-11:
> Does this apply only to the RP2040? Or does the RP2350 have the same issue with a lack of PIO memory?

PIO blocks in the RP2350 have the same amount of memory as the RP2040.  

I suspect there could be problems implementing more SDIO features in PIO.  I was only able to implement the most basic SDIO features with PIO.

Stass48, 2025-03-11:
Thank you!

Sorry for the off-topic.
I just got the right SD card slots for SDIO, wrote the code, but for some reason it doesn't work. For some reason the write() function returns 0. I can't figure out why this is happening. In SPI mode this code works. Are the wires really long? I'm using standard Arduino wires about 15 cm long. The memory card is successfully initialized in SDIO, but no data is written.

```
void CopyFile(const char* fileName) {
  unsigned long copyTime = 0;
  uint16_t len = 0;
  uint8_t fileBuff[fileBuffSize];

  FsFile srcFile;
  FsFile dstFile;

  srcFile = SrcSD.open(fileName, O_RDONLY);
  if (!srcFile) {
    Serial.println("Cannot open file on SrcSD!");
    while (1) delay(10);
  }

  dstFile = DstSD.open(fileName, O_RDWR | O_CREAT | O_TRUNC);
  if (!dstFile) {
    Serial.println("Cannot open file on DstSD!");
    while (1) delay(10);
  }

  Serial.println("Files are opened/created. Lets start!");
  copyTime = millis();

  while (true) {
    int n = srcFile.read(fileBuff, sizeof(fileBuff));
    if (n < 0) {
      Serial.println("read error");
      return;
    }
    if (n == 0) {
      break;
    }

    int x = dstFile.write(fileBuff, n);

    if (x != (unsigned)n) {
      Serial.println(n);
      Serial.println(x);
      Serial.println("write error");
      return;
    }
  }

  srcFile.close();
  dstFile.close();

  copyTime = millis() - copyTime;
  Serial.print("File was copied at ");
  Serial.print(copyTime);
  Serial.println(" ms!");
  Serial.println("Done!");
}
```

_Source SD card ready!
Dest SD card ready!
Ready for copying. Send any data to start.
Files are opened/created. Lets start!
512
0
write error_

Thank you!

Stass48, 2025-03-11:
Thank you, Bill. You're amazing!

bastian2001, 2025-03-27:
@greiman Thank you for the amazing library! While I was trying to find out which resources the SDIO implementation requires, I found out that there might be a bug:
PioSdioCard.cpp claims two state machines, but it only checks (twice) if the first one (g_sm0) was assigned successfully (in lines 284 + 289).

Furthermore, with regards to this issue, if I am correct about this two state machine assumption, doesn't that mean that the next two state machines on the same block can run another SDIO interface? You can reuse the same instructions on multiple SMs at once, by just referencing them to the same program offsets. So the only limitation would be to check whether there is already an instance of an SDIO interface active, and just return the same program offsets with the other two SMs of the same block. Also, if I understand the code correctly, it is already possible to run multiple interfaces on multiple blocks as well? So up to 4 SDIO cards on RP2040 and 6 on RP2350 (or with the current implementation 2 or 3 SDIO cards, one on each block?). I don't have a way to test this, as I am just about to order my first PCB with _one_ SDIO card on it, but I think it should be possible.

greiman, 2025-03-27:
I am currently making many mods in the programs. I will fix the check for assignment program.

The normal way to run multiple cards in on a single bus.  I have done none of the code to handle this.

I also modify the program in memory to embed a GPIO number.