greiman/SdFat #517: Problem in PioSdioCard.cpp file

scrippa74, 2025-03-20:
Hi,

I have some problems to understand the initialization loop located in the "SdioCard::begin(SdioConfig sdioConfig)" function at the line 635 of thePioSdioCard.cpp file.

I suppose that the goal of this loop is to retry the "CMD0_R0" command in case of a power up delay of the SD card, is right?
If this is the case, the loop has a problem, in fact if the command "CMD0_R0" fails the loop will not be executed "CMD0_RETRIES" times, but it will exit immediately with the error code "SD_CARD_ERROR_CMD0".

This loop doesn't work also if the goal is to send the 74 dummy clock's shots needed by the SD card after the power up sequence. In fact, also in this case, the loop will terminate immediately as the "CMD0_R0" command fails. 

[image]

Also the "CMD8_R7" loop located at the line 642 could have some problems.

Thanks a lot

greiman, 2025-03-20:
> I have some problems to understand the initialization loop 

Yes this doesn't look like the SD Association Standard flow charts.  A few cards will not initialize without retries.

I have tried sending dummy clocks also but it it not as effective as retrying CMD0.

I think this mainly happens when a card is reinitialized without cycling power.

I should initialize the card at a much lower clock rate but the SD cards have strange timing and just reducing the clock rate doesn't work.  PIO just isn't like a real SDIO controller.  I gave up trying for "High Speed mode" so the card id overclocked since Default mode is 25 MHz max.

So I can't match a real SDIO controller with PIO and this seems to work.
 Here are time charts:

[image]

[image]

scrippa74, 2025-03-20:
Ok, thanks for your explanation.

closed 2025-03-20 by scrippa74

greiman, 2025-03-21:
I want to thank you for reminding me about the strange code in `SdioCard::begin`.

I have been working on fixing it. I was able to slow SD init to 400 kHz.

`SdioCard::begin` seems to work with the standard initialization sequence.

```
  if (!cardCmd(CMD0_R0, 0)) {
    sdError(SD_CARD_ERROR_CMD0);
    goto fail;
  }
  if (cardCmd(CMD8_R7, 0X1AA)) {
    if (g_cardRsp != 0X1AA) {
      sdError(SD_CARD_ERROR_CMD8);
      goto fail;
    }
    g_version2 = true;
  } else {
    g_version2 = false;
    g_errorCode = SD_CARD_ERROR_NONE;
  }
```

Here is a Canvas Go! Plus at 276 MHz on a AdaFruit Merto RP2040.  The Metro has an onboard socket so it is best for high speed.

> clk_sys: 276000000
> g_clkDiv: 690.00
> Type is exFAT
> Card size: 124.49 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X9F
> OEM ID: TI
> Product: SD128
> Revision: 6.1
> Serial number: 0X827800F2
> Manufacturing date: 9/2020
> 
> FILE_SIZE_MB = 20
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 26809.31,85,18,18
> 26315.45,91,18,18
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 26109.33,497,19,19
> 26143.46,502,19,19

It even works at  276 MHz with a cheap 128 MB Chinese card. Slow write but what you would expect for a $2 card with no SD Association logo.

> clk_sys: 276000000
> g_clkDiv: 690.00
> Type is FAT16
> Card size: 0.13 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X0
> OEM ID: 0
> Product: APPSD
> Revision: 0.0
> Serial number: 0X6B8
> Manufacturing date: 5/2018
> 
> FILE_SIZE_MB = 20
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4757.37,48526,18,107
> 4853.19,47886,18,105
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 18993.35,50,18,26
> 19011.41,50,19,26

scrippa74, 2025-03-24:
Oh, well, really thanks to you for the great job you have done! ;)