greiman/SdFat #519: SdFormatter error

rtek1000, 2025-04-22:
Hello,

I tried the example but it gave these errors below. How can I make it work?

---- SdFormatter:

https://github.com/greiman/SdFat/tree/master/examples/SdFormatter

> Warning, all data on the card will be erased.
> Enter 'Y' to continue: Y
> error: Get sector count failed.
> SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x19
> SD errorData = 0x20

---- SdInfo:

https://github.com/greiman/SdFat/blob/master/examples/SdInfo/SdInfo.ino

> SdFat version: 2.3.0
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CS_PIN to disable an SPI device.
> 
> Assuming the SD chip select pin is: 5
> Edit SD_CS_PIN to change the SD chip select pin.
> 
> type any character to start
> init time: 3 ms
> 
> Card type: SDHC
> sdSpecVer: 3.00
> HighSpeedMode: true
> 
> Manufacturer ID: 0X6F
> OEM ID: 
> Product: CBADS
> Revision: 1.0
> Serial number: 0XAA000609
> Manufacturing date: 10/2022
> CID HEX: 6F0303434241445310AA000609016A01
> 
> cardSize: 15849.75 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> CSD HEX: 400E00325B59000076167F800A404001
> 
> OCR: 0XC0FF8080
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X80,0XFF,0XFF,0XFF,0XC,0XFF,0XFF,0XFF,2048,30954496
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT32
> sectorsPerCluster: 64
> fatStartSector:    2108
> dataStartSector:   9664
> clusterCount:      483545
> freeClusterCount:  483510
> 
> type any character to start
>

greiman, 2025-04-22:
Looks like a card not certificated by the SD Association.  Reading the card size failed.

Since the SdInfo example read the card size I suspect your setup/card has a problem with SPI speed.  

What card are you using?  Some boards select a speed that is marginal.  ESP32 is one case.

SdFormatter requests the highest supported speed less than or equal to 50 MHz [here](https://github.com/greiman/SdFat/blob/a4ecec60554e8071e3f15a4407cad3908519c06e/examples/SdFormatter/SdFormatter.ino#L41).

SdInfo uses a slow SPI clock.  See [this](https://github.com/greiman/SdFat/blob/a4ecec60554e8071e3f15a4407cad3908519c06e/examples/SdInfo/SdInfo.ino#L40).

Try 16 MHz or less.

rtek1000, 2025-04-22:
Ok, using slow SPI clock works.

- Is there any way to test what speed the card supports before trying to format it?

Board: https://github.com/rtek1000/2.4inch_ESP32-2432S024

[image]

[image]

Thank you.

> Assuming the SD is the only SPI device.
> Edit DISABLE_CS_PIN to disable an SPI device.
> 
> Assuming the SD chip select pin is: 5
> Edit SD_CS_PIN to change the SD chip select pin.
> 
> Type any character to start
> 
> This program can erase and/or format SD/SDHC/SDXC cards.
> 
> Erase uses the card's fast flash erase command.
> Flash erase sets all data to 0X00 for most cards
> and 0XFF for a few vendor's cards.
> 
> Cards up to 2 GiB (GiB = 2^30 bytes) will be formated FAT16.
> Cards larger than 2 GiB and up to 32 GiB will be formatted
> FAT32. Cards larger than 32 GiB will be formatted exFAT.
> 
> Warning, all data on the card will be erased.
> Enter 'Y' to continue: Y
> 
> Card size: 15.85 GB (GB = 1E9 bytes)
> Card size: 14.76 GiB (GiB = 2^30 bytes)
> Card will be formated FAT32
> 
> Options are:
> E - erase the card and skip formatting.
> F - erase and then format the card. (recommended)
> Q - quick format the card without erase.
> 
> Enter option: F
> 
> Erasing
> ................................................................
> .......................................................
> All data set to 0xff
> Erase done
> Writing FAT ................................
> Format Done
> Run the SdInfo example for format details.

greiman, 2025-04-22:
> Is there any way to test what speed the card supports before trying to format it?

SPI is poorly supported on ESP32 so it's trial and error.

Here is how it should work, for SD cards the max supported speed is 50 MHz.  

A quote from Arduino:

> What is the maximum SPI speed your device can use? This is controlled by the first parameter in SPISettings. If you are using a chip rated at 15 MHz, use 15000000. Arduino will automatically use the best speed that is equal to or less than the number you use with SPISettings.