greiman/SdFat #520: Example program BackwardCompatiblity.ino doesn't work with SDfat

monsteroyd, 2025-04-25:
I tried this while trying to learn how to use sdfat. I can't get beyond SD.Begin(). It always returns zero when using the SDFat library. it works fine with the SD.h library. I think there needs to be an sdconfig thing defined and passed to SD.begin. I am guessing this based on the code for begin. 

I tested this on an Arduino Uno R4 wifi.  

The backward compatibility program works find when run with sd.h but fails when using sdfat.h.

Monty Shaw
mshaw@microft.com

greiman, 2025-04-26:
It may be your SD socket/module or wiring.  SdFat uses a much higher default SPI clock than SD.h.  SD.h is based on an old version of SdFat that I wrote in 2009 and used 4 MHz as the default clock.

To get full performance from SdFat you will need a quality SD module with short clean wiring and learn to use SdSpiConfig.

See [this](https://github.com/greiman/SdFat/blob/a4ecec60554e8071e3f15a4407cad3908519c06e/examples/bench/bench.ino#L53) area of the bench example.

For modern SD cards you can only get high performance with dedicated SPI.  Also note that the Uno R4 has a very poor implementation of SPI.

Here is the default result of bench on an Uno R4 with an Adafruit SD [shield](https://www.adafruit.com/product/1141), no better than a Uno R3:

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 612.67,851,830,832
> 612.59,851,830,833
> 
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 591.30,865,860,863
> 591.30,865,860,863

Here is an old Arduino Due:

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4537.21,1149,110,111
> 4537.21,128,110,111
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4545.45,113,110,110
> 4553.73,112,110,110

monsteroyd, 2025-04-26:
Thanks for the quick reply. 

I did get the example to work by adding this around line 20:
#define SPI_CLOCK 25
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)

I also added this further down in the setup function:
#if USE_SD_H          
  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
#else
  if (!SD.begin(SD_CONFIG)) {
    SD.initErrorHalt(&Serial);
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
#endif

OK I probably brought the clock down by using 25.

What is a good value for SPI_CLOCK? I don't know where this is defined, as I had to add my own #define SPI_CLOCK to get it to compile.

Monty

greiman, 2025-04-26:
> What is a good value for SPI_CLOCK? I don't know where this is defined, as I had to add my own #define SPI_CLOCK to get it to compile.

See [this](https://github.com/greiman/SdFat/issues/519#issuecomment-2821931864).

With Uno R4 and an Adafruit SD Shield I use 50 MHz.  The SPI library should choose the highest speed the board will support that is less than or equal to 50 MHz.