greiman/SdFat #521: RP2040 fast logging

Anjali8340, 2025-05-05:
I wanted to log data into sd card using RP2040. There is no example code for this, so can I use TeensySDIOlogger example for reference and use Ring buffer with RP2040 also?

greiman, 2025-05-05:
The basic idea of using `isBusy()` in TeensySdioLogger works with RP2040 SDIO, Teensy SDIO or dedicated SPI.

I tested with a Adafruit Metro RP2040 and a CPU clock of 200 MHz.  The max time to write a 512 byte sector when the SD is not busy is about 35 µs.

The max busy time depends on the SD so using an SD rated [U3 A3 V30](https://www.sdcard.org/press/thoughtleadership/the-more-you-know-the-evolution-of-sd-ratings-for-new-applications/).

greiman, 2025-05-05:
I did a quick hack of TeensySdioLogger to test RP2040 results.  I found a limit of about 100 µs interval between points.

The reason for the limit is there is enough code that executes from flash using XIP.  I forced much of the PIO SDIO code to be in ram with the `__time_critical_func` macro but there are still delays for remaining code needed to log data.

Here is the hacked program, Metro RP2350 is not yet supported:

```
#include "SdFat.h"
#include "RingBuf.h"

// Example SDIO definition for RP2040/RP2350. See the Rp2040SdioSetup example.
#if defined(ARDUINO_ADAFRUIT_METRO_RP2040)
#define RP_CLK_GPIO 18
#define RP_CMD_GPIO 19
#define RP_DAT0_GPIO 20  // DAT1: GPIO21, DAT2: GPIO22, DAT3: GPIO23.
#elif defined(ARDUINO_ADAFRUIT_METRO_RP2350)
#define RP_CLK_GPIO 34
#define RP_CMD_GPIO 35
#define RP_DAT0_GPIO 36  // DAT1: GPIO37, DAT2: GPIO38, DAT3: GPIO39.
#endif  // defined(ARDUINO_ADAFRUIT_METRO_RP2040)

// Try to select the best SD card configuration.
#if defined(HAS_TEENSY_SDIO)
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif defined(RP_CLK_GPIO) && defined(RP_CMD_GPIO) && defined(RP_DAT0_GPIO)
// See the Rp2040SdioSetup example for RP2040/RP2350 boards.
#define SD_CONFIG SdioConfig(RP_CLK_GPIO, RP_CMD_GPIO, RP_DAT0_GPIO)
#elif ENABLE_DEDICATED_SPI
// SDCARD_SS_PIN is defined for the built-in SD on some boards.
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = SS;
#else   // SDCARD_SS_PIN
// Assume built-in SD is used.
const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
#endif  // SDCARD_SS_PIN
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_TEENSY_SDIO
#error "Shared SPI is not supported"
#endif  // HAS_TEENSY_SDIO

#define LOG_INTERVAL_USEC 100  // 200 would be safer

// Size to log 10 byte lines at 25 kHz for more than ten minutes.
#define LOG_FILE_SIZE 10 * 25000 * 600  // 150,000,000 bytes.

// Space to hold 40 sectors
#define RING_BUF_CAPACITY 40 * 512
#define LOG_FILENAME "SdioLogger.csv"

SdFs sd;
FsFile file;

// RingBuf for File type FsFile.
RingBuf<FsFile, RING_BUF_CAPACITY> rb;

void __time_critical_func(logData)() {
  // Initialize the SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  // Open or create file - truncate existing file.
  if (!file.open(LOG_FILENAME, O_RDWR | O_CREAT | O_TRUNC)) {
    Serial.println("open failed\n");
    return;
  }
  // File must be pre-allocated to avoid huge
  // delays searching for free clusters.
  if (!file.preAllocate(LOG_FILE_SIZE)) {
    Serial.println("preAllocate failed\n");
    file.close();
    return;
  }
  // initialize the RingBuf.
  rb.begin(&file);
  Serial.println("Type any character to stop");

  // Max RingBuf used bytes. Useful to understand RingBuf overrun.
  size_t maxUsed = 0;

  // Min spare micros in loop.
  int32_t minSpareMicros = INT32_MAX;

  // Start time.
  uint32_t logTime = micros();
  // Log data until Serial input or file full.
  while (!Serial.available()) {
    // Amount of data in ringBuf.
    size_t n = rb.bytesUsed();
    if ((n + file.curPosition()) > (LOG_FILE_SIZE - 20)) {
      Serial.println("File full - quitting.");
      break;
    }
    if (n > maxUsed) {
      maxUsed = n;
    }
    if (n >= 512 && !file.isBusy()) {
      // Not busy only allows one sector before possible busy wait.
      // Write one sector from RingBuf to file.
      if (512 != rb.writeOut(512)) {
        Serial.println("writeOut failed");
        break;
      }
    }
    // Time for next point.
    logTime += LOG_INTERVAL_USEC;
    int32_t spareMicros = logTime - micros();
    if (spareMicros < minSpareMicros) {
      minSpareMicros = spareMicros;
    }
    if (spareMicros <= 0) {
      Serial.print("Rate too fast ");
      Serial.println(spareMicros);
      break;
    }
    // Wait until time to log data.
    while (micros() < logTime) {
    }

    // Read ADC0 - about 17 usec on Teensy 4, Teensy 3.6 is faster.
    uint16_t adc = analogRead(A0);
    // Print spareMicros into the RingBuf as test data.
    rb.print(spareMicros);
    rb.write(',');
    // Print adc into RingBuf.
    rb.println(adc);
    if (rb.getWriteError()) {
      // Error caused by too few free bytes in RingBuf.
      Serial.println("WriteError");
      break;
    }
  }
  // Write any RingBuf data to file.
  rb.sync();
  file.truncate();
  file.rewind();
  // Print first twenty lines of file.
  Serial.println("spareMicros,ADC0");
  for (uint8_t n = 0; n < 20 && file.available();) {
    int c = file.read();
    if (c < 0) {
      break;
    }
    Serial.write(c);
    if (c == '\n') n++;
  }
  Serial.print("fileSize: ");
  Serial.println((uint32_t)file.fileSize());
  Serial.print("maxBytesUsed: ");
  Serial.println(maxUsed);
  Serial.print("minSpareMicros: ");
  Serial.println(minSpareMicros);
  file.close();
  sd.end();
}
void clearSerialInput() {
  for (uint32_t m = micros(); micros() - m < 10000;) {
    if (Serial.read() >= 0) {
      m = micros();
    }
  }
}
void setup() {
  Serial.begin(9600);
  while (!Serial) {
  }
  // Go faster or log more channels.  ADC quality will suffer.
  // analogReadAveraging(1);
}

void loop() {
  clearSerialInput();
  Serial.println("Type any character to start");
  while (!Serial.available()) {
  }
  clearSerialInput();
  logData();
}
```