greiman/SdFat #525: Issue with timestamp() and openNext() in SdFs File Traversal

Bob128x, 2025-06-26:
Dear Mr. Greiman,

I hope you're doing well.
I have another question and would appreciate your help once again.

I am trying to implement a function that:

1.  Counts the total size of all files in a directory (including subdirectories).
2.  Optionally updates the timestamp (date/time) of each file and folder to a user-defined value.

The file size counting works perfectly. However, when I enable the timestamp update using entry.timestamp(), only the first file is updated. After that, the while (entry.openNext(...)) loop stops working — it seems that modifying the timestamp interferes with the directory traversal.

If I disable timestamp updates, the full counting process completes successfully. So the problem seems to occur only after calling timestamp() while traversing the directory with openNext().

Would you be willing to look over my code snippet and possibly explain why this behavior occurs?

Here is the relevant part:

```
SdFs sd;
FsFile binFile;

vod startCountPlusDate() {

 if (sd.volumeBegin() && binFile.open("/")) {
  
            m_dirCount = 0;
            m_fileCount = 0;
            m_totalSize = 0;
            countFiles(&binFile, true);
            binFile.close();
  }
}

static void countFiles(FsFile *dir, bool ApplyNewDate) {
  FsFile entry;
  bool openMode = ApplyNewDate ? O_RDWR : O_RDONLY;

  while (entry.openNext(dir, openMode)) {
    uint8_t attr = entry.attrib();

    if (attr & (ATTR_HIDDEN | ATTR_SYSTEM | ATTR_VOLUME_ID)) {
      entry.close();
      continue;
    }

    if (entry.isDir()) {
      if (ApplyNewDate) {
        entry.timestamp(T_CREATE | T_WRITE | T_ACCESS,
                        (uint16_t)(m_myDate[0] + 1970),
                        m_myDate[1], m_myDate[2],
                        m_myDate[3], m_myDate[4], m_myDate[5]);
      }

      countFiles(&entry, ApplyNewDate); // recursive
    } else if (entry.isFile()) {
      m_fileCount++;
      m_totalSize += entry.fileSize();

      if (ApplyNewDate) {
        entry.timestamp(T_CREATE | T_WRITE | T_ACCESS,
                        (uint16_t)(m_myDate[0] + 1970),
                        m_myDate[1], m_myDate[2],
                        m_myDate[3], m_myDate[4], m_myDate[5]);
      }
    }

    entry.close();
  }
}
```

Thank you very much in advance for your time and for your great work on the SdFs library!

Kind regards

greiman, 2025-06-26:
I see one problem, openNext() will fail for a directory file if open mode is O_RDWR.  Any open call will fail for a directory file if the mode is write.

You should not need write mode to change a files timestamp since it does not change the file's content.

Bob128x, 2025-06-26:
Thank you very much for your previous help — your suggestion was exactly right.

Now the code successfully iterates through all files and updates the timestamps correctly. However, I’ve noticed that the folder timestamps are not being changed, even though I call timestamp() on directories as well.

From what I observe, directories usually seem to have only the creation date/time set. I also tried updating only the creation time using:

`entry.timestamp(T_CREATE, year, month, day, hour, minute, second);`

But this didn’t seem to have any effect.

Is it generally possible to update folder timestamps using timestamp() in the SdFs library? Or are there limitations from the FAT32 specification or the library’s implementation?

Any further insight or suggestions would be greatly appreciated.

greiman, 2025-06-27:
Older versions of SdFat did not allow setting times on directory files.  See [this](https://github.com/greiman/SdFat/issues/448).

I believe this version has the mod.

Try this test program:

```
/*
 * This program tests the timestamp() function.
 */
#include "SdFat.h"

SdFat sd;
FsFile file;

// SDCARD_SS_PIN is defined for the built-in SD on some boards.
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = SS;
#else   // SDCARD_SS_PIN
// Assume built-in SD is used.
const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
#endif  // SDCARD_SS_PIN

// Try max SPI clock for an SD. Reduce SPI_CLOCK if errors occur.
#define SPI_CLOCK SD_SCK_MHZ(50)

// Try to select the best SD card configuration.
#if defined(HAS_TEENSY_SDIO)
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif defined(HAS_BUILTIN_PIO_SDIO)
// See the Rp2040SdioSetup example for boards without a builtin SDIO socket.
#define SD_CONFIG SdioConfig(PIN_SD_CLK, PIN_SD_CMD_MOSI, PIN_SD_DAT0_MISO)
// Definitions for my Pico debug tests when to zero and // are removed.
#elif 0  // defined(ARDUINO_RASPBERRY_PI_PICO) || defined(ARDUINO_RASPBERRY_PI_PICO_2)
// CLK: GPIO10, CMD: GPIO11, DAT[0,3]: GPIO[12, 15].
#define SD_CONFIG SdioConfig(10u, 11u, 12u)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_TEENSY_SDIO
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
#endif  // HAS_TEENSY_SDIO
//------------------------------------------------------------------------------
// store error strings in flash to save RAM
#define error(s) sd.errorHalt(F(s))
//------------------------------------------------------------------------------
/*
 * Function to print all timestamps.
 */
void printTimestamps(FsFile& f) {
  Serial.print("Name: ");
  f.printName(&Serial);
  Serial.print("\nCreation: ");
  f.printCreateDateTime(&Serial);
  Serial.print("\nModify: ");
  f.printModifyDateTime(&Serial);
  Serial.print("\nAccess: ");
  f.printAccessDateTime(&Serial);
  Serial.println();
}
//------------------------------------------------------------------------------
void setup(void) {
  Serial.begin(9600);
  // Wait for USB Serial
  while (!Serial) {
    yield();
  }
  Serial.println("Type any character to start");
  while (!Serial.available()) {
    yield();
  }
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt();
  }

#define USE_TEST_DIR 1
#if USE_TEST_DIR
  // remove TestDir if it exist
  sd.rmdir("TestDir");
  if (!sd.mkdir("TestDir")) {
    error("make TesrDir failed");
  }
  if (!file.open("TestDir")) {
    error("open TestDir failed");
  }
#else
  sd.remove("TestFile.txt");  
    // create a new file with default timestamps
  if (!file.open("TestFile.txt", O_WRONLY | O_CREAT)) {
    error("open stamp.txt failed");
  }
#endif
   Serial.println("\nDefault Times");
  printTimestamps(file);
  // set creation date time
  if (!file.timestamp(T_CREATE, 2024, 11, 10, 1, 2, 3)) {
    error("set create time failed");
  }
  // set write/modification date time
  if (!file.timestamp(T_WRITE, 2024, 11, 11, 4, 5, 6)) {
    error("set write time failed");
  }
  // set access date
  if (!file.timestamp(T_ACCESS, 2024, 11, 12, 7, 8, 9)) {
    error("set access time failed");
  }
  Serial.println("\nTimes after timestamp() calls");
  printTimestamps(file);

  file.close();
  Serial.println("\nDone");
}

void loop() {}
```
When I run it with FAT32:

> Default Times
> Name: TestDir
> Creation: 2025-01-01 00:00
> Modify: 2025-01-01 00:00
> Access: 2025-01-01
> 
> Times after timestamp() calls
> Name: TestDir
> Creation: 2024-11-10 01:02
> Modify: 2024-11-11 04:05
> Access: 2024-11-12

When I run it with exFAT:

> Default Times
> Name: TestDir
> Creation: 2025-01-01 00:00
> Modify: 2025-01-01 00:00
> Access: 2025-01-01 00:00
> 
> Times after timestamp() calls
> Name: TestDir
> Creation: 2024-11-10 01:02
> Modify: 2024-11-11 04:05
> Access: 2024-11-12 07:08

greiman, 2025-06-27:
I am curious, what version of SdFat are you using?

The current version, 2.3.0, has the correct calls [here](https://github.com/greiman/SdFat/blob/a4ecec60554e8071e3f15a4407cad3908519c06e/src/ExFatLib/ExFatFileWrite.cpp#L466) and [here](https://github.com/greiman/SdFat/blob/a4ecec60554e8071e3f15a4407cad3908519c06e/src/FatLib/FatFile.cpp#L1284).

It is best to use the current release version since there are usually many fixes and feature changes in each release.  If you find a problem in a new release, try [SdFat-beta](https://github.com/greiman/SdFat-beta), and if the beta still has the problem, post an issue.

I limit the number of releases so the Arduino IDE does not press users to upgrade too often.

Bob128x, 2025-06-27:
I decided to stick with SdFat v2.2.2. 

The frequent updates didn’t bring anything relevant to my project, so I saw no reason to update.

But you’re right — I’ll plan to bring it up to date later this year.

greiman, 2025-06-28:
> I decided to stick with SdFat v2.2.2.

There have been a number of bugs fixed since that version and I will be posting a new version with more bugs fixed including a fix to timestamp() that rejects invalid dates that are not caught in the current version.  Many useful feature like times on directory files are added to each release.

I suggest you wait till the next release which will be in about two months since I am about to post what I hope is the final beta. 

 I tend to post one or two releases per year.

Bob128x, 2025-06-29:
Thanks for the advice!

closed 2025-06-29 by Bob128x