greiman/SdFat #533: UnicodeFilenames

VantagePro2, 2025-08-13:
Good evening!
I'm trying to run an example of a UnocodeFilenames sketch on an Arduino Due board.
I am not changing the source code.
My first message in Serial Monitor is
_Type any character to begin
begin() failed
Do not reformat the SD.
No card, wrong chip select pin, or wiring error?
SdError: 0X1,0XFF__

When I use the usual example from the Arduino IDE CardInfo, I get the message:

_Initializing SD card...Wiring is correct and a card is present.

Card type:         SD2
Clusters:          61109
Blocks x Cluster:  64
Total Blocks:      3910976

Volume type is:    FAT16
Volume size (KB):  1955488
Volume size (MB):  1909
Volume size (GB):  1.86

Files found on the card (name, date and size in bytes): 
RECYCLER/     2025-07-23 09:16:00
  S-5-3-~1/     2025-07-23 09:16:00
    JWGKVSQ.VMX   2008-04-15 15:00:00 164746
_~1           2025-01-01 00:00:00 0
AUTORUN.INF   2008-04-15 15:00:00 59306_

Do I need additional settings for the SdFat.h library for this example?

greiman, 2025-08-14:
Looks like you need to edit [this](https://github.com/greiman/SdFat/blob/cda057318bec196183d4cc92b01bc1dd64bbfb02/examples/UnicodeFilenames/UnicodeFilenames.ino#L27) line to use pin 4 for SD chip select.

If there is still a problem reduce SPI CLK rate [here](https://github.com/greiman/SdFat/blob/cda057318bec196183d4cc92b01bc1dd64bbfb02/examples/UnicodeFilenames/UnicodeFilenames.ino#L40) and [here](https://github.com/greiman/SdFat/blob/cda057318bec196183d4cc92b01bc1dd64bbfb02/examples/UnicodeFilenames/UnicodeFilenames.ino#L42).

closed 2025-08-16 by VantagePro2

VantagePro2, 2025-08-19:
Good morning,
The QuickStart works normal:
I
SPI pins:
MISO: 74
MOSI: 75
SCK:  76
SS:   10

Be sure to edit DISABLE_CHIP_SELECT if you have
a second SPI device.  For example, with the Ethernet
shield, DISABLE_CHIP_SELECT should be set to 10
to disable the Ethernet controller.

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Enter the chip select pin number: 4

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Card successfully initialized.

Card size: 15829 MB (MB = 1,000,000 bytes)

Volume is FAT32, Cluster size (bytes): 8192

Files found (date time size name):

This card should be reformatted for best performance.
Use a cluster size of 32 KB for cards larger than 1 GB.
Only cards larger than 2 GB should be formatted FAT32.
Try reformatting the card.  For best results use
the SdFormatter program in SdFat/examples or download
and use SDFormatter from www.sdcard.org/downloads.

Restarting

Enter the chip select pin number:

VantagePro2, 2025-08-19:
I use two different SD cards. 
I don't think they both refused at the same time.
The error appeared after I tried to change the code of the UnicodeFilenames example.
I wanted to disable file deletion and commented out the lines

_#if REMOVE_UTF8_FILES  // For debug test of remove and rmdir.
  for (uint8_t i = 0; names[i]; i++) {
    sd.remove(names[i]);
  }
  sd.chdir();
  sd.rmdir(UTF8_FOLDER);
  Serial.println("After remove and rmdir");
  sd.ls(LS_SIZE | LS_R);
#endif  // REMOVE_UTF8_FILES_

Could this be the cause of the error?

greiman, 2025-08-19:
> Could this be the cause of the error?

The startup error: SdError: 0X1,0XFF

Has nothing to do with what is on the SD or how the SD is formatted.  The program is not able to access the SD.  You need to be sure all the options in the Unicode example match your wiring.

VantagePro2, 2025-08-19:
I compared QuickStart & UnicodeFilenames.
Now it works.
Thanks