greiman/SdFat #536: type u8 in UnicodeFilenames

VantagePro2, 2025-09-03:
Good evening!

There are such stroke
_const char* names[] = {u8"россиянин", u8"très élégant", u8"狗.txt", nullptr};_
in UnicodeFilenames.ino

Would you like to you explain me, please?

I form the file name using ds3231 library such way:

_HR+MIN+DATA+MONTH+".dat"_

**See example:**

// Get data from DS3231
int minute,hour,date,month; 
minute=Clock.getMinute();
hour=Clock.getHour(h12, PM);
date=Clock.getDate();
month=Clock.getMonth(Century);

//Hours
if (hour < 10)
  HR =  "0"+String(hour, DEC); 
else
  HR =  String(hour, DEC); 
// Minutes
if (minute < 10)
  MIN =  "0"+String(minute, DEC); 
else
  MIN =  String(minute, DEC); 
// DATA
if (date < 10)
 DATE = "0"+String(date, DEC); 
else
 DATE = String(date, DEC); 
// MONTH 
if (month < 10)
 MTH = "0"+String(month, DEC); 
else
 MTH =  String(month, DEC);

 String stroke = MTH+DATE+HR+MIN+".dat";

How do I define the array _names[]_ using the variable _stroke_ ?
How do I set the type _u8_ and use quotation marks _""_?

VantagePro2, 2025-09-04:
In the example, you specify the file name explicitly in quotation marks.
For example, u8"россиянин"
Can I define the file name as a variable (for example, String stroke) without explicitly specifying it in quotes?

greiman, 2025-09-04:
Yes you can specify the file name as a variable.  The variable can be a literal, String, char array, or char pointer.

 The current version of SdFat requires you to use c_str() with String.

Here is an example:
```
#include "SdFat.h"
FsFile file0, file1, file2, file3;

void utf8Example() {
  String name1 = "09040853.dat";
  const char* name2 = "09040854.dat";
  char name3[] = "0904085.dat";

  file0.open("09040854.dat", O_RDWR | O_CREAT);
  file1.open(name1.c_str(), O_RDWR | O_CREAT);
  file2.open(name2, O_RDWR | O_CREAT);
  file3.open(name3, O_RDWR | O_CREAT);
}
```

VantagePro2, 2025-09-06:
Thanks for the hint, it really helped.

closed 2025-09-06 by VantagePro2